JMRI/JMRI #1597: Unit Tests cause settings: to be created.

rhwood, 2016-06-29:
On a computer where a JMRI release has never been run, running the unit tests (```ant all-tests```) causes the ```settings:``` directory to be populated. It should not touch that directory.

I think the solution is to ensure that ant and any processes that run tests ensure that the ```jmri.prefsdir``` property is set to a temporary file (or to a directory ignored by git within a local repository) before tests are run.

bobjacobsen, 2016-06-29:
Good point.

Some of the tests now create & use a `temp` directory in the program directory. Would that be a good convention for setting `settings:`?

pabender, 2016-06-29:
I think that would be a good idea.

I went through a lot of headache to make sure the operations tests were not writing to the user's directory last year.  That was certainly a beneficial task though.  

Those tests write to the "temp" directory in the root of the source tree.

Paul

rhwood, 2016-06-30:
While working on this, I found the ```alltest``` and ```quicktest``` targets are not in ```junit``` ant tasks, but all other test targets are. Is this deliberate, an oversight, or a historical artifact?

rhwood, 2016-06-30:
#1600 fixes this for tests run through ant.

rhwood, 2016-07-23:
This is fixed as originally proposed in #1600 and #1691/#1692.

closed 2016-07-23 by rhwood