JMRI/JMRI #1602: Create Test Release 4.5.1

mattharris, 2016-06-30:
For discussion of timing and what to include in 4.5.1

bobjacobsen, 2016-07-05:
Current plan is to build this from HEAD sometime late Friday night or early morning Saturday (California time). Please post here if that's inconvenient or of any killer conditions, please.

bobjacobsen, 2016-07-09:
The release-4.5.1 branch has been created. 

From now on, please document your changes in the [jmri4.5.2.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.5.2.shtml) release note file.

Maintainers, please set the 4.5.2 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [new server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.1/)

bobjacobsen, 2016-07-16:
A belated close of this, sorry.

closed 2016-07-16 by bobjacobsen