JMRI/JMRI #1639: Errors in console when using ArduinoCMRI?

k4kfh, 2016-07-09:
I'm not sure if this is a JMRI bug or an ArduinoCMRI bug. Whenever I use ArduinoCMRI, I get a handful of these errors in the console.

```
2016-07-07 16:21:27,084 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 10 was 41 52 00 00 00 [Receive]
2016-07-07 16:21:27,089 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 10 was 41 52 00 00 00 [Receive]
2016-07-07 16:21:33,494 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 15 was 41 52 00 00 00 [Receive]
2016-07-07 16:21:33,498 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 15 was 41 52 00 00 00 [Receive]
2016-07-07 16:21:33,554 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 15 was 41 52 00 00 00 [Receive]
```

I'm not knowledgeable enough in programming the Arduino to do any kind of debugging on that library, or I'd provide more information than this. Happy to do testing/provide more logs if needed.

bobjacobsen, 2016-07-09:
> On Jul 9, 2016, at 12:52 PM, Hampton <notifications@github.com> wrote:
> 
> I'm not sure if this is a JMRI bug or an ArduinoCMRI bug. Whenever I use ArduinoCMRI, I get a handful of these errors in the console.
> 
> 2016-07-07 16:21:27,084 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 10 was 41 52 00 00 00 [Receive]

This means that the reply from the Arduino arrived, but later than it should have.  Is the Arduino doing something that keeps it busy?

k4kfh, 2016-07-09:
It may be, I'm not really sure. I wasn't sure what the error even meant, so thanks for letting me know. I'll tinker with it a little, but if you want we can close this issue since it's most likely an ArduinoCMRI bug or a bug in my own Arduino code.

madleech, 2016-07-13:
All fixed, see madleech/ArduinoCMRI#6

bobjacobsen, 2016-07-13:
Glad you were able to work it out. Thanks for letting us know.

closed 2016-07-13 by bobjacobsen