JMRI/JMRI #1656: Linux user gets wrong Java version with JMRI 4.5.1

bobjacobsen, 2016-07-13:
A Linux user with both Java 1.5 and Java 1.8 installed isn't able to start ./PanelPro because it selects the Java 1.5 JRE.

````
~/JMRI> /usr/lib/jvm/java/bin/java -version
java version "1.5.0"
gij (GNU libgcj) version 4.8.3 20140627 [gcc-4_8-branch revision 212064]
````

````
~/JMRI> ./PanelPro -d
PWD: '/home/john/JMRI'
CLASSPATH: '.:classes:java/classes:jmri.jar:lib/beansbinding-1.2.1.jar:lib/bluecove-2.1.1-SNAPSHOT.jar:lib/bluecove-bluez-2.1.1-SNAPSHOT.jar:lib/bluecove-gpl-2.1.1-SNAPSHOT.jar:lib/ch.ntb.usb.jar:lib/commons-io-1.3.1.jar:lib/commons-lang3-3.2.1.jar:lib/gluegen-rt.jar:lib/jackson-annotations-2.0.6.jar:lib/jackson-core-2.0.6.jar:lib/jackson-databind-2.0.6.jar:lib/jakarta-regexp-1.5.jar:lib/javacsv.jar:lib/javax.servlet-api-3.1.0.jar:lib/jdom-2.0.5.jar:lib/jdom.jar:lib/jetty-http-9.3.9.v20160517.jar:lib/jetty-io-9.3.9.v20160517.jar:lib/jetty-security-9.3.9.v20160517.jar:lib/jetty-server-9.3.9.v20160517.jar:lib/jetty-servlet-9.3.9.v20160517.jar:lib/jetty-util-9.3.9.v20160517.jar:lib/jhall.jar:lib/jhidrawplugin.jar:lib/jinput.jar:lib/jlfgr-1_0.jar:lib/jmdns.jar:lib/joal.jar:lib/jul-to-slf4j-1.7.13.jar:lib/jython.jar:lib/log4j-1.2.17.jar:lib/mailapi.jar:lib/openlcb.jar:lib/pi4j-core.jar:lib/pi4j-device.jar:lib/pi4j-gpio-extension.jar:lib/pi4j-service.jar:lib/RXTXcomm.jar:lib/Serialio.jar:lib/slf4j-api-1.7.13.jar:lib/slf4j-log4j12-1.7.13.jar:lib/smtp.jar:lib/vecmath-1.5.2.jar:lib/websocket-api-9.3.9.v20160517.jar:lib/websocket-client-9.3.9.v20160517.jar:lib/websocket-common-9.3.9.v20160517.jar:lib/websocket-server-9.3.9.v20160517.jar:lib/websocket-servlet-9.3.9.v20160517.jar:lib/WinRegistry-4.4.jar:lib/xAPlib.jar:lib/xbee-api-0.9.jar:lib/xercesImpl.jar'
Java CMD: '/usr/lib/jvm/java/bin/java'
APPNAME: 'PanelPro'
OPTIONS: ' -noverify -Djava.security.policy=lib/security.policy -Djava.rmi.server.codebase=file:java/classes/ -Djava.library.path=.:lib/linux/i386:lib/linux:lib -Xms96m -Xmx1002m -Dgnu.io.rxtx.NoVersionOutput=true'
ALTPORTS: '-Dgnu.io.rxtx.SerialPorts=/dev/ttyS0:/dev/ttyS1:/dev/ttyS2:/dev/ttyS3:/dev/ttyS4:/dev/ttyS5:/dev/ttyS6:/dev/ttyS7:/dev/ttyUSB0'
Exception in thread "main" java.lang.ClassFormatError: apps.PanelPro.PanelPro (unrecognized class file version)
at java.lang.VMClassLoader.defineClass(libgcj.so.14)
at java.lang.ClassLoader.defineClass(libgcj.so.14)
at java.security.SecureClassLoader.defineClass(libgcj.so.14)
at java.net.URLClassLoader.findClass(libgcj.so.14)
at gnu.gcj.runtime.SystemClassLoader.findClass(libgcj.so.14)
at java.lang.ClassLoader.loadClass(libgcj.so.14)
at java.lang.ClassLoader.loadClass(libgcj.so.14)
at gnu.java.lang.MainThread.run(libgcj.so.14)
Exit Status: 1
````

See the [JMRIusers message thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/129896;_ylc=X3oDMTM4b3BkNHE4BF9TAzk3MzU5NzE0BGdycElkAzY2MDI3NTgEZ3Jwc3BJZAMxNzA1MDYzNTc2BG1zZ0lkAzEyOTkwOARzZWMDZnRyBHNsawN2dHBjBHN0aW1lAzE0Njg0MTc4MDQEdHBjSWQDMTI5ODk2) for more details.

rhwood, 2016-07-13:
We need to be more specific about what "Linux" means in this case. The solution to this problem differs between Fedora/RedHat/CentOS, Debian/Ubuntu, and Suse/OpenSuse, since these operating system families have different mechanisms for dealing with multiple versions of the same application.

bobjacobsen, 2016-07-13:
Thanks. The user mentioned "OpenSUSE 13.2" in a JMRIusers post.

rhwood, 2016-07-13:
In this case, the user's environment has either:

1. a JAVA_HOME environment variable defined that is pointing to ```/usr/lib/jvm/java``` If this is the case, it needs to be removed from the environment.
2. ```/usr/lib/jvm/java/bin``` in his or her PATH. On Suse/OpenSuse the ```update-alternatives --config java``` command will most likely fix this issue.

The new launcher will use JAVA_HOME if it set, while the old launcher would ignore JAVA_HOME and search for java only in the shell's path.

PapaJohn88, 2016-07-13:
The simple solution was to unset JAVA_HOME before running PanelPro. I simply added the unset JAVA_HOME command before the $HOME/JMRI/PanelPro command associated with the desktop icon. Most likely not the best way of getting around the problem but it works.

John

closed 2016-07-15 by bobjacobsen