JMRI/JMRI #1662: 4.5.1 Hangs

fiz1962, 2016-07-15:
I've installed 4.5.1 on 2 Windows systems, Window 7 and Windows 10.  They both hang if loading a profile from 4.4 but can create a new profile.

rhwood, 2016-07-15:
What connections were expected to work in the breaking profile?

We are trying to determine if all profiles have an issue, or only if certain connection types break.

fiz1962, 2016-07-15:
Here's the profile.  Only 1 connection and it is disabled because it is not currently plugged in.

<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.dccpp.serial.configurexml.ConnectionConfigXml" disabled="yes" manufacturer="DCC++" port="COM7" speed="115,200 baud" systemPrefix="DCCPP" userName="DCC++">
            <options/>
        </connection>
    </connections>
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
        <perform xmlns="" class="apps.configurexml.PerformFileModelXml" name="preference:MyJMRIRailRoadPanels.xml" type="XmlFile"/>
    </startup>
</auxiliary-configuration>

fiz1962, 2016-07-15:
The problem seems to be in profile.properties.  Specifically

jmri-managers.defaults.jmri-CommandStation=DCC++

commenting out this line allows PanelPro to load.  The only DCC++ connection is disabled as I don't have the hardware at home to plug in while editing other parts of the system.

bobjacobsen, 2016-07-16:
There's a problem, first visible in C/MRI but apparently also in SPROG and perhaps others, where the SystemConnectionMemo returns a null TrafficController and Bad Stiff Ensues. Don't know if this is related or not.  Are you seeing NPE errors? If so, a traceback would be helpful.

DanielBoudreau, 2016-07-16:
Bob,

 

I tried a simple but incorrect configuration that any user could do using the latest commits.  I set the connect to NCE USB and I didn’t specify a comm port.  I get the following NPE and the program hangs, no way to fix it once it happens:

 

2016-07-15 20:16:58,814 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]

2016-07-15 20:16:58,846 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Daniel Boudreau\JMRI\log\messages.log [main]

2016-07-15 20:16:58,846 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Daniel Boudreau\JMRI\log\session.log [main]

2016-07-15 20:16:58,861 apps.Apps                             INFO  - PanelPro version 4.5.1ish-201607031624-Daniel Boudreau starts under Java 1.8.0_25 on Windows 7 amd64 v6.1 at Fri Jul 15 20:16:58 EDT 2016 [main]

2016-07-15 20:16:59,688 apps.Apps                             INFO  - Starting with profile Brian_Keay.3f5240fd [main]

2016-07-15 20:17:00,219 node.NodeIdentity                     INFO  - Using jmri-0C84DCBAA785-3f5240fd as the JMRI Node identity [main]

2016-07-15 20:17:00,641 usbdriver.UsbDriverAdapter            ERROR - Serial port (none selected) not found [main]

2016-07-15 20:17:00,641 configurexml.ErrorHandler             ERROR - (none selected) not found while opening connection [main]

2016-07-15 20:17:03,823 jmri.InstanceManager                  ERROR - Should not set default of type jmri.ConsistManager to null value [main]

2016-07-15 20:17:03,823 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]

java.lang.NullPointerException

       at jmri.InstanceManager.setDefault(InstanceManager.java:235)

       at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:159)

       at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:214)

       at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:249)

       at jmri.implementation.JmriConfigurationManager.lambda$2(JmriConfigurationManager.java:245)

       at jmri.implementation.JmriConfigurationManager$$Lambda$30/243745864.accept(Unknown Source)

       at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)

       at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)

       at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:244)

       at jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManager.java:178)

       at jmri.implementation.JmriConfigurationManager$$Lambda$29/1253946629.accept(Unknown Source)

       at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)

       at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)

       at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:177)

       at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:166)

       at apps.Apps.<init>(Apps.java:266)

       at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)

       at apps.PanelPro.PanelPro.main(PanelPro.java:98)

fiz1962, 2016-07-16:
Maybe this helps...

2016-07-15 18:21:19,542 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-07-15 18:21:19,557 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\willif\JMRI\log\messages.log [main]
2016-07-15 18:21:19,557 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\willif\JMRI\log\session.log [main]
2016-07-15 18:21:19,573 apps.Apps                             INFO  - PanelPro version 4.5.1-R44ce773 starts under Java 1.8.0_91 on Windows 7 amd64 v6.1 at Fri Jul 15 18:21:19 CDT 2016 [main]
2016-07-15 18:21:29,463 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3edbb29a [main]
2016-07-15 18:21:29,931 node.NodeIdentity                     INFO  - Using jmri-D4BED94B59BF-3edbb29a as the JMRI Node identity [main]
2016-07-15 18:21:30,337 jmri.InstanceManager                  ERROR - Should not set default of type jmri.CommandStation to null value [main]
2016-07-15 18:21:30,337 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.NullPointerException
	at jmri.InstanceManager.setDefault(InstanceManager.java:235)
	at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:159)
	at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:214)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:249)
	at jmri.implementation.JmriConfigurationManager.lambda$2(JmriConfigurationManager.java:245)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:244)
	at jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManager.java:178)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:177)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:166)
	at apps.Apps.<init>(Apps.java:266)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:98)

bobjacobsen, 2016-07-16:
It looks like we have several problems, not quite sure how they separate yet. I'm working on this one in C/MRI(post from jmri-developers)
````
I had my son recreate the (related) problem I’m seeing, then run “git bisect” to track it down.

To recreate:
*) Run PanelPro, configured to run from Digitrax LocoNet Simulator.
*) From a clean start on that configuration:
**) Go to preferences
**) Select CMRI
**) Select Simulator
**) Click Configure Nodes
**) Without changing anything, click “Add Node”

A fault looks like:

    [java] 2016-07-11 21:17:38,419 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
    [java] java.lang.NullPointerException
    [java] 	at jmri.jmrix.cmri.serial.nodeconfig.NodeConfigFrame.addButtonActionPerformed(NodeConfigFrame.java:354)
    [java] 	at jmri.jmrix.cmri.serial.nodeconfig.NodeConfigFrame$3.actionPerformed(NodeConfigFrame.java:283)
    [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
    [java] 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
    [java] 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)

due to a problem with the SystemConnectionMemo.

“git bisect” found that it comes from this commit:

678c19f5dca9f31c85fae2b3d4e9666114f582f7 is the first bad commit
commit 678c19f5dca9f31c85fae2b3d4e9666114f582f7
Author: Paul Bender <paul.bender@acm.org>
Date:   Fri Jun 24 00:29:44 2016 -0400

   remove ActiveFlag references from C/MRI

:040000 040000 ff4c04faee7e3eab8cb8468c4bb83d67c165c58c 2d7cdb5f0168c00e47dd5b23280efc5c0ad2cd59 M	java

Paul, if you have a chance to look at this, that would be great. I haven’t been keeping up on the ActiveFlag changes.  I can probably find time later in the week if you’ve not got a chance.
````

and have that basically fixed, but don't yet know how many other systems are affected.

pabender, 2016-07-17:
Sent from my iPad

On Jul 16, 2016, at 3:25 PM, Bob Jacobsen <notifications@github.com> wrote:
> due to a problem with the SystemConnectionMemo.
>
> “git bisect” found that it comes from this commit:
>
> 678c19f5dca9f31c85fae2b3d4e9666114f582f7 is the first bad commit
> commit [...]

I'll take a look at that tonight...

And I wouldn't be surprised if there are other systems I have touched that have similar problems.

Paul

pabender, 2016-07-17:
On 07/16/2016 08:17 PM, Paul Bender wrote:
> I'll take a look at that tonight...

For the C/MRI case, the issue is that I missed a few places where 
SerialTrafficController.instance() was called... AND I converted 
SerialTrafficController.instance() to return a null object.

Setting a null object in this case did just what I wanted it to, it let 
me find the calls I missed.

Paul

bobjacobsen, 2016-07-17:
> On 07/16/2016 08:17 PM, Paul Bender wrote:
> > I'll take a look at that tonight...
> 
> For the C/MRI case, the issue is that I missed a few places where 
> SerialTrafficController.instance() was called... AND I converted 
> SerialTrafficController.instance() to return a null object.
> 

I found and fixed a couple of those places in the branch "cmri-crash-from-TC” in the bobjacobsen/JMRI  repository, if that’s helpful. Not sure I got them all.

pabender, 2016-07-17:
I just merged in PR #1668 which SHOULD take care of this for C/MRI systems.

I was able to verify the issue is corrected when the simulator is used, but I don't have any real hardware to test with right now.

pabender, 2016-07-24:
Pr #1684 moves the SPROG system(s) to multi-system code.  None of this was committed prior to the 4.5.1 release.

I was not able to duplicate SPROG systems hanging ( testing with a Pi - SPROG and a Pi-SPROG Nano )

One issue i do know about is there are apparently hanging problems when a user upgrades from 4.0 directly to 4.4.  I've seen this happen with two XPressNet users in the last 24 hours.  Both users were able to get JMRI 4.4 to load properly by going back to 4.2, loading the profile  , and then going to 4.4.

rhwood, 2016-07-24:
Do you have the original 4.0 profiles available?

pabender, 2016-07-25:
> Do you have the original 4.0 profiles available?

No, because neither of the users followed my instructions (zip up the profile and send them to me...)

But, I do have a profile that hangs with head of master.  It was last written with 3.11.5 and has both an XBee and a RFID connection configured.

I'm attaching that profile here.

[JMRIProfile.zip](https://github.com/JMRI/JMRI/files/380643/JMRIProfile.zip)

bobjacobsen, 2016-08-15:
I believe this is fixed in JMRI 4.5.2, now being created.  If crashes are observed in that version or later, please open a new Issue and reference this one.  Thanks.

closed 2016-08-15 by bobjacobsen