JMRI/JMRI #1666: Create Test Release 4.5.2

bobjacobsen, 2016-07-16:
For discussion of timing and what to include in 4.5.2

bobjacobsen, 2016-07-21:
(Only occasional contact this week, sorry to be having trouble keeping up)

Do we want to move JMRI 4.5.2 up (sooner) from it's current schedule?  

Normally, it would be built on the weekend of August 13, just over three weeks from now.

JMRI 4.5.1 had some issues which seem to be coming under control.  Do we want to move up the 4.5.2 target to try to get them out earlier?  If so, when should we aim for?  

Would a week from now (weekend of July 30) make sense as a two-week gain in getting those out, or should we just stick to the usual interval?

DanielBoudreau, 2016-07-21:
Bob,

 

Still seeing an NPE when setting up a NCE connection and selecting “none selected” for the comm port.  Once this happens, you cannot change the connection as JMRI has crashed. This needs to be fixed before we release 4.5.2.

 

Dan

 

2016-07-21 18:43:39,710 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]

2016-07-21 18:43:39,730 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Daniel Boudreau\JMRI\log\messages.log [main]

2016-07-21 18:43:39,731 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Daniel Boudreau\JMRI\log\session.log [main]

2016-07-21 18:43:39,739 apps.Apps                             INFO  - PanelPro version 4.5.1ish-201607031624-Daniel Boudreau starts under Java 1.8.0_25 on Windows 7 amd64 v6.1 at Thu Jul 21 18:43:39 EDT 2016 [main]

2016-07-21 18:43:40,187 apps.Apps                             INFO  - Starting with profile Brian_Keay.3f5240fd [main]

2016-07-21 18:43:40,398 node.NodeIdentity                     INFO  - Using jmri-0C84DCBAA785-3f5240fd as the JMRI Node identity [main]

2016-07-21 18:43:40,574 serialdriver.SerialDriverAdapter      ERROR - Serial port (none selected) not found [main]

2016-07-21 18:43:40,575 configurexml.ErrorHandler             ERROR - (none selected) not found while opening connection [main]

2016-07-21 18:43:43,444 jmri.InstanceManager                  ERROR - Should not set default of type jmri.CommandStation to null value [main]

2016-07-21 18:43:43,445 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]

java.lang.NullPointerException

       at jmri.InstanceManager.setDefault(InstanceManager.java:235)

       at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:159)

       at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:214)

       at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:249)

       at jmri.implementation.JmriConfigurationManager.lambda$2(JmriConfigurationManager.java:245)

       at jmri.implementation.JmriConfigurationManager$$Lambda$30/1613255205.accept(Unknown Source)

       at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)

       at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)

       at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:244)

       at jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManager.java:178)

       at jmri.implementation.JmriConfigurationManager$$Lambda$29/352359770.accept(Unknown Source)

       at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)

       at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)

       at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:177)

       at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:166)

       at apps.Apps.<init>(Apps.java:266)

       at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)

       at apps.PanelPro.PanelPro.main(PanelPro.java:98)

bobjacobsen, 2016-08-05:
Normally, we'd build JMRI 4.5.2 this weekend.  What are the critical issues to be addressed before that?

(Sorry to be out of touch, traveling a lot recently, but at least I've got a bunch of time on airplanes later today to do some JMRI programming!)

Among the open Issues, #1662 seems to be still a problem, and should be fixed.  Any others?

Anything that's not on the Issues list?

silverailscolo, 2016-08-05:
PR #1682 should be included in 4.5.2 to fix (part of) the Conditional expression i18n issue present in 4.5.1 and earlier.

Egbert Broerse

> Op 5 aug. 2016, om 17:02 heeft Bob Jacobsen <notifications@github.com> het volgende geschreven:
> 
> Normally, we'd build JMRI 4.5.2 this weekend. What are the critical issues to be addressed before that?
> 
> Anything that's not on the Issues list?

DanielBoudreau, 2016-08-12:
Still seeing NPE when one of the two default connections is selected for any manufacture.  To repeat the following NPE, Select "none selected" when setting up a connection.  This is a really bad bug and it locks up JMRI and you can't fix the problem without going back to one of the previous production releases.  I don't understand why it is taking so long to resolve this issue.  I know Randall has been making changes in this area.

Dan

2016-08-12 09:27:43,248 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-08-12 09:27:43,266 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Daniel Boudreau\JMRI\log\messages.log [main]
2016-08-12 09:27:43,267 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Daniel Boudreau\JMRI\log\session.log [main]
2016-08-12 09:27:43,274 apps.Apps                             INFO  - PanelPro version 4.5.2ish-201608060219-Daniel Boudreau starts under Java 1.8.0_25 on Windows 7 amd64 v6.1 at Fri Aug 12 09:27:43 EDT 2016 [main]
2016-08-12 09:27:43,710 apps.Apps                             INFO  - Starting with profile PanelPro.3ef219d3 [main]
2016-08-12 09:27:43,925 node.NodeIdentity                     INFO  - Using jmri-0C84DCBAA785-3ef219d3 as the JMRI Node identity [main]
2016-08-12 09:27:44,082 serialdriver.SerialDriverAdapter      ERROR - Serial port (none selected) not found [main]
2016-08-12 09:27:44,087 configurexml.ErrorHandler             ERROR - (none selected) not found while opening connection [main]
2016-08-12 09:27:46,527 jmri.InstanceManager                  ERROR - Should not set default of type jmri.PowerManager to null value [main]
2016-08-12 09:27:46,527 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.NullPointerException
	at jmri.InstanceManager.setDefault(InstanceManager.java:235)
	at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:159)
	at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:214)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:249)
	at jmri.implementation.JmriConfigurationManager.lambda$2(JmriConfigurationManager.java:245)
	at jmri.implementation.JmriConfigurationManager$$Lambda$31/1613255205.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:244)
	at jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManager.java:178)
	at jmri.implementation.JmriConfigurationManager$$Lambda$30/243745864.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:177)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:166)
	at apps.Apps.<init>(Apps.java:266)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:98)

bobjacobsen, 2016-08-12:
> On Aug 12, 2016, at 6:34 AM, Daniel Boudreau <notifications@github.com> wrote:
> 
> 2016-08-12 09:27:43,925 node.NodeIdentity INFO - Using jmri-0C84DCBAA785-3ef219d3 as the JMRI Node identity [main]
> 2016-08-12 09:27:44,082 serialdriver.SerialDriverAdapter ERROR - Serial port (none selected) not found [main]
> 2016-08-12 09:27:44,087 configurexml.ErrorHandler ERROR - (none selected) not found while opening connection [main]
> 2016-08-12 09:27:46,527 jmri.InstanceManager ERROR - Should not set default of type jmri.PowerManager to null value [main]
> 2016-08-12 09:27:46,527 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
> java.lang.NullPointerException
> at jmri.InstanceManager.setDefault(InstanceManager.java:235)
> at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:159)
> at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:214)

I’m not able to recreate this.  Perhaps it depends on a race condition, or I’ve got some default persisting somewhere, but I haven’t been able to recreate it.

Could you rerun with DEBUG set for:

	jmri.InstanceManager
	jmri.managers.ManagerDefaultSelector
	jmri.implementation.JmriConfigurationManager
        jmri.jmrix.SystemConnectionMemo

please? 

The problem seems to be that jmri.managers.ManagerDefaultSelector is working with a busted SystemConnectionMemo.  But at this point, I don’t even know what concrete type that is, etc.

Bob
--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

rhwood, 2016-08-12:
> at apps.Apps.(Apps.java:266)
> at apps.PanelPro.PanelPro.(PanelPro.java:41)

I haven't been making changes in this area for months, but if you see a problem, you are free to fix it.

That said, I see how to protect the system against this, but I do not know why the NCE system connection memo would be returning null for any of the managers that can be configured as defaults.

DanielBoudreau, 2016-08-12:
Bob,

 

Not a race condition.  Set up a connection where the serial port is one of the two defaults:

 

jacobsen@mac.com <mailto:jacobsen@mac.com>  +1-510-708-5988 AIM, Skype JacobsenRG

DanielBoudreau, 2016-08-12:
Randall,

 

It’s been broken since we created the first test releases.  4.5.1 and 4.5.2 have this problem, 4.4 does not.

 

Dan

pabender, 2016-08-12:
Sent from my iPad

On Aug 12, 2016, at 10:26 AM, Bob Jacobsen <notifications@github.com> wrote:
> I’m not able to recreate this. Perhaps it depends on a race condition, or I’ve got some default persisting somewhere, but I haven’t been able to recreate it.
<snip>
> The problem seems to be that jmri.managers.ManagerDefaultSelector is working with a busted SystemConnectionMemo. But at this point, I don’t even know what concrete type that is, etc.

It's possible this is related to removing the references to the active flag in the NCE code.  

That said, I can't replicate this either.

Paul

DanielBoudreau, 2016-08-13:
Randall's latest commit has fixed the NPE, and we now get a nice pop-up message stating that there's a configuration issue and the preferences window opens..  Thank-you.  Good to go for 4.5.2.

Dan

bobjacobsen, 2016-08-14:
The release-4.5.2 branch has been created. 

From now on, please document your changes in the [jmri4.5.2.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.5.2.shtml) release note file.

Maintainers, please set the 4.5.2 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [new server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.2/)

dheap, 2016-08-15:
Maybe 4.5.3?
> @bobjacobsen From now on, please document your changes in the jmri4.5.2.shtml release note file.
> 
> Maintainers, please set the 4.5.2 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

bobjacobsen, 2016-08-15:
> Maybe 4.5.3?

Oops. Thanks for catching that. Fixed!

bobjacobsen, 2016-08-15:
Released and announced.

closed 2016-08-15 by bobjacobsen