JMRI/JMRI #1667: A little help on a new System & Preferences test

bobjacobsen, 2016-07-16:
In the short term, we have multiple issues with broken startup, probably due to a combination of insufficient test coverage for the (hardware) system setup code, combined with trying to move forward some complex legacy code. (In other words, another demonstration of how code gets brittle without tests). We'll just have to work through that for 4.5.2.

In the longer term, I'd like to create some release tests that can check this by checking the configuration and startup process, up to and including specifying real ports.  We may not (or may) be able to run them for CI, but as a minimum making it possible for developers to run them on their own machines would help.

Happy to take another approach, but I'm imagining JFCUnit tests that 
- Start with a clean profile directory, e.g. not the current user's local setup, but something specific such as "internal system only"
- Walks through GUI selection to configure the specific system, e.g. selects C/MRI
- Configures a specific connection, e.g. C/MRI Serial, using a port defined in an environment variable
- Saves and exits
- Does a check of the stored preferences
- Does a start and checks that goes OK
- All the while, detecting exceptions, etc.

I'm happy to do a lot of the work on this. Clearly, it's going to take some time to get good coverage, but I think it's worth a start. I've created the basic scaffold and pushed it to my fork:

[java/test/jmri/jmrix/AbstractSystemStartupTest.java](https://github.com/bobjacobsen/JMRI/blob/better-startup-unit-tests/java/test/jmri/jmrix/AbstractSystemStartupTest.java)

I'd very much appreciate help with this, especially that first bullet.  
- [] The code linked above does open a pseudo-working Preferences window, but it's pretty ad-hoc. (See the setUp() method)  Is there a better way to know what needs to be initialized? To do the initialization?
- [] What's the right approach to having reads and writes go to a temp directory and be able to reset that directory between tests?
- [] Does this need to be connected to the profile code too?
- [] It only works once (see suppressed 2nd test is file linked above) Any ideas on why? Is that a setup issue? 

@rhwood, you've done a lot of work in that area, so any help would be appreciated, as would help from anybody else. Feel free to either modify and PR the code itself, or tell me what's needed (in detail, please)

Thanks in advance.

(Eventually, I'd like to refactor the code of this up as a basic preference window test, and down to test specific system setup, but for now I just want to get something cleanly running)

rhwood, 2016-07-17:
Given a Profile object, a jmri.implementation.JmriConfigurationManager object will initialize an instance of every class listed in java/src//META-INF/services/jmri.spi.PreferencesManager when constructed (the JmriConfigurationManager itself is initialized in ```JUnitUtil.initConfigureManager()```. An empty or arbitrary Profile object can be created by instantiating a NullProfile object, or an existing Profile in SCM for testing can be created using the jmri.profile.Profile(path/to/profile/dir) constructor (Call ProfileManager.getDefault().setActiveProfile(Profile) before creating the JmriConfigureManager).

This might work in tests (I haven't tested this):

```
Profile p = new Profile(new File(path/to/test/profile));
ProfileManager.getDefault().setActiveProfile(p);
JUnitUtil.InitConfigureManager();
```

Regarding the bullets:
- Start with a clean profile directory, e.g. not the current user's local setup, but something specific such as "internal system only"
  - All tests run by ant do this already, but tests run by their ```main()``` methods do not. This is why #1584 proposed changing runtest.csh the way it did.
- Walks through GUI selection to configure the specific system, e.g. selects C/MRI
Configures a specific connection, e.g. C/MRI Serial, using a port defined in an environment variable
  - Instead of walking through the GUI, set the preferences items programmatically - this allows us to catch problems *not* in the GUI at the correct level.
- Saves and exits
- Does a check of the stored preferences
  - You can save and test that stored preferences are correct without exiting - we are no longer dependent on exiting to store a complete set of preferences like we used to be.
- Does a start and checks that goes OK
  - You can also start from stored preferences, which might help break these tests into smaller pieces.
- All the while, detecting exceptions, etc.

bobjacobsen, 2020-07-25:
Travis is now running some startup tests, in large part through the work of @silverailscolo and @rhwood, so I'm going to close this.

closed 2020-07-25 by bobjacobsen