JMRI/JMRI #1670: Instructions for converting to native JUnit4 format?

bobjacobsen, 2016-07-17:
@rhwood moved the JMRI JUnit tests to JUnit4 in #1376, which opened up some great new possibilities.

JUnit4 has a different approach to test class structure, which you can see by comparing:
- the new form [jmri/profile/ProfileTest.java](https://github.com/JMRI/JMRI/blob/master/java/test/jmri/profile/ProfileTest.java) which uses annotations like `@Test` and `@Before`
- the older form [jmri/IdTagTest.java](https://github.com/JMRI/JMRI/blob/master/java/test/jmri/IdTagTest.java) which has specific method names.

Should we add some hints and tips about this to our [JUnit recommendations page](http://jmri.org/help/en/html/doc/Technical/JUnit.shtml#write4NewClass)?

Are there new capabilities we want to call out?

rhwood, 2016-07-17:
We might want to consider:
- using [Jemmy](https://jemmy.java.net) for UI tests instead of JFCUnit, since Jemmy can be used by Junit 4 tests, while JFCUnit requires extending Junit 3 TestCase class.
- [Ignoring Tests](https://github.com/junit-team/junit4/wiki/Ignoring-tests) - marking a test as not to be run because the test itself is broken
- [Timeouts for Tests](https://github.com/junit-team/junit4/wiki/Timeout-for-tests) - failing a test because it takes too long
- [Categories](https://github.com/junit-team/junit4/wiki/Categories) - useful ones in our case could be headless/not headless, hardware specific (loco buffer attached, NCE PowerPro attached, etc)

pabender, 2016-07-18:
I've actually already been using the ignoring tests.  

There are also assumptions ( http://junit.sourceforge.net/javadoc/org/junit/Assume.html ) to conditionally ignore a test.

Incidentally, I've also written a few tests using JUnit4 that have a main see, for example:
https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/cmri/PackageTest.java

rhwood, 2016-07-22:
We may want to document that a test that would fail in a headless environment can be ignored in headless mode if the first line of the test method is:
```
Assume.assumeTrue(!GraphicsEnvironment.isHeadless());
```

bobjacobsen, 2016-08-08:
The comments here are now on our [JUnit page](http://jmri.sourceforge.net/help/en/html/doc/Technical/JUnit.shtml#junit4) along with some pointers to examples.  It's a start....

bobjacobsen, 2016-08-22:
A bit more info added via #1840

closed 2016-08-22 by bobjacobsen