JMRI/JMRI #1674: Slow close of Tables window

bobjacobsen, 2016-07-17:
A JMRIusers report of a freeze of up to 22 seconds(!) when closing the Tables window prompted me to look into this.  Although his experience is worse than most, I think there's an issue here.

On my Mac, lots of memory and SSD file system, the GUI is frozen for about 3 seconds when closing the tables window.  It doesn't seem to matter which pane is open, or what else is happening.

I've attached a typical thread dump, and a trace-level log of what happens, along with the UserPrefsProfileConfig.xml file that ends up getting written:
[Typical stack trace](https://github.com/JMRI/JMRI/files/367976/TypicalTraceback.txt)
[log output at trace level](https://github.com/JMRI/JMRI/files/367977/trace-log.txt)
[Written UserPrefsProfileConfig.xml as .txt](https://github.com/JMRI/JMRI/files/367978/UserPrefsProfileConfig.xml.txt)

The `JmriUserPreferencesManager.saveTableColumnPreferences()` is apparently supposed to format and write a single `tableDetails` element by calling `saveElement()`, which invokes `jmri.util.xml.XMLUtil.write(..)`. That element contains the info to recreate the format of all the tables.  

The trace shows that `saveElement` is invoked for `tableDetails` 161 (!) times, even though that element ends up in the file only once.  Each call on my machine takes on average about 12msec, so those calls are the majority of the total time. (The processing is all on the AWT-EventQueue, so that explains the freeze).  Doing that much I/O on an older machine with a rotating disk might take much longer.

The code seems to intend to write much less.  `saveTableColumnPreferences()` is one of five things sequentially invoked by `savePreferences()`:  That might mean five writes of the file, still more than optimal, but not so bad. 

You can see signs of that in the trace:  `Saving windowDetails element`, which is the last of the five to be called, appears (twice, though) at the end.  But note that `Saving settings element`, which is invoked in `savePreferences()` _before_ `saveTableColumnPreferences()`, appears almost at the end of the trace. There are 16 reports of `Saving tableDetails element.` between those, still too many, but only a tenth of the total.  There seems to be multiple things going on here.

A clue is in the thread dump:
```
     [java] 	at jmri.util.xml.XMLUtil.write(XMLUtil.java:427)
     [java] 	at jmri.util.prefs.JmriConfiguration.putConfigurationFragment(JmriConfiguration.java:112)
     [java] 	- locked <0x0000000798609d10> (a jmri.util.prefs.JmriUserInterfaceConfigurationProvider$Configuration)
     [java] 	at jmri.managers.JmriUserPreferencesManager.saveElement(JmriUserPreferencesManager.java:1337)
     [java] 	at jmri.managers.JmriUserPreferencesManager.saveTableColumnPreferences(JmriUserPreferencesManager.java:1224)
     [java] 	at jmri.managers.JmriUserPreferencesManager.setTableColumnPreferences(JmriUserPreferencesManager.java:887)
     [java] 	at jmri.jmrit.beantable.BeanTableDataModel.saveTableColumnDetails(BeanTableDataModel.java:970)
     [java] 	at jmri.jmrit.beantable.BeanTableDataModel.saveTableColumnDetails(BeanTableDataModel.java:954)
     [java] 	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.dispose(ListedTableFrame.java:450)
     [java] 	at jmri.jmrit.beantable.ListedTableFrame.dispose(ListedTableFrame.java:206)
     [java] 	at javax.swing.JFrame.processWindowEvent(JFrame.java:305)
```
Here, `saveTableColumnPreferences(..)`, hence `saveElement`, is being invoked out of `setTableColumnPreferences(..)`, which in turn was invoked out of the closing of the BeanTable by the closing of the window. Each table in the window will do something like:

```
    public void saveTableColumnDetails(JTable table, String beantableref) {
        jmri.UserPreferencesManager p = jmri.InstanceManager.getDefault(jmri.UserPreferencesManager.class);
        XTableColumnModel tcm = (XTableColumnModel) table.getColumnModel();
        TableSorter tmodel = ((TableSorter) table.getModel());
        Enumeration<TableColumn> en = tcm.getColumns(false);
        while (en.hasMoreElements()) {
            TableColumn tc = en.nextElement();

            try {
                String columnName = (String) tc.getHeaderValue();
                //skip empty or blank columns
                if (columnName != null && !columnName.equals("")) {
                    int index = tcm.getColumnIndex(tc.getIdentifier(), false);
                    p.setTableColumnPreferences(beantableref, columnName, index, tc.getPreferredWidth(), TableSorter.getSortOrder(tmodel.getSortingStatus(tc.getModelIndex())), !tcm.isColumnVisible(tc));
                }
            } catch (Exception e) {
                log.warn("unable to store settings for table column " + tc.getHeaderValue());
                e.printStackTrace();
            }
        }
    }
```

For each table, for each column, we're calling `jmri.UserPreferencesManager.setTableColumnPreferences(..)`, which in turn is writing the file immediately.  Then, just for good effect, we write it _again_ at the end.

How is this stuff supposed to be used?

rhwood, 2016-07-17:
> I've attached a typical thread dump, and a trace-level log of what happens, along with the UserPrefsProfileConfig.xml file that ends up getting written:
> Typical stack trace <https://github.com/JMRI/JMRI/files/367976/TypicalTraceback.txt>
> log output at trace level <https://github.com/JMRI/JMRI/files/367977/trace-log.txt>
> Written UserPrefsProfileConfig.xml as .txt <https://github.com/JMRI/JMRI/files/367978/UserPrefsProfileConfig.xml.txt>UserPrefsProfileConfig.xml file *should not* be written. The file <profile>/profile/<node>/user-interface.xml is what *should* be written.
> The JmriUserPreferencesManager.saveTableColumnPreferences() is apparently supposed to format and write a single tableDetails element by calling saveElement(), which invokes jmri.util.xml.XMLUtil.write(..). That element contains the info to recreate the format of all the tables.
> 
> The trace shows that saveElement is invoked for tableDetails 161 (!) times, even though that element ends up in the file only once. Each call on my machine takes on average about 12msec, so those calls are the majority of the total time. (The processing is all on the AWT-EventQueue, so that explains the freeze). Doing that much I/O on an older machine with a rotating disk might take much longer.
> 
It should be invoked as a user preferences change happens. The point of doing it on demand is that the previous method of doing this would only be invoked once per application instance, often prior to windows having finalized those details, so that, for example, the preferences of the main window of DecoderPro were *never* saved on OS X (because the window would be closed when exiting the application only after all shutdown hooks had been run, so its state was irrelevant). See #1126 and #1110.
> The code seems to intend to write much less. saveTableColumnPreferences() is one of five things sequentially invoked by savePreferences(): That might mean five writes of the file, still more than optimal, but not so bad.
> 
It should be writing to the file on every change. There are methods to prevent the save within the PreferencesManager. They need to be set when closing the table window (see last paragraph). The code that actually does the saving implements the NetBeans Platform’s mechanism for saving UI state, but does not include the write-reduction and scheduling code in their implementation, although its apparent it should include some optimizing code.
> You can see signs of that in the trace: Saving windowDetails element, which is the last of the five to be called, appears (twice, though) at the end. But note that Saving settings element, which is [...]
This should all be inside calls to jmri.UserPreferencesManager.disallowSave() and jmri.UserPreferencesManager.allowSave() to prevent the massive saving when the window is saved. In most windows (i.e. a Roster window), this could be ignored, but its clear this Window needs to explicitly block the saving until a single save call can be made. You can probably get improvements by replacing jmri.jmrit.beantable.ListedTableFrame.dispose() with:

```
    public void dispose() {
        pref.disallowSave();
        for (int x = 0; x < tabbedTableArray.size(); x++) {
            tabbedTableArray.get(x).dispose();
        }
        if (list.getListSelectionListeners().length > 0) {
            list.removeListSelectionListener(list.getListSelectionListeners()[0]);
        }
        pref.allowSave();
        super.dispose();
    }
```

bobjacobsen, 2016-07-18:
Thanks @rhwood!  Will look into whether the changes are being saved when they're made.  If so, I think we can skip the store cycle within dispose.

In the meantime, am trying the `disallowSave()` then 'allowSave()` you suggest above.  Since all the writing is in the contained loop, the preferences don't get written by this.  Is there a call that says "force a write"?

rhwood, 2016-07-18:
> Thanks @rhwood <https://github.com/rhwood>! Will look into whether the changes are being saved when they're made. If so, I think we can skip the store cycle within dispose.
>
> In the meantime, am [...]
Not within the jmri.UserPreferencesManager interface. If we want to preserve the interface’s API, our best bet would be to flush changes when allowSave() is called. Adding something like:

```
if (this.dirty) {
   this.savePreferences();
}
```

within the allowSave() method in jmri.managers.JmriUserPreferencesManager should do the trick.

rhwood, 2016-07-18:
Reading over this again, I need to clarify something:

Changes to UI state are saved when the UserPreferencesManager is aware of them, not when they occur (ideally a table or window would save its state to the UserPreferencesManager when changes are complete). So you can only stop saving state in the dispose method of a window or table if the window's or table's state is updated in the UserPreferencesManager by a listener to the window or table as the changes occur.

rhwood, 2016-07-22:
Calling ```allowSave()``` in master now automatically triggers a save if there is anything to save.

rhwood, 2016-07-23:
#1694 contains a quick fix for this issue within the Tables window specifically.

A more general and longer term fix still needs to be figured out.

bobjacobsen, 2016-08-15:
This should be fixed in JMRI 4.3.2 due to the work of @rhwood (#1694 and see above).  I'm closing this. If it recurs, please open a new Issue referencing this one.

closed 2016-08-15 by bobjacobsen