JMRI/JMRI #1675: Loosing AppleScript support

rhwood, 2016-07-17:
Oracle is removing AppleScript support in JDK 9 without replacement per [JEP 272](http://openjdk.java.net/jeps/272).

Since we added AppleScript support only in the past year, should we remove it entirely and immediately, or should we provide a separate JAR for JDK 8 users who want it?

bobjacobsen, 2016-07-18:
Java 8 should be supported through [at least a year after Java 9 is released](http://www.oracle.com/technetwork/java/eol-135779.html).  

I've seen various things about the Java 9 release date, but the [best estimate](http://openjdk.java.net/projects/jdk9/) seems to be late March 2017, about nine months from now.

That's a big enough window that I think it's OK to leave the support in for now. Maybe there will be enough interest for some other solution to emerge.

In the meantime, we should probably add a warning about this to the JMRI [AppleScript page](http://jmri.org/help/en/html/tools/scripting/AppleScript.shtml).

rhwood, 2016-07-18:
Unfortunately, half the links on that page are now seriously dated (Java/Cocoa support was removed as of Snow Leopard) or will be completely obsolete with JDK 9. I'm thinking that other than showing how ```osascript``` can be used within a script, we need to remove any other claims of support.

rhwood, 2016-07-18:
Accidentally closed and commented when I meant to not comment at all.

rhwood, 2016-07-19:
#1680 removes direct support for AppleScript. A separately downloadable JAR and documentation for using the mechanisms in #1472 will be published soon.

rhwood, 2016-07-21:
Draft enablement of direct AppleScript scripting is at https://rhwood.github.io/applescript

rhwood, 2016-10-23:
Direct AppleScript scripting enablement is available and documented at [JMRI AppleScript Support](http://jmri.github.io/applescript).

closed 2016-10-23 by rhwood