JMRI/JMRI #1678: Turnout Control tool gets wrong default layout connection

bobjacobsen, 2016-07-18:
Entering a number, e.g. "11", in the Turnout Control (TC) tool should work with the default layout connection for turnouts.  For example, if the default connection is to a LocoNet, then "11" and "LT11" should work the same.

Currently, this is fine if you open the TC from the menu bar.  If you configure it to open via as a startup action in Preferences, it comes up connected to the Internal system, e.g. "11" and "IM11" are equivalent.  "LT11" will still work, but it's not what people expect.

rhwood, 2016-07-18:
Is this not a symptom of #1014?

bobjacobsen, 2016-08-12:
I believe this is now fixed.  At least, I haven't been able to recreate my test case.  Will leave open pending confirmation.

Still working on the more general case in #1014.

bobjacobsen, 2016-10-06:
The general case (#1014) isn't complete yet, but this particular case is done.

closed 2016-10-06 by bobjacobsen