JMRI/JMRI #1697: I18N of JUnit tests

bobjacobsen, 2016-07-25:
In #1682, @silverailscolo reported:

> Local travis-ci test result seems to have nothing to do with my commits: 
Testcase: testSVProgrammingProtocolV2(jmri.jmrix.loconet.locomon.LlnmonTest): FAILED SV2 test 2 expected:<... number 4, product 1[,541, serial number 2,]055 &gt; but was:<... number 4, product 1[.541, serial number 2.]055 [...] at jmri.jmrix.loconet.locomon.LlnmonTest.testSVProgrammingProtocolV2(LlnmonTest.java:64)
What seems strange/a possible cause is use of the Comma as a thousands separator in numerals 1541 and 2055 in the test code, line 64, interfering with the Comma as an item delimiter:
Device characteristics are manufacturer 3, developer number 4, product 1,541, serial number 2,055
On my system the period . is set as thousands separator (Can't succesfully run the test under a non-english locale either).

I haven't recreated this yet, but it sounds like the Llnmon tests are checking internationalized outputs against non-internationalized constants.  

Should the JUnit tests, as a general practice, be able run successfully with the Locale set to something other than the default?

This isn't a trivial question to answer.  A separate translation of the comparison strings for the java/test classes is too much work to be realistic.  But some of the test value is lost if we just check output against the same I18N strings that we're using to generate the message in the first place.  Having a constant String value in an Assert statement provides a bit of a sanity check, and has caught errors before in the Logix code, as at least one sample case.

rhwood, 2016-07-26:
```./runtest.csh -Duser.country=DE -Duser.language=de apps.tests.AllTest``` has these results:
```Tests run: 6162,  Failures: 129,  Errors: 37``` and about a dozen UI tests did not complete (presumably because the test assumed a certain button would have a certain text and could not find it to click).

bobjacobsen, 2020-07-25:
AFAIK, this has been resolved with a documentation update and some locale-setting in the test infrastructure.

closed 2020-07-25 by bobjacobsen