JMRI/JMRI #1706: Connection Modes

rhwood, 2016-07-28:
This is a proposal for an enhancement. Feedback is desired.

# Connection Modes
Within a single profile, on a single computer, there should be multiple sets of connections (or connection modes within a single profile). The primary use case is that a single profile on a single computer can be used connected (real USB connection) or disconnected (simulated connection). A secondary use case is that a single computer may be connected to different hardware at different times (a laptop with a single profile - a single connections with panels and scripts started in a specific order - may be connected to a PR3 at the club and to a LocoBuffer USB at home).

## User Interface
If there are multiple connection modes (called Modes in the System Connection Preferences and Connections Mode(s) elsewhere), JMRI will have a 10 second countdown to select a connection mode (identically to the countdown on Profiles) after JMRI attempts to confirm which one is correct.
The Preferences UI will be above the connections tabs as a combobox (the list of current modes, and a selection “Manage Modes” that opens a “Manage Connection Modes” dialog allowing modes to be added, renamed, or deleted).
Adding a new mode will duplicate the current mode.
Modes would be per-computer within a single profile, such that if profile A is shared between computers B and C, computer B could have modes D, E, and F, while computer C only has mode G.

## Storage
-  ```/profile/profile.xml``` will contain the list of connections for the profile (critically this will contain the user and system names of the connections and all connection sets will mirror this)
- ```/profile/jmri-MACADDR-########/profile.xml``` will contain the connection configuration for the computer with an interface with the MAC address MACADDR (this is the *only* way a JVM exposes a unique identifier for a computer that is tied to physical hardware)
- The ```connections``` element will be extended to allow a child element ```mode``` with an attribute ```name```, which would be unique within a set of mode elements. The ```mode``` element can contain multiple ```connection``` elements. Once a user selects to begin using modes, this is how the per-computer profile.xml files will store modes.

bobjacobsen, 2016-07-29:
The combination of this and multiple profiles seems confusing to me. 

Is there any intent for how those should be used together or separately?

rhwood, 2016-07-29:
> The combination of this and multiple profiles seems confusing to me.
> 
Sorry.
> Is there any intent for how those should be used together or separately?
> 
Connection Modes are contained entirely within a Profile.

A profile is not just connections — it is connections + startup actions + file locations + server (web, withrottle, etc) configs + …

A user, John, has a laptop with three layouts he’s maintaining panels for: his own (HOME), the club’s (CLUB), and a friend he’s helping (FRND), and he sometimes works on these while the laptop is disconnected using simulated (fake) connections, he has six profiles:
HOME-Locobuffer
HOME-simulator
CLUB-PR3+DCS100
CLUB-simulator
FRND-PowerCab
FRND-simulator
So John has six profiles—that’s easy enough, except that HOME and CLUB are both extensively automated and load multiple scripts and panels, so John has to take notes to ensure that the real/fake profile pairs for the HOME layout are reasonably the same (loading the same panels, running the same versions of startup scripts, etc), and has to do the same for the profile pairs for the CLUB layout.

Connection modes will help John by giving him two modes (real and fake) for each profile (HOME, CLUB, and FRND), such that when he starts PanelPro, after he selects the profile (HOME, CLUB, or FRND), he then selects the mode (real or fake).

If John’s club, which uses a PR3 connected to a DCS100, begins testing a DCS240 instead, he adds a mode “DCS240" to his CLUB profile, so that the CLUB profile can be used with both types of hardware (he has three modes on the CLUB profile now — simulator, DCS240, and PR3+DCS100). Without connection modes, this would require a third profile for the same layout.

I hope this helps clarify the intent of Connection Modes.

dheap, 2016-07-29:
Are connection modes like sub-profiles? Are the following covered?
- Multiple simultaneously active connections, e.g. an NCE SB5 (no program track) with SPROG (for programming) or an EasyDCC system with LocoNet (for detection and signalling)? Both connections have to be active in the one mode. 
- Defaults with multiple simultaneously active connections. The defaults will differ by connection mode.
- Handling system or mode specific startup actions, script, buttons. They may differ by mode.
- Common startup actions, script, buttons.

Dave

KenC57, 2016-07-29:
Randall,

 

Given the effort it has taken to get the ideas behind a profile into the
hands of the users, I'm hesitant to say this idea would lead to more
confusion. My own thought however would be leveraging the profile concept
and instead use a wizard approach to building these extra profiles to
reflect the other connection setups. This would help since it would use the
existing profile concept which users are finally getting the hang of using.

 

On the other hand I understand that knowing a set of profiles are related
and that the connection details (real vs simulated) are the true
differences. But one thing I've done frequently with simulated connections
is adding scripts to initialize those turnouts and sensors. So for me an
offline profile would be different by having the simulated connection but
also a few more scripts.

 

Now would your idea understand that when a panel file is updated and changed
in the startup of the one, would it show as changed in the other
connection?? And how would it deal with things like a need for an offline
script for initialization issues?? I'm not sure the added complexity vs
improving how profiles can be cloned is worth the effort given some of the
impact and how to train the users on the proper use of the feature.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-07-29:
> Connection modes will help John by giving him two modes (real and fake) for each profile (HOME, CLUB, and FRND), such that when he starts PanelPro, after he selects the profile (HOME, CLUB, or FRND), he then selects the mode (real or fake).

Thanks, that helps a lot.  I can see where this would simplify profiles for some people quite a bit.  A few more questions:

1. If the user _doesn't_ define multiple connections, will there be a startup choice involved? In that case, it could just skip the choice dialog. Or will there be a bit of configuration to say "open the Connections dialog anyway with a New... button"?  I'd argue for the 1st one, as we've seen in the profile discussion that the 2nd generates some confusion.

2. (This is the thing that was originally confusing me)  Can we discuss a bit about what's included vs not included in a "Connection"?  This is the GUI from the current code for a particular set of settings [image] (The colored circles where added)  

  a. It sounds to me like the variable Connections correspond to different "System connection" values within the "System manufacturer" selection. In other words, I can pick "LocoNet Simulator" vs "LocoNet PR3" by picking a Connection at start up. This basically works because the _rest_ of the startup is expecting a LocoNet link, but script files (to first order) will be happy with that being via a PR3 vs LocoBuffer-USB vs a simulator link.

  b. That selection then sets the specific values for "Serial port", "Command station type", "Baud rate", "Turnout command handling" and "Connection uses (flow control)".  Different Connections will have different values to set, but each will remember their own.

  c. But does that include the settings (in this example) for "Connection Prefix", "Connection Name", and "Disable this Connection"?  If those change within a profile, really confusing things can happen. Perhaps they should be held at the "Link" (word chosen to separate it from connection) level.

  d.  Within that same block, some types of links  have even more specialized configuration, e.g. the C/MRI node definitions.  There are use cases for "I only want to set that once, at the Link level, and will use Profiles to change it", but there are just as many use cases for "I want to have different ones for different Connections".  I have profiles configured to implement both of those; I certainly agree that it can get confusing and frustrating! (Not a criticism of Profiles, still _way_ more helpful than not having them, they just allow me to do things that are complicated to maintain from one layout visit to another) 

It's points 2.c and 2.d that might make this really confusing to people. Or not.  On the one hand, people don't generally mess with those settings, and they'd maybe never even notice. On the other hand, if they do get in trouble, it can be messy getting them out short of "delete the profile and start over".

Within the code, "Connection" is carrying a particular set of information, which forms an operational definition.  Things like C/MRI node definitions are outside that, as is the "disable this connection". A first step, perhaps sufficient, is to update the GUI to make it really clear what's inside and what's outside.  A colored box around the Connection, with another tab line like this to select one to edit and create a new one?  [image] That would address a lot of my concerns. Even better would be moving the "Connection Prefix" and "Connection Name" outside that box, into the Link-level configuration.

But these comments are all along the lines of suggestions. I think the "code wins" principle applies strongly here.  Since, as I now understand, users who don't use this feature don't have to deal with it, i.e. there's no dialog at startup time if there's only one Connection defined, then it's a pure addition and it should be done the way that the person doing the work (i.e. @rhwood) thinks best.

rhwood, 2016-07-29:
> If the user doesn't define multiple connections, will there be a startup choice involved?

I'm assuming you meant "multiple connection modes". If there are not multiple connection modes already defined for the loaded profile at application startup, the user will see no prompting to select a connection mode.

bobjacobsen, 2016-07-29:
>>If the user doesn't define multiple connections, will there be a startup choice involved?

>I'm assuming you meant "multiple connection modes". If there are not multiple connection modes already defined for the loaded profile at application startup, the user will see no prompting to select a connection mode.

Thanks, that clarifies a piece of terminology that I'd missed:  These are called "Connection Modes", not "Connections". 

So is the hierarchy & nomenclature:
- One or more "Profile" choices
- each of which contains pre-set values for (zero, but usually 1) or more "Connection" definitions
- each of which contains one or more selectable "Connection Mode" settings?

That's consistent with the "Connections" in the left-side Preferences selector, but it seems to be a recipe for confusion.  Can I suggest "Connection Specifics" or "Connection Details" or something along those lines?  "Connection Mode" seems to me to be a bit ambiguous about what it contains.

rhwood, 2016-07-29:
> (This is the thing that was originally confusing me) Can we discuss a bit about what's included vs not included in a "Connection"?

Yes. I was thinking that the parts of a "Connection" that are common to modes and enforced, will be the System Prefix and System Name. (i.e. if only connections L1 and L2 exist in the profile, every mode must have connections L1 and L2. Adding connection L3 in one mode will cause every other mode to request that connection L3 be configured when next used. Later removing connection L1 will cause L1 to be removed from every mode). That would definitely be the easiest level of enforcement of similarity.

If you think there should a stronger enforcement of similarity (for example, if L1 is LocoNet in one mode, L1 must be LocoNet in all modes), it would be very helpful to hash that out sooner rather than later.

If its appropriate for a specific connection type to make non-standard information common across modes (like C/MRI nodes), the connection classes will have to ensure that's handled correctly. The API design to handle that will require some thought.

KenC57, 2016-07-29:
I agree that one family would have to have identical connection id's and
requiring the same connection type fits. The connection type will have range
limits etc and those should remain constant between the modes.

 

I've long thought that some is needed for systems like CMRI to keep that set
of node config data out of the connection. I've had to reenter that way to
many times when making a connection change (like real and simulated). Same
on cloning a profile, it doesn't retain that when you have to edit the
connection config. At least that how it seemed last time I had to do that.
But I also know the nodes are important for knowing what addresses are out
there and which to poll, etc. so that data is required for a connection
before it can start polling the nodes.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-07-30:
> (i.e. if only connections L1 and L2 exist in the profile, every mode must have connections L1 and L2. Adding connection L3 in one mode will cause every other mode to request that connection L3 be configured when next used. Later removing connection L1 will cause L1 to be removed from every mode

Back to being confused again, unfortunately.

I don’t understand the singular and plurals in those examples.

Is there one “Connection Mode” setting that cuts across the entire set of Connections?  Or is a Connection Mode a subset of the information in a Connection.

Concrete question:

You have a railroad with simultaneous SPROG and NCE links.  You want to use a single profile for this.  You want to use Connection Mode capability so you can either use the serial ports when at the railroad, or simulators when away from it. You start PanelPro, the profile starts. Does PanelPro ask once for _the_ Connection Mode, which selections both?  Or does it ask you to select a Connection Mode for SPROG, then ask again for NCE?

KenC57, 2016-07-30:
Bob,

 

My understanding of the idea would let you have a number of 'modes' for your
profile that had SPROG and NCE. One mode could be hardware for both, another
mode for both simulated, another for simulated NCE but real SPROG, and last
could be real NCE but simulated SPROG. If you only had one mode, you
wouldn't see anything extra. If you then wanted to, you could create as many
of these different modes as you felt like.

 

So a profile would be some mix of one or more system types, but the modes
would be mixes of the system connections. 

 

Randall, does that sound like what you are planning??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2016-07-30:
> My understanding of the idea would let you have a number of 'modes' for your
> profile that had SPROG and NCE. One mode could be hardware for both, another
> mode for both simulated, another for [...]

Yes, this sounds like what I am planning.

“Connection Mode” is not the best term for this, but its what I managed to come up with to describe this (so its the working term). Since it’s causing confusion, it does need to be changed before putting this into a test release.

rhwood, 2016-07-30:
The conceptual model I have for Connections in JMRI 4.4 is a hierarchical model within each Profile:

* Profile
  * Computer

The __Prefix__, __Name__, and __Manufacturer__ for each and every connection is set at the _Profile_ level. The last used configuration on any computer is also stored at the _Profile_ level.

The _Computer_ level contains everything else needed for the connection to work. Connections not matching the __Prefix__, __Name__, and __Manufacturer__ set within the Profile on a given computer are discarded, and missing connections at the computer level are copied from the last used configuration stored at the _Profile_ level.

The conceptual model I have for Connection Modes extends that hierarchical model within each Profile:

* Profile
  * Computer
    * Mode

such that the _Mode_ level contains everything else needed for the connection to work, but that any Mode remains local to the _Computer_ on which it was created. What's set at the _Profile_ level remains unchanged in the model from 4.4, but what's set at the _Computer_ level in 4.4 is moved to the _Mode_ level.

pabender, 2016-08-01:
On Jul 30, 2016 12:30 PM, "Randall Wood" <notifications@github.com> wrote:
> The conceptual model I have for Connection Modes extends that
hierarchical model within each Profile:
>
> Profile
> Computer
> Mode
>
> such that the Mode level contains everything else needed for the
connection to work, but that any Mode remains local to the Computer on
which it was created. What's set at the Profile level remains unchanged in
the model from 4.4, but what's set at the Computer level in 4.4 is moved to
the Mode level.
>

Let me throw a bit of a monkey wrench into all of this....

A while ago, I introduced a few stream port connection types  This was
specifically introduced to handle connections that live on the remote
serial port of an XBee connection, but it is useful in many situations ( I
used it to provide the XPressNet tunneling supprt for the roco Z21, for
example )

I need to get back to this, but with the XBee connections, we end up with a
tree in the configuration:

Profile
Connection
Node (s)
Remote Connection ( one per node for XBee )

And there isn't anything that says the remote connection can't be an XBee
node that tunnels some other set of protocols, which could even be another
XBee network.

So how do we handle nested connections that require configuration? ( the
Z21 connection does not.  The XBee network, and some of our supported
network connections may... )

Paul

bobjacobsen, 2016-08-02:
I think I had a different conceptual model for this:
a) I was thinking that a "Connection Mode" was, well, one of a few specific modes for a single Connection.  The plural "modes" in Connection Modes was that e.g. a LocoNet connection L2 could have simulator vs PR3 selected at start-up time as one of two preconfigured modes.  

   I then assumed that if more than one Connection was defined with more than one Mode, you'd get multiple requests to pick a Mode.  I still think this is the most conceptually simple approach, particularly if the name is Connection Modes. It does have the disadvantage that you're selecting by-connection, therefore in some cases more than once, but you're actually selecting the _mode_.

b) It them seemed that "Connection Modes" refers to a higher-level thing, a mode that lives across all configured Connection items, e.g. all Connections, at once.  If it didn't sound go bad, perhaps it should be called "_Connections_ Modes".   OK, if that's how the implementors want it, fine, though I think the language could use some work to avoid later confusion.  

c) But now there seems to be a mix of "computer" and "mode" (in the (b) sense) levels that I don't fully understand the value of.  Why is this not just more complexity?  E.g. how does "computer" and "mode" inherently differ if the original use case for "mode" was a set of different connections?

The basic approach seems to be adding _multiple_ _levels_ of commonality in the configuration:  In level N, configurations from higher are global (change once), and the ones specific to N break out into multiple choices (change multiple), which can then include further specialization at lower levels.

But I'm not sure this scales, because one person's hierarchy is not the same as somebody else's.  Some have a constant computer, e.g. a single laptop, but have different layout connection details at different times.  Some share a single layout connection config across different computers. Some have different panel&script setups for different operating modes on a single common layout configuration, with or without computer sharing.

I suspect any single hierarchy will work brilliantly for some set of (advanced) users, and will be just annoyingly confusing for others. Yes, those others don't have to use it, but if it's the thing JMRI has, somebody on JMRIusers will tell them a weird & confusing way to use it to sort-of meet their needs, and everybody in the thread will end up a bit unhappy.

Hierarchies impose structure.  They work when that structure has a 1-to-1 mapping to reality:  When they're expressing something essential.  They cause trouble when they're forcing something that's not essential, when reality (or common mental models of it) and the hierarchy don't match up _always_. 

In this case, I think (a) is a better solution than (b), which is a better solution than (c) as I understand it. I base that opinion on "How can I explain this to an average JMRI user who's trying to do something just a bit more complicated than he has been?" and "Does it provide as much flexibility as possible?"

geowar1, 2017-01-22:
Maybe its the programmer in me but… ;-)
Why not use profile inheritance? New profiles can be based on (inherit from) existing profile and just override anything they want to do different. Is that too complicated for non-programmers? ;-)

rhwood, 2017-01-22:
> Why not use profile inheritance? New profiles can be based on (inherit from) existing profile and just override anything they want to do different. Is that too complicated for non-programmers? ;-)

One of the goals of profiles is to be able make them entirely self-contained so a single profile can be (for example) copied to a thumb drive from one computer to another and "just work".

geowar1, 2017-01-22:
On Jan 22, 2017, at 8:57 AM, Randall Wood <notifications@github.com> wrote:
> One of the goals of profiles is to be able make them entirely self-contained so a single profile can be (for example) copied to a thumb drive from one computer to another and "just work”
Implementation detail: The profiles could still be self-contained on disk… they could store their parent profile as a baseline and then the overridden properties seperately. At runtime if the parent profile is available load it and then the overridden properties if the parent profile isn’t available then load the baseline and then the overridden properties.

implementation details aside; conceptually, might this be better than introducing “modes”?
-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

rhwood, 2017-01-22:
That "implementation detail" is a very explicitly *not self-contained* profile that prevents a profile from being self-contained by introducing a situation where a change in one profile's configuration can have the effect of breaking another profile. A self-contained profile has *no* dependency on another profile.

closed 2018-03-05 by rhwood