JMRI/JMRI #1721: Control Panel Editor NPE

silverailscolo, 2016-07-31:
Choosing **Item Palette** from the **Add Items** menu on a new empty Control Panel (Editor) gives an NPE on 4.5.1 (on macOS, no errors in 4.4):
``
2016-07-31 20:54:42,091 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.display.palette.FamilyItemPanel.makeFamilyButtons(FamilyItemPanel.java:356)
	at jmri.jmrit.display.palette.FamilyItemPanel.initIconFamiliesPanel(FamilyItemPanel.java:277)
	at jmri.jmrit.display.palette.FamilyItemPanel.makeBottomPanel(FamilyItemPanel.java:120)
	at jmri.jmrit.display.palette.FamilyItemPanel.init(FamilyItemPanel.java:72)
	at jmri.jmrit.display.palette.TableItemPanel.init(TableItemPanel.java:73)
	at jmri.jmrit.display.palette.ItemPalette.buildTabPane(ItemPalette.java:379)
	at jmri.jmrit.display.palette.ItemPalette.<init>(ItemPalette.java:362)
	at jmri.jmrit.display.controlPanelEditor.ControlPanelEditor$1.actionPerformed(ControlPanelEditor.java:206)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
``
I am inclined to read this upwards, but spot no direct recent change.

bobjacobsen, 2016-08-01:
I wasn't able to recreate this. From reading the code, I think #1722 (now merged) should fix it.  Could you try that and let me know?  Thanks.

silverailscolo, 2016-08-01:
Thanks, this fixed it! (Which BTW demonstrates that even small things matter)

closed 2016-08-02 by silverailscolo