JMRI/JMRI #1731: NPE when starting JMRI

DanielBoudreau, 2016-08-03:
I'm seeing an NPE at start up.  The only connection is NCE simulator.

2016-08-02 21:01:24,903 apps.Apps                             ERROR - Error trying to setup preferences null [init prefs]
java.lang.NullPointerException
	at jmri.jmrix.ConnectionConfigManager$ConnectionTypeManager.getConnectionTypes(ConnectionConfigManager.java:328)
	at jmri.jmrix.ConnectionConfigManager.getConnectionTypes(ConnectionConfigManager.java:231)
	at jmri.jmrix.JmrixConfigPane.<init>(JmrixConfigPane.java:214)
	at jmri.jmrix.JmrixConfigPane.createPanel(JmrixConfigPane.java:88)
	at jmri.jmrix.JmrixConfigPane.instance(JmrixConfigPane.java:59)
	at jmri.jmrix.swing.ConnectionsPreferencesPanel.<init>(ConnectionsPreferencesPanel.java:72)
	at jmri.jmrix.swing.ConnectionsPreferencesPanel.<init>(ConnectionsPreferencesPanel.java:85)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
	at java.lang.Class.newInstance(Class.java:438)
	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:172)
	at apps.Apps$3.run(Apps.java:379)
	at java.lang.Thread.run(Thread.java:745)

rhwood, 2016-08-04:
I am unable to recreate this using either a built and installed version of PanelPro from master, or by running ```ant panel pro```. How are you launching a JMRI application to get this?

DanielBoudreau, 2016-08-05:
From Eclipse, Run PanelPro.

DanielBoudreau, 2016-08-05:
I just pulled all recent changes and did a clean, and the NPE seems to be fixed.  Still have a problem with any type of manufacture and not selecting a valid port. That causes an NPE:

2016-08-04 20:58:43,518 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-08-04 20:58:43,535 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Daniel Boudreau\JMRI\log\messages.log [main]
2016-08-04 20:58:43,536 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Daniel Boudreau\JMRI\log\session.log [main]
2016-08-04 20:58:43,542 apps.Apps                             INFO  - PanelPro version 4.5.2ish-201608042319-Daniel Boudreau starts under Java 1.8.0_25 on Windows 7 amd64 v6.1 at Thu Aug 04 20:58:43 EDT 2016 [main]
2016-08-04 20:58:43,912 apps.Apps                             INFO  - Starting with profile PanelPro.3ef219d3 [main]
2016-08-04 20:58:44,115 node.NodeIdentity                     INFO  - Using jmri-0C84DCBAA785-3ef219d3 as the JMRI Node identity [main]
2016-08-04 20:58:44,359 serialdriver.SerialDriverAdapter      ERROR - Serial port (none selected) not found [main]
2016-08-04 20:58:44,361 configurexml.ErrorHandler             ERROR - (none selected) not found while opening connection [main]
2016-08-04 20:58:46,829 jmri.InstanceManager                  ERROR - Should not set default of type jmri.PowerManager to null value [main]
2016-08-04 20:58:46,830 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.NullPointerException
	at jmri.InstanceManager.setDefault(InstanceManager.java:235)
	at jmri.managers.ManagerDefaultSelector.configure(ManagerDefaultSelector.java:159)
	at jmri.managers.ManagerDefaultSelector.initialize(ManagerDefaultSelector.java:214)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:249)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$119(JmriConfigurationManager.java:245)
	at jmri.implementation.JmriConfigurationManager$$Lambda$30/1897115967.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:244)
	at jmri.implementation.JmriConfigurationManager.lambda$load$118(JmriConfigurationManager.java:178)
	at jmri.implementation.JmriConfigurationManager$$Lambda$29/243745864.accept(Unknown Source)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:177)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:166)
	at apps.Apps.<init>(Apps.java:266)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:98)

rhwood, 2016-08-05:
How do you "Run PanelPro" in Eclipse?

DanielBoudreau, 2016-08-05:
Select the app you want to run, right click, Run As, and select "Java Application".  

More info here:

http://jmri.sourceforge.net/help/en/html/doc/Technical/Eclipse.shtml

rhwood, 2016-08-05:
This now duplicates #1662

closed 2016-08-05 by rhwood