JMRI/JMRI #1751: Set up Jenkins to run non-headless tests

bobjacobsen, 2016-08-06:
Currently, Jenkins is running the headless versions of the CI-tests.

- [ ] install xvfb and its Jenkins plugin
- [ ] reconfigure Builds task and test
- [ ] reconfigure JaCoCo and test

bobjacobsen, 2016-08-06:
[Jenkins plugin doc](https://wiki.jenkins-ci.org/display/JENKINS/Xvfb+Plugin)

Unfortunately, the first attempt to install xvfb failed:
````
$ sudo apt-get install xvfb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libgl1-mesa-dri libgl1-mesa-glx
  libglapi-mesa libllvm3.4 libpciaccess0 libtxc-dxtn-s2tc0 libx11-xcb1
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-sync1
  libxdamage1 libxfixes3 libxkbfile1 libxmuu1 libxshmfence1 libxxf86vm1
  x11-xkb-utils xauth xfonts-base xserver-common
Suggested packages:
  libglide3 xfs xserver
The following NEW packages will be installed:
  libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libgl1-mesa-dri libgl1-mesa-glx
  libglapi-mesa libllvm3.4 libpciaccess0 libtxc-dxtn-s2tc0 libx11-xcb1
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-sync1
  libxdamage1 libxfixes3 libxkbfile1 libxmuu1 libxshmfence1 libxxf86vm1
  x11-xkb-utils xauth xfonts-base xserver-common xvfb
0 upgraded, 26 newly installed, 0 to remove and 136 not upgraded.
Need to get 18.9 MB/19.0 MB of archives.
After this operation, 75.7 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Err http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libdrm-intel1 amd64 2.4.64-1~ubuntu14.04.1
  404  Not Found [IP: 2001:67c:1560:8001::11 80]
Err http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libdrm-nouveau2 amd64 2.4.64-1~ubuntu14.04.1
  404  Not Found [IP: 2001:67c:1560:8001::11 80]
Err http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libdrm-radeon1 amd64 2.4.64-1~ubuntu14.04.1
  404  Not Found [IP: 2001:67c:1560:8001::11 80]
Get:1 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libgl1-mesa-dri amd64 10.1.3-0ubuntu0.6 [4,823 kB]
Get:2 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxmuu1 amd64 2:1.1.1-1 [11.0 kB]
Get:3 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libllvm3.4 amd64 1:3.4-1ubuntu3 [6,540 kB]
Get:4 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libglapi-mesa amd64 10.1.3-0ubuntu0.6 [21.0 kB]
Get:5 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libxfixes3 amd64 1:5.0.1-1ubuntu1.1 [10.4 kB]
Get:6 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main libgl1-mesa-glx amd64 10.1.3-0ubuntu0.6 [113 kB]
Get:7 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main xserver-common all 2:1.15.1-0ubuntu2.7 [27.9 kB]
Get:8 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/main xvfb amd64 2:1.15.1-0ubuntu2.7 [747 kB]
Get:9 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libx11-xcb1 amd64 2:1.6.2-1ubuntu2 [9,426 B]
Get:10 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxcb-dri2-0 amd64 1.10-2ubuntu1 [6,710 B]
Get:11 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxcb-dri3-0 amd64 1.10-2ubuntu1 [5,118 B]
Get:12 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxcb-glx0 amd64 1.10-2ubuntu1 [20.0 kB]
Get:13 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxcb-present0 amd64 1.10-2ubuntu1 [5,254 B]
Get:14 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxcb-sync1 amd64 1.10-2ubuntu1 [8,090 B]
Get:15 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxdamage1 amd64 1:1.1.4-1ubuntu1 [7,612 B]
Get:16 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxshmfence1 amd64 1.1-2 [4,644 B]
Get:17 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxxf86vm1 amd64 1:1.1.3-1 [11.7 kB]
Get:18 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libxkbfile1 amd64 1:1.0.8-1 [74.2 kB]
Get:19 http://ubuntu.positive-internet.com/ubuntu/ trusty/main libtxc-dxtn-s2tc0 amd64 0~git20131104-1.1 [51.8 kB]
Get:20 http://ubuntu.positive-internet.com/ubuntu/ trusty/main xauth amd64 1:1.0.7-1ubuntu1 [25.8 kB]
Get:21 http://ubuntu.positive-internet.com/ubuntu/ trusty/main x11-xkb-utils amd64 7.7+1 [156 kB]
Get:22 http://ubuntu.positive-internet.com/ubuntu/ trusty/main xfonts-base all 1:1.0.3 [6,180 kB]
Fetched 18.9 MB in 2s (9,245 kB/s)        
E: Failed to fetch http://gb.archive.ubuntu.com/ubuntu/pool/main/libd/libdrm/libdrm-intel1_2.4.64-1~ubuntu14.04.1_amd64.deb  404  Not Found [IP: 2001:67c:1560:8001::11 80]

E: Failed to fetch http://gb.archive.ubuntu.com/ubuntu/pool/main/libd/libdrm/libdrm-nouveau2_2.4.64-1~ubuntu14.04.1_amd64.deb  404  Not Found [IP: 2001:67c:1560:8001::11 80]

E: Failed to fetch http://gb.archive.ubuntu.com/ubuntu/pool/main/libd/libdrm/libdrm-radeon1_2.4.64-1~ubuntu14.04.1_amd64.deb  404  Not Found [IP: 2001:67c:1560:8001::11 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
````

More to follow after discussion...

bobjacobsen, 2016-08-15:
xvfb installed at:
````
/usr/bin/xvfb-run
/usr/bin/Xvfb
````

bobjacobsen, 2016-08-16:
I think this is working now, starting with [JaCoCo build 200](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/200/).

In addition to xvfb, also had to install libxi6 and libxtst6.

pabender, 2016-08-16:
On Aug 15, 2016, at 8:22 PM, Bob Jacobsen <notifications@github.com> wrote:
> I think this is working now, starting with JaCoCo build 200.
> 
That helped.  Class coverage in the tests went from 26% to 41% and line coverage went from 16% to 25%

Thanks!

Paul

bobjacobsen, 2016-08-16:
> Class coverage in the tests went from 26% to 41% and line coverage went from 16% to 25%

Because we run with TRACE and DEBUG off, some statements can’t be reached.  

Should we run this (but not the regular tests) with those set on?  It’ll take a bit of infrastructure work, but not much.

pabender, 2016-08-16:
On 08/15/2016 10:13 PM, Bob Jacobsen wrote:
>> Class coverage in the tests went from 26% to 41% and line coverage
> went from 16% to 25%
>
> Because we run with TRACE and DEBUG off, some statements can’t be reached.
>
> Should we run this (but not the regular tests) with those set on? It’ll
> take a bit of infrastructure work, but not much.

For now, no.

The issue is when you turn on DEBUG and/or TRACE, globally, we can't 
test the branch points that depend on those values.  (Branch points is 
one of the other metrics).

We actually need to be able to run each test with DEBUG or TRACE (as 
appropriate) both on and off to catch both sides of the branch point.

Paul

bobjacobsen, 2016-08-16:
> The issue is when you turn on DEBUG and/or TRACE, globally, we can't 
> test the branch points that depend on those values. (Branch points is 
> one of the other metrics).

OK, but there shouldn’t be much of the form:
````
if isDebugEnabled() { /* stuff */ }
else { /* other stuff*/ }
````
because there shouldn’t be much (if anything) that’s _not_ run when logging is turned up.  So we’re still doing the same thing at `ifDebugEnabled()` branch points: Just taking one path. But the statement coverage would be a more realistic indicator, particularly in short methods with logging.

What we really need is a way to tell JaCoCo “isDebugEnabled() will always evaluate to false, ignore the other branch”, but I haven’t been able to find anything like that using a Google search.

pabender, 2016-08-16:
On 08/15/2016 10:58 PM, Bob Jacobsen wrote:
> OK, but there shouldn’t be much of the form:
> ````
> if isDebugEnabled() { /* stuff */ }
> else { /* other stuff*/ }
> ````
> because there shouldn’t be much (if anything) that’s _not_ run when
> logging is [...]

True, I hadn't thought about it that way.

> What we really need is a way to tell JaCoCo “isDebugEnabled() will
> always evaluate to false, ignore the other branch”, but I haven’t been
> able to find anything like that using a Google search.

Right.  It looks like there has been some effort for building line level 
filtering into Jacoco, but the Jacoco developers think it is pretty low 
prioroity

see:
https://github.com/jacoco/jacoco/wiki/FilteringOptions
https://github.com/jacoco/jacoco/issues/14
https://github.com/jacoco/jacoco/issues/15

Paul

bobjacobsen, 2016-08-16:
> It looks like there has been some effort for building line level filtering into Jacoco, but the Jacoco developers think it is pretty low prioroity

Doesn't sound like it'll happen soon.

Our log4j-under-JUnit code gets confused by DEBUG et al messages, but it wouldn't be tricky to fix that. Would it make sense to do that and then put two `<junit>` blocks in the JaCoCo Ant target, one without and one with enhanced logging?

Bob

pabender, 2016-08-16:
On 08/16/2016 12:16 PM, Bob Jacobsen wrote:
> filtering into Jacoco, but the Jacoco developers think it is pretty low
> prioroity
>
> Doesn't sound like it'll happen soon.
>
> Our log4j-under-JUnit code gets confused by DEBUG et al messages, but it [...]

I think that's worth a shot.

Paul

bobjacobsen, 2016-08-17:
>> Our log4j-under-JUnit code gets confused by DEBUG et al messages, but it
>> wouldn't be tricky to fix that. Would it make sense to do that and then
>> put two `<junit>` blocks in the JaCoCo Ant target, one without and one
>> with enhanced logging?
> 
> I think that's worth a shot.

Unfortunately, that doesn’t quite work.  When I try it, Ant complains:

````
build.xml:1309: Only one child task can be supplied to the coverge task
````

Using a different .lcf file and message handler does raise the coverage, so even if we’re only getting one side of the `if isDebugEnabled()` I think it’s still a nice reduction in false-negatives.

Having a special version of AllTest that runs twice with different setups is also possible, I think.

rhwood, 2016-08-17:
If I'm understanding this correctly, there is concern about ensuring that all log.ifDebugEnabled() branches are tested.

Would removing all ifDebugEnabled protections and changing manually formatted log messages to use the slf4j formatters (except where documented in comments) help?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

DanielBoudreau, 2016-08-17:
I like Randall’s solution.

 

Dan

 

randall.h.wood@alexandriasoftware.com <mailto:randall.h.wood@alexandriasoftware.com> 
http://alexandriasoftware.com

pabender, 2016-08-17:
On Aug 17, 2016, at 7:40 AM, Randall Wood <notifications@github.com> wrote:

> If I'm understanding this correctly, there is concern about ensuring that all log.ifDebugEnabled() branches are tested.

Yes and no.  Jenkins once to see every branch of every if statement covered.  We can't get to 100% test coverage if all branches are not tested, but 100% may not be a reasonable goal for now.  Maybe we need to strive for 80% now, and once we are there, then we set a higher goal.

> Would removing all ifDebugEnabled protections and changing manually formatted log messages to use the slf4j formatters (except where documented in comments) help?

To the best of my knowledge, there is still a performance penalty building the output strings.  That is the main reason for the if DebugEnabled statements.

> 

Paul

rhwood, 2016-08-17:
> To the best of my knowledge, there is still a performance penalty building the output strings. That is the main reason for the if DebugEnabled statements.

Using slf4j’s formatters ensures the output strings are not built unless actually logged. See http://www.slf4j.org/faq.html#logging_performance

bobjacobsen, 2016-08-17:
> Using slf4j’s formatters ensures the output strings are not built unless actually logged.

That works for 
````
   if (log.isDebugEnabled()) log.debug("Decoded as {}, ret);
````

but not so well for 
````
  if (log.isDebugEnabled()) {
     log.debug("forwarding message: {}", m.getMessage());
  }
````
where there's a lot of work in creating the arguments.

A quick grep scan found:
- 7393 log.debug calls, of which 6693 have a ; on the same line so are likely to be single-line statements
- 3074 `if (log.isDebugEnabled())` lines, of which almost all (2890) are `if (log.isDebugEnabled()) {` e.g. the actual log statement is on the next line (makes it a little harder to automate removal)
- 1672 of the log.debug calls contain a comma, 1258 contain an open brace on the same line, from which I conclude that about a fifth of the log.debug statements are already using the slf4j formatters
- 2902 (about a half to a third) of the log.debug calls contain a `.`; a hand-scan indicates that about 95% of those are method calls, with maybe 2/3 calls to getters (which are hopefully cheap), but maybe 20% looking more expensive. There are also a hundred or so array element retrievals, some with calculations for the index.

Clearly, it's a good thing to move toward slf4j formatters everywhere, with only expensive calls protected by `if (log.isDebugEnabled())`.  That makes the code cleaner and even helps with getting a correct coverage number.  That's a long pull that we should start.

But I also think we should try (at least as an experiment for a while) running the JaCoCo coverage numbers with INFO/DEBUG/TRACE turned on. Other than about 20% on run-time, is there a reason not to do that?

pabender, 2016-08-17:
On Aug 17, 2016 11:52 AM, "Bob Jacobsen" <notifications@github.com> wrote:
> But I also think we should try (at least as an experiment for a while)
running the JaCoCo coverage numbers with INFO/DEBUG/TRACE turned on. Other
than about 20% on run-time, is there a reason not to do that?
>

Runtime is relatively cheap, so i say go ahead and turn on trace, which
should be debyg plus other information, and we can work towards coverting
as many of the debug statements to the newer form.   Over time.

Paul

bobjacobsen, 2016-08-18:
I just ran a [Jenkins test job](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/206/) from the `two-pass-jacoco` branch which runs with DEBUG on, and it seems to be working:

[image]

It didn't make much of a difference to the overall statistics, though, which is a surprise to me.

pabender, 2016-08-18:
On 08/17/2016 08:03 PM, Bob Jacobsen wrote:
> It didn't make much of a difference to the overall statistics, though,
> which is a surprise to me.

Remember that we're talking about somewhere in the neighborhood of 
350,000 total lines of code, and the debugging statements are a  small 
percentage of that.

If you look at smaller pieces, this is a more impressive change.  As an 
example, look at the lenz results between build 205:

http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/205/jacoco/jmri.jmrix.lenz/

and 206

http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/206/jacoco/jmri.jmrix.lenz/

In 205, there is 56% line coverage.  In 206, there is 58% line coverage. 
  A 2% improvement is not insignificant here*.

Individual classes are even more impressive, for example, the 
jmri.jmrix.lenz.XNetSensor class went from 86% to 90% between the two 
builds.  More impressive is that the instruction coverage went from 71% 
to 92% for the same class.

Paul

*For my dissertation, I had to explain why we only saw a 1-2% 
performance increase in battery life on a wireless sensor network 
instead of the 16 times improvement the math predicted.  It turned out 
the hardware wasn't designed to do what we needed to to get the 
predicted results (transmit power was adjustable, but not the receive 
power.)

bobjacobsen, 2016-08-18:
PR #1823 implements "run with DEBUG" for the Jenkins JaCoCo job.  

I'm going to close this Issue (which is really about non-headless tests).  Please open another if there's further discussion needed on e.g. use of `isDebugEnabled()`

closed 2016-08-18 by bobjacobsen