JMRI/JMRI #1775: Can't create new Xbee connection

KenC57, 2016-08-12:
After selecting the connection type (IEEE 802.15.4) I get this trace and can't get to selecting com ports etc... It looks like it is in a try/catch looking for the NPE, but misses the fact that the list is zero length. Now as for why the getList puts a null list in there, I'm not sure of why that's useful.

-ken c

2016-08-12 11:32:39,819 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-08-12 11:32:39,841 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\kcameron\JMRI\log\messages.log [main]
2016-08-12 11:32:39,842 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\kcameron\JMRI\log\session.log [main]
2016-08-12 11:32:39,848 apps.Apps                             INFO  - PanelPro version 4.5.2ish-201608041309-kcameron starts under Java 1.8.0_101 on Windows 10 amd64 v10.0 at Fri Aug 12 11:32:39 EDT 2016 [main]
2016-08-12 11:32:44,459 apps.Apps                             INFO  - Starting with profile Xbee_ZB.3f7dfdee [main]
2016-08-12 11:32:44,724 node.NodeIdentity                     INFO  - Using jmri-84A6C8D92D62-3f7dfdee as the JMRI Node identity [main]
2016-08-12 11:32:46,308 util.FileUtilSupport                  INFO  - File path program: is C:\Users\kcameron\workspace\jmri-git6\JMRI\ [main]
2016-08-12 11:32:46,308 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\kcameron\JMRI\Xbee_ZB\ [main]
2016-08-12 11:32:46,309 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\kcameron\JMRI\Xbee_ZB\ [main]
2016-08-12 11:32:46,309 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\kcameron\JMRI\ [main]
2016-08-12 11:32:46,309 util.FileUtilSupport                  INFO  - File path home: is C:\Users\kcameron\ [main]
2016-08-12 11:32:46,310 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\kcameron\workspace\jmri-git6\JMRI\jython\ [main]
2016-08-12 11:33:04,644 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(Unknown Source)
	at java.util.ArrayList.get(Unknown Source)
	at jmri.jmrix.ieee802154.swing.nodeconfig.NodeConfigAction.<init>(NodeConfigAction.java:29)
	at jmri.jmrix.ieee802154.xbee.swing.nodeconfig.NodeConfigAction.<init>(NodeConfigAction.java:23)
	at jmri.jmrix.ieee802154.xbee.swing.nodeconfig.NodeConfigAction.<init>(NodeConfigAction.java:40)
	at jmri.jmrix.ieee802154.xbee.ConnectionConfig.loadDetails(ConnectionConfig.java:35)
	at jmri.jmrix.JmrixConfigPane.selection(JmrixConfigPane.java:330)
	at jmri.jmrix.JmrixConfigPane.lambda$1(JmrixConfigPane.java:258)
	at javax.swing.JComboBox.fireActionEvent(Unknown Source)
	at javax.swing.JComboBox.setSelectedItem(Unknown Source)
	at jmri.jmrix.JmrixConfigPane.updateComboConnection(JmrixConfigPane.java:310)
	at jmri.jmrix.JmrixConfigPane.lambda$0(JmrixConfigPane.java:210)
	at javax.swing.JComboBox.fireActionEvent(Unknown Source)
	at javax.swing.JComboBox.setSelectedItem(Unknown Source)
	at javax.swing.JComboBox.setSelectedIndex(Unknown Source)
	at javax.swing.plaf.basic.BasicComboPopup$Handler.mouseReleased(Unknown Source)
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at javax.swing.plaf.basic.BasicComboPopup$1.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

rhwood, 2016-08-12:
From [FindBug's Bug Descriptions]:
>It is often a better design to return a length zero array rather than a null reference to indicate that there are no results (i.e., an empty list of results). This way, no explicit check for null is needed by clients of the method.

This implies that you should be checking that the list is not empty instead of checking that the list is null. The quick fix is to change the NPE catch to a catch for this exception.

pabender, 2016-08-13:
On 08/12/2016 11:49 AM, Ken Cameron wrote:
> After selecting the connection type (IEEE 802.15.4) I get this trace
> and can't get to selecting com ports etc... It looks like it is in a
> try/catch looking for the NPE, but misses the fact that [...]
<snip>
> 2016-08-12 11:33:04,644 ptionhandler.UncaughtExceptionHandler ERROR - 
> Uncaught Exception caught by 
> jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
>
> java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
>

Regardless of whether or not this is useful, I've added the ability to 
catch to the IndexOutOfBoundsException.

Of course, this needs to be revisited anyway.  The current 
implementation only looks at the FIRST entry in the list of system 
connection memos, and it needs to be associated with a specific memo 
that is passed to the constructor.

I'll work on that, but for the moment I'll check in the updated 
exception handling.

Paul
>

KenC57, 2016-08-18:
Paul,
That cures what I was having trouble with.
-ken c

closed 2016-11-13 by pabender