JMRI/JMRI #1792: Create test release 4.5.3

bobjacobsen, 2016-08-15:
For discussion of timing and what to include in 4.5.3

bobjacobsen, 2016-08-17:
(This is a copy of a post to the jmri-developers list, would prefer to have this discussion there right now)

Randall (@rhwood) has requested a quick release of 4.5.3 to fix a problem in the Mac OS X (macOS) launcher.

Happy to do that tomorrow, which is the soonest I’ll have a good enough connection.

Our choices are:

A) Take v4.5.2 plus just this change

B) Take v4.5.2 plus N other changes

C) Take the head of master at that time.

I’m happy with either (A) or (C).  (A) is a bit of hand work, nothing terrible.  (C) picks up other changes that have been made (good, may fix some other issues), with a bit more risk, and requires more work on the release note.

Not a big fan of (B).  If there are multiple requests, I’d prefer to go to (C).

rhwood, 2016-08-18:
The web server is now working, so (C) works for me.

bobjacobsen, 2016-08-20:
Starting (C) now.

bobjacobsen, 2016-08-20:
The release-4.5.3 branch has been created. 

From now on, please document your changes in the [jmri4.5.4.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.5.4.shtml) release note file.

Maintainers, please set the 4.5.4 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [new server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.3/)

closed 2016-08-22 by bobjacobsen