JMRI/JMRI #1797: RE Issue 1662

fiz1962, 2016-08-15:
JMRI 4.5.2 on Windows 10

Profiles now load with ver 4.5.2 but every time it loads a dialog pops up saying 

"System connection LocoNet provides a null manager for interface jmri.ConsistManager\n Please check the logs for more details.\n The Preferences window will open so this can be fixed"

Clicking OK opens the preferences window with the following fields filled in
System Manufacturer - Digitrax
System Connection - LocoNet PR3
Settings - COM5, DCS100 (Chief), L, LocoNet for Serial Port, Command Station Type, Prefix and Name.

I can save those settings and restart JMRI, and it starts with the same Dialog wanting to open Preferences again.

rhwood, 2016-08-15:
What do the logs state when this happens?

rhwood, 2016-08-15:
Also what are you connection defaults (Preferences->Defaults) set to?

DanielBoudreau, 2016-08-15:
Randall,

 

FYI. I tried the same settings on Win7 and I don’t see a problem when using 4.5.2.

 

System Manufacturer - Digitrax
System Connection - LocoNet PR3
Settings - COM5, DCS100 (Chief),

 

Dan

rhwood, 2016-08-15:
@fiz1962 Within your configuration profile, there are two files:
- profile\profile.properties
- profile\profile.xml

Can you attach those files to this issue (it may be easier to simply zip up the entire profile directory and attach that)?

fiz1962, 2016-08-22:
Sorry guys.  Things have gotten a bit hectic.  I've reverted back to 4.4 for now.  Hopefully can take this back up in a week or so...

DanielBoudreau, 2016-09-01:
One of our users emailed to me his console log and asked why he was seeing warning and error messages.  Here's the log:
2016-08-29 20:07:04,875 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-08-29 20:07:04,875 util.Log4JUtil                        INFO  - This log is appended to file: C:\Documents and Settings\Jean A. Waraxa\JMRI\log\messages.log [main]
2016-08-29 20:07:04,890 util.Log4JUtil                        INFO  - This log is stored in file: C:\Documents and Settings\Jean A. Waraxa\JMRI\log\session.log [main]
2016-08-29 20:07:04,906 apps.Apps                             INFO  - PanelPro version 4.5.3-R7b2f835 starts under Java 1.8.0_31 on Windows XP x86 v5.1 at Mon Aug 29 20:07:04 CDT 2016 [main]
2016-08-29 20:07:05,921 apps.Apps                             INFO  - Starting with profile PanelPro.3f0a01fa [main]
2016-08-29 20:07:06,265 node.NodeIdentity                     INFO  - Using jmri-00132043A5CF-3f0a01fa as the JMRI Node identity [main]
2016-08-29 20:07:06,453 managers.JmriUserPreferencesManager   WARN  - java.lang.NoSuchMethodException: apps.AppConfigBase.setMessagePreferencesDetails() [main]
2016-08-29 20:07:08,015 simpleserver.SimpleServer             INFO  - JMRI SimpleServer started on port 2048 [main]
2016-08-29 20:07:08,062 configurexml.PerformFileModelXml      INFO  - Load file C:\Documents and Settings\Jean A. Waraxa\JMRI\operationsComp.xml [main]
2016-08-29 20:07:08,734 loconetovertcp.Server                 INFO  - Starting new LocoNetOverTcpServer listener on port 1234 [main]
2016-08-29 20:07:08,734 loconetovertcp.Server                 INFO  - Starting ZeroConfService _loconetovertcpserver._tcp.local for LocoNetOverTCP Server [main]
2016-08-29 20:07:08,750 managers.DefaultShutDownManager       ERROR - already contains jmri.util.zeroconf.ZeroConfService$ShutDownTask@4fa968 [main]
2016-08-29 20:07:09,187 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-08-29 20:07:09,187 util.FileUtilSupport                  INFO  - File path preference: is C:\Documents and Settings\Jean A. Waraxa\JMRI\ [main]
2016-08-29 20:07:09,187 util.FileUtilSupport                  INFO  - File path profile: is C:\Documents and Settings\Jean A. Waraxa\JMRI\PanelPro\ [main]
2016-08-29 20:07:09,187 util.FileUtilSupport                  INFO  - File path settings: is C:\Documents and Settings\Jean A. Waraxa\JMRI\ [main]
2016-08-29 20:07:09,203 util.FileUtilSupport                  INFO  - File path home: is C:\Documents and Settings\Jean A. Waraxa\ [main]
2016-08-29 20:07:09,203 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-08-29 20:07:42,171 setup.BackupBase                      INFO  - Saving copy of operations files to: C:\Documents and Settings\Jean A. Waraxa\JMRI\operations\autoBackups\2016_08_29_00 [AWT-EventQueue-0]
2016-08-29 20:07:45,859 cars.CarRoads                         INFO  - Max road name (EBNSF) length 5 [Build Train]
2016-08-29 20:07:45,859 locations.LocationManager             INFO  - Max track name (KR Intg BNSF east cars) at (KR INTG) length 22 [Build Train]
2016-08-29 20:07:45,875 locations.LocationManager             INFO  - Max location name (BURMAN QUARRY) length 13 [Build Train]
2016-08-29 20:07:45,875 locations.LocationManager             INFO  - Max location and track name (CLARK FORK, *09CF grain off spot) length 30 [Build Train]
2016-08-29 20:07:45,875 cars.CarTypes                         INFO  - Max car type name () length 4 [Build Train]
2016-08-29 20:08:46,156 loconetovertcp.LnOverTcpPacketizer    WARN  - run: input stream returned null, exiting loop [LocoNet receive handler]
2016-08-29 20:09:43,031 setup.AutoSave                        INFO  - Operation files automatically saved [Operations Auto Save]
2016-08-29 20:15:07,890 managers.DefaultShutDownManager       ERROR - already contains jmri.util.zeroconf.ZeroConfService$ShutDownTask@4fa968 [AWT-EventQueue-0]
2016-08-29 20:15:07,890 managers.DefaultShutDownManager       ERROR - already contains jmri.util.zeroconf.ZeroConfService$ShutDownTask@4fa968 [Thread-16]
2016-08-29 20:15:08,796 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

Any ideas?

Dan

rhwood, 2016-09-01:
For some reason, his ZeroConfService is not publishing its connection, so its attempting to re-register itself whenever it attempts with the shutdown manager. I don’t know why its not.

Can you ask for the contents of the following files?
- C:\Documents and Settings\Jean A. Waraxa\JMRI\PanelPro\profile\profile.properties
- C:\Documents and Settings\Jean A. Waraxa\JMRI\PanelPro\profile\jmri-00132043A5CF-3f0a01fa\profile.properties

There are potentially settings in one of those files that may be disabling the ZeroConfService on either IPv6 or IPv4. If its not those settings, then there is something on that computer’s network stack that is preventing the publication of the service—we deliberately log those only at the debug level, since that’s an incredibly common, noisy, and usually safe to ignore issue (a modern computer might have many interfaces, only some of which allow the publication of the service; its easiest to publish on all interfaces and see which stick than it is to bother the user with these details, and its rare that none of them work).

(Why the ShutDownManager is throwing an error when an existing ShutDownTask is reregistered is also an interesting question—I would silently ignore the reregistration (or log it at the debug level).)
 
Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

DanielBoudreau, 2016-09-02:
- C:\Documents and Settings\Jean A. Waraxa\JMRI\PanelPro\profile\profile.properties:

#JMRI Preferences version 4.5.3ish-201608181240-jenkins-R8afdadc
#Sun Aug 28 10:24:34 CDT 2016
jmri-implementation.scripts=program\:jython/
jmri-implementation.user-files=settings\:

- C:\Documents and Settings\Jean A. Waraxa\JMRI\PanelPro\profile\jmri-00132043A5CF-3f0a01fa\profile.properties:

#JMRI Preferences version 4.5.3ish-201608181240-jenkins-R8afdadc
#Sun Aug 28 10:24:34 CDT 2016
apps-gui.fontSize=15
apps-gui.locale=en-US
apps-gui.lookAndFeel=Windows Classic
apps-gui.nonstandardMouseEvent=false
apps-gui.showToolTipDismissDelay=4000
apps-systemconsole.fontFamily=Monospaced
apps-systemconsole.fontSize=12
apps-systemconsole.fontStyle=0
apps-systemconsole.scheme=0
apps-systemconsole.wrapStyle=2
jmri-implementation.scripts=program\:jython/
jmri-implementation.user-files=settings\:
jmri-jmris-json.heartbeatInterval=15000
jmri-jmris-json.port=2056
jmri-jmris-simpleserver.port=2048
jmri-jmris-srcp.port=4303
jmri-jmrit-roster.defaultOwner=Inspector Dave
jmri-jmrit-roster.directory=preference\:
jmri-jmrit-symbolicprog.defaultFile=Basic
jmri-jmrit-symbolicprog.showCvNumbers=false
jmri-jmrit-symbolicprog.showEmptyPanes=true
jmri-managers.defaults.jmri-AddressedProgrammerManager=LocoNet
jmri-managers.defaults.jmri-CommandStation=LocoNet
jmri-managers.defaults.jmri-ConsistManager=LocoNet
jmri-managers.defaults.jmri-GlobalProgrammerManager=LocoNet
jmri-managers.defaults.jmri-PowerManager=LocoNet
jmri-managers.defaults.jmri-ThrottleManager=LocoNet
jmri-web-server.allowRemoteConfig=false
jmri-web-server.clickDelay=1
jmri-web-server.disableFrames=false
jmri-web-server.disallowedFrames.0=DecoderPro
jmri-web-server.disallowedFrames.1=PanelPro
jmri-web-server.disallowedFrames.2=LocoNet Simulator
jmri-web-server.disallowedFrames.3=Preferences
jmri-web-server.disallowedFrames.4=WiThrottle
jmri-web-server.disallowedFrames.5=Change Signal Head Icons
jmri-web-server.disallowedFrames.6=Item Palette
jmri-web-server.port=12080
jmri-web-server.railRoadName=BNSF Kootenai River sub
jmri-web-server.readonlyPower=true
jmri-web-server.redirectFramesToPanels=true
jmri-web-server.refreshDelay=5
jmri-web-server.simple=false
jmri-web-server.useAjax=true
profile.id=PanelPro.3f0a01fa
profile.name=PanelPro

rhwood, 2016-09-02:
@DanielBoudreau I don't see anything obvious in your configurations. BTW, when dealing with different issues than the original poster, can you please open a new issue? The ZeroConfServices not publishing is a different issue than the null consist manager issue (sorry I did not notice this earlier).

rhwood, 2016-09-28:
@fiz1962 are you still seeing this with 4.5.4?

rhwood, 2016-11-27:
Closing since me response from reporter.

closed 2016-11-27 by rhwood