JMRI/JMRI #1806: NPE at startup in jmrit.Sound

bobjacobsen, 2016-08-16:
From @KenC57 via jmri-developers:

````
I was testing a panel file I had brought from another system. The file path
for a couple of sounds in the conditionals had full path, but not a valid
path for my test system. It would pass that path to jmri.jmrit.Sound() and
that would throw an exception. Problem was the normal menu never came up. I
let it sit for 15 minutes, still no main menu, had to kill. After I edited
the path (to a proper program: prefix instead of absolute path) it worked
fine.

I don't know why the NPE at that point hung everything, but hope somebody
else can figure it out.

Here is the trace from when it failed:

2016-08-16 15:08:59,795 util.FileUtilSupport                  INFO  - File
path scripts: is C:\Users\kcameron\workspace\jmri-git6\JMRI\jython\ [main]
2016-08-16 15:14:58,673 ptionhandler.UncaughtExceptionHandler ERROR -
Uncaught Exception caught by
jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.NullPointerException
	at jmri.jmrit.Sound.<init>(Sound.java:75)
	at jmri.jmrit.Sound.<init>(Sound.java:53)
	at
jmri.implementation.DefaultConditional.takeActionIfNeeded(DefaultConditional
.java:934)
	at
jmri.implementation.DefaultConditional.calculate(DefaultConditional.java:313
)
	at
jmri.implementation.DefaultLogix.calculateConditionals(DefaultLogix.java:215
)
	at
jmri.implementation.DefaultLogix.activateLogix(DefaultLogix.java:244)
	at
jmri.managers.DefaultLogixManager.activateAllLogixs(DefaultLogixManager.java
:165)
	at apps.Apps.<init>(Apps.java:456)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:98)
````

rhwood, 2016-08-16:
Sound is behaving [as documented](http://jmri.org/JavaDoc/doc/jmri/jmrit/Sound.html) in response to earlier complaints that it failed to handle portable paths silently (see #1272 and #1274).

Either things loading sound files should ensure they are loading valid sound files before creating a Sound object, or it needs to be possible to create Sound objects that only log errors (I'm a proponent of having the creating process validate the sound file or catch  the NPE to notify the user).

rhwood, 2016-08-18:
This appears to be resolved in #1808

closed 2016-08-18 by rhwood