JMRI/JMRI #1834: Potential issue with IDTag and RailCom objects.

pabender, 2016-08-19:
While working on support for RailCom on Roco Z21 systems, I discovered a potential problem that needs to be investigated.

The issue comes about because we have IdTag objects and RailCom objects (jmri.RailCom extends jmri.IdTag).

IdTag objects are provided by the default IdTagManager.

Railcom objects are provided by the default RailComManager, but they don't appear to ever be registered with the IdTag manager.  (There is currently only one system (Dcc4Pc) actually registering RailCom objects, but that really should change.  There are at least 3 other systems that support RailCom, and we eventually need to also ).

The potential issue we have is that when we are trying to identify things by IdTag, we're not currently going to see anything that is registered as a RailCom object, but we should.

We also need to look at how RailCom objects are stored and restored.  They really should be written to the same file as the IdTags, so the Ids are not duplicated between the two managers.

There is a further potential issue down the road, if we ever move LocoNet reporters off of producing strings as the reports, because transponding messages are also identifiable objects.

(The long term vision for this issue is that some day we will have a user who wants to mix identification systems.  The Locomotives already have decoders in them, so if they can identify themselves on the rails, then that solves the identification problem for them.  Rolling stock usually doesn't have a decoder, so you have to add something to it before it can identify itself, and RFID and barcodes are cheap....  but of course the decoders are able to provide more information than the barcodes and RFID tags do...)

pabender, 2016-08-29:
I labelled this as an infrastructure enhancement.  We need to build a proxy manager, much like we have for other bean types, to handle IdTags and derivative types.

rhwood, 2016-08-29:
We had been using "infrastructure" to indicate issues/enhancements to the JMRI development infrastructure (CI services, sites, etc), not to JMRI code itself.

pabender, 2016-08-29:
I've been using infrastructure to indicate code changes that should not be visible to users also.  Mostly this has applied to test related items, but it fits here as well.

rhwood, 2016-08-29:
This is the first code change that shops with JMRI that has been labeled as infrastructure. If the meaning is going to change to indicate JMRI intervals that should not (but given past experience will) be user visible, how should we now label project infrastructure issues to be distinguishable from issues with the JMRI programs?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2016-08-29:
On Mon, Aug 29, 2016 at 2:58 PM, Randall Wood <notifications@github.com>
wrote:

> This is the first code change that shops with JMRI that has been labeled
> as infrastructure. If the meaning is going to change to indicate JMRI
> intervals that should not (but given past experience [...]
In this case, this is infrastructure because it is REQUIRED infrastructure
for an object type.  This is also an enhancement (and was labelled as such)
because it adds new functionality in support of that object type.

We can disagree about what constitutes infrastructure for the project, but
I think this is an infrastructure change, so I'm not changing the label.

pabender, 2019-02-07:
Closed via #6493

closed 2019-02-07 by pabender