JMRI/JMRI #1842: Create test release 4.5.4

bobjacobsen, 2016-08-21:
For discussion of timing and what to include in 4.5.4

bobjacobsen, 2016-09-18:
The release-4.5.4 branch has been created. 

From now on, please document your changes in the [jmri4.5.5.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.5.5.shtml) release note file.

Maintainers, please set the 4.5.5 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.4/)

silverailscolo, 2016-09-19:
Thanks.
In 4.5.4. under English locale on OS X, the OBlocks table doesn’t open (TableSorter). Issue #1980 https://github.com/JMRI/JMRI/issues/1980 <https://github.com/JMRI/JMRI/issues/1980>

Egbert

> Op 18 sep. 2016, om 19:13 heeft Bob Jacobsen <notifications@github.com> het volgende geschreven:
> The release-4.5.4 branch has been created.
>

closed 2016-09-24 by bobjacobsen