JMRI/JMRI #1845: roco.z21 errors that only happen running tests with DEBUG set

bobjacobsen, 2016-08-22:
The following errors are seen when doing `ant ci-test` with DEBUG set, but not when only WARN is set.  

They might be due to e.g. code protected by `if (log.isDebugEnabled())`, but I haven't been able to track that down.  Hopefully somebody familiar with the Roco code will see wha the problem is.

````
    [junit] java.lang.ClassNotFoundException: jmri.jmrix.roco.z21.configurexml.Z21XNetConnectionConfigXml
    [junit] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:372)
    [junit] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
    [junit] 	at java.security.AccessController.doPrivileged(Native Method)
    [junit] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:360)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
    [junit] 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
    [junit] 	at java.lang.Class.forName0(Native Method)
    [junit] 	at java.lang.Class.forName(Class.java:260)
    [junit] 	at jmri.configurexml.ConfigXmlManager.confirmAdapterAvailable(ConfigXmlManager.java:68)
    [junit] 	at jmri.configurexml.ConfigXmlManager.registerUser(ConfigXmlManager.java:162)
    [junit] 	at jmri.implementation.JmriConfigurationManager.registerUser(JmriConfigurationManager.java:85)
    [junit] 	at jmri.jmrix.roco.z21.Z21XPressNetTunnel.setStreamPortController(Z21XPressNetTunnel.java:240)
    [junit] 	at jmri.jmrix.roco.z21.Z21XPressNetTunnel.<init>(Z21XPressNetTunnel.java:61)
    [junit] 	at jmri.jmrix.roco.z21.Z21SystemConnectionMemo.configureManagers(Z21SystemConnectionMemo.java:140)
    [junit] 	at jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapter.configure(Z21SimulatorAdapter.java:62)
    [junit] 	at jmri.jmrix.roco.z21.simulator.Z21SimulatorAdapterTest.testConnection(Z21SimulatorAdapterTest.java:43)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    [junit] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    [junit] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [junit] 	at java.lang.reflect.Method.invoke(Method.java:483)
    [junit] 	at junit.framework.TestCase.runTest(TestCase.java:176)
    [junit] 	at junit.framework.TestCase.runBare(TestCase.java:141)
    [junit] 	at junit.framework.TestResult$1.protect(TestResult.java:122)
    [junit] 	at junit.framework.TestResult.runProtected(TestResult.java:142)
    [junit] 	at junit.framework.TestResult.run(TestResult.java:125)
    [junit] 	at junit.framework.TestCase.run(TestCase.java:129)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.run(JUnitTestRunner.java:523)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.launch(JUnitTestRunner.java:1063)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.main(JUnitTestRunner.java:914)
    [junit] java.lang.ClassNotFoundException: jmri.jmrix.roco.z21.configurexml.Z21XNetTurnoutManagerXml
    [junit] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:372)
    [junit] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
    [junit] 	at java.security.AccessController.doPrivileged(Native Method)
    [junit] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:360)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
    [junit] 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
    [junit] 	at java.lang.Class.forName0(Native Method)
    [junit] 	at java.lang.Class.forName(Class.java:260)
    [junit] 	at jmri.configurexml.ConfigXmlManager.confirmAdapterAvailable(ConfigXmlManager.java:68)
    [junit] 	at jmri.configurexml.ConfigXmlManager.registerConfig(ConfigXmlManager.java:131)
    [junit] 	at jmri.implementation.JmriConfigurationManager.registerConfig(JmriConfigurationManager.java:75)
    [junit] 	at jmri.managers.AbstractManager.registerSelf(AbstractManager.java:43)
    [junit] 	at jmri.managers.AbstractManager.<init>(AbstractManager.java:27)
    [junit] 	at jmri.managers.AbstractTurnoutManager.<init>(AbstractTurnoutManager.java:20)
    [junit] 	at jmri.jmrix.lenz.XNetTurnoutManager.<init>(XNetTurnoutManager.java:24)
    [junit] 	at jmri.jmrix.roco.z21.Z21XNetTurnoutManager.<init>(Z21XNetTurnoutManager.java:21)
    [junit] 	at jmri.jmrix.roco.z21.Z21XNetInitializationManager.init(Z21XNetInitializationManager.java:49)
    [junit] 	at jmri.jmrix.lenz.AbstractXNetInitializationManager.<init>(AbstractXNetInitializationManager.java:51)
[... 63 lines cut ...]
    [junit] 	at junit.framework.TestCase.run(TestCase.java:129)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at junit.framework.TestSuite.runTest(TestSuite.java:252)
    [junit] 	at junit.framework.TestSuite.run(TestSuite.java:247)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.run(JUnitTestRunner.java:523)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.launch(JUnitTestRunner.java:1063)
    [junit] 	at org.apache.tools.ant.taskdefs.optional.junit.JUnitTestRunner.main(JUnitTestRunner.java:914)
    [junit] ERROR - jmri.jmrix.roco.z21.configurexml.Z21XNetConnectionConfigXml could not load adapter class jmri.jmrix.roco.z21.configurexml.Z21XNetConnectionConfigXml [main] jmri.configurexml.ConfigXmlManager.?()
    [junit] ERROR - jmri.jmrix.roco.z21.configurexml.Z21XNetTurnoutManagerXml could not load adapter class jmri.jmrix.roco.z21.configurexml.Z21XNetTurnoutManagerXml [main] jmri.configurexml.ConfigXmlManager.?()
    [junit] ERROR - jmri.jmrix.roco.z21.configurexml.Z21ReporterManagerXml could not load adapter class jmri.jmrix.roco.z21.configurexml.Z21ReporterManagerXml [main] jmri.configurexml.ConfigXmlManager.?()
    [junit] WARN  - getDefault found no default object for type "jmri.RailComManager" [main] jmri.InstanceManager.?()
    [junit] WARN  - getDefault found no default object for type "jmri.UserPreferencesManager" [main] jmri.InstanceManager.?()
    [junit] WARN  - getDefault found no default object for type "jmri.RailComManager" [main] jmri.InstanceManager.?()
````

pabender, 2016-08-22:
On 08/21/2016 09:27 PM, Bob Jacobsen wrote:
> The following errors are seen when doing |ant ci-test| with DEBUG set,
> but not when only WARN is set.
>
> They might be due to e.g. code protected by |if
> (log.isDebugEnabled())|, but I haven't been [...]
|<snip>|
> |[junit] java.lang.ClassNotFoundException: 
> jmri.jmrix.roco.z21.configurexml.Z21XNetTurnoutManagerXml |
|<snip>|
> |[junit] java.lang.ClassNotFoundException: 
> jmri.jmrix.roco.z21.configurexml.Z21ReporterManagerXml|
|<snip>|
> |[junit] ERROR -
> jmri.jmrix.roco.z21.configurexml.Z21XNetConnectionConfigXml could not
> load adapter class
> jmri.jmrix.roco.z21.configurexml.Z21XNetConnectionConfigXml [main] [...]
|Hmmm.. Only one of those three classes (the reporter manager) actually 
existed, and I had mis-named it.  I just created the TurnoutManagerXml 
class.  PR# 1848 should fix these.

The Z21XNetConnectionConfigXml class isn't supposed to exist. That 
particular connection is dependent on the Z21 connection and isn't 
independently configurable.

Now, why those are failing in DEBUG but not without DEBUG, I don't know.

Paul
|

bobjacobsen, 2016-08-22:
Turns out that those only show up in DEBUG because there's a check for whether the persistence (`*Xml`) class exists was only done if DEBUG was set.  And _that_ check was buried down in, instead of being readable at the call level.  #1814 has discussion on the general issue.

bobjacobsen, 2016-08-22:
The Z21XNetConnectionConfigXml error shows up because line 236 in `jmri.jmrix.roco.z21.Z21XPressNetTunnel` registers one at user level:
````
        jmri.InstanceManager.getDefault(jmri.ConfigureManager.class).registerUser(new Z21XNetConnectionConfig(xsc));
````
Registering the object at user level means that the ConfigureXml system sets up to store and load it, which trips the error about a missing class.  That error will also happen if you `store` after that, so it's not just at DEBUG time.  That should perhaps be a `registerPref(..)` instead.

bobjacobsen, 2016-08-22:
#1850 clarifies the code that's posting these errors, but doesn't change the behavior. They're still there in DEBUG (because they'd still be there if somebody did a "store" operation from a running system).  I haven't yet tried the changes in #1848, but it sounds like those will fix some of them.

pabender, 2016-08-22:
PR #1851 changes the registrer on Z21XPressNetTunnel to registerPref.

pabender, 2016-08-23:
#1856 seems to have fixed this.

closed 2016-08-25 by bobjacobsen