JMRI/JMRI #1865: Bundle classes to look at

pabender, 2016-08-24:
I just pushed up PR #1864 which makes sure that most of the current *Bundle.java files have tests associated with them.

There are still 6 *Bundle classes that need to be looked at (out of 161 total...)

@bobjacobsen do you know why we have both jmri.jmrix.loconet.bundle and jmri.jmrix.loconet.LocoNetBundle?  Both classes reference the same file, and the LocoNetBundle version doesn't follow the design pattern.  It looks like this is a historic artefact we can probably eliminate.

We also have a whole bunch (8 total) bundles in the jmri.jmrix.mrc package (and it's subpackages).  5 of those don't have tests, yet, because it looks like they are something that may be able to be eliminated or combined into a smaller number of bundles (see discussion on PR #1861 ).  The 3 Bundle.java files in the package are fine, its the Mrc*Bundle.java files we need to think about.

rhwood, 2016-08-24:
What classes use the LocoNetBundle? It may be in place for a class outside the Loconet package tree to access content in the Loconet properties file.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2016-08-24:
> On Aug 24, 2016, at 11:44 AM, Paul Bender <notifications@github.com> wrote:
> 
> do you know why we have both jmri.jmrix.loconet.bundle and jmri.jmrix.loconet.LocoNetBundle? Both classes reference the same file, and the LocoNetBundle version doesn't follow the design pattern. It looks like this is a historic artefact we can probably eliminate.

Yes, I think it can be merged into jmrix.loconet.Bundle class & .properties files.

It seems to exist as an easy way to have multiple sub-packages reference one bundle, but that’s what Bundle itself is meant to do.

So a straight-forward “LocoNetBundle” -> “Bundle” replacement, along with a merge of .properties files, should do it.

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

pabender, 2016-08-25:
I'm working on the Loconet changes now.  

Removing LocoNetBundle only caused 41 compile errors, most in sub-packages.

KenC57, 2016-08-25:
Bob, Paul,

 

So should I work on consolidating the MRC stuff??

 

As I see it, there should just be the one Bundle.java (per directory) and
all the files in that directory uses that file. And that file is using a
single properties, which means the existing files consolidate into one
properties file.

 

Or is it more than that??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-08-25:
That is basically it @KenC57 .

KenC57, 2016-08-25:
I see the _da and _fr files are identical, no translations, so I figure
those get deleted. Same for the other properties parts after rolled into the
main one.

 

Paul, I had 96 errors after taking out the partial files (simple rename for
now) and it should take a while for the rest to get worked over. I'm
figuring sometime tomorrow to finish the edits.

 

One thing I'm not sure of is how to delete the files I won't need when it
comes time commit/push them back to the master. But that's another day.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-08-25:
PR #1867 takes care of the LocoNet portion of this (and adds tests, because I had to add more Bundle classes).

@KenC57 Once you merge all the files, I would leave the merged _da and _fr files, even if they are blank,  just in case someone adds translations later.

as for deleting files, I typically use `git rm` on the command line...

KenC57, 2016-08-25:
Paul,

 

My issue on the _da and _fr files is they aren't blank but just the same
pairs as in the English version. Would it be correct to delete those
duplicates so if anybody did translate it later they would know to grab the
current contents of the English properties. But deleting the split files
that we are merging into one, those get the delete.

-ken c

pabender, 2016-08-25:
@KenC57,

In that case, yes, go ahead and delete them.

If they had actually been translated to different text from the English, then I would leave them.

silverailscolo, 2016-08-25:
Duplicated values from the default .properties indeed add nothing in terms of localization.
As a way to work on a translation, some users started with a complete copy of the default and just edit that file, replacing string after string. 
But as no work is being done, and it complicates your job te merge several Bundles, go ahead.
I suggest to leave the (almost) blank files themselves, but strip the lines that are just copies.

Egbert Broerse

> Op 25 aug. 2016, om 05:35 heeft Ken Cameron <notifications@github.com> het volgende geschreven:
> 
> Paul,
> My issue on the _da and _fr files is they aren't blank but just the same
> pairs as in the English version. Would it be correct to delete those
> duplicates so if anybody did translate it later they would know to grab the
> current contents of the English properties. But deleting the split files
> that we are merging into one, those get the delete.
> 
> -ken c

closed 2016-11-13 by pabender