JMRI/JMRI #1871: JMRI cannot run on Fedora 24

rhwood, 2016-08-25:
The RXTX package included in JMRI fails to function on Fedora 24 due to [the fix for a latent security issue](https://bugzilla.redhat.com/show_bug.cgi?id=581884).

Working around this requires removing __all__ instances of ```librxtxSerial.so``` in ```JMRI_HOME/lib/linux/*/```, before creating a symlink to the Fedora provided RXTX libraries.

Fixes for this include:
1. Updating the launcher so that users can set the Java property ```java.library.path``` and having the JMRI launcher append (not prepend) its library paths to the user-provided property. (Prepending the JMRI paths would defeat the purpose of this change -- if a user wants to append their paths to that setting, they can including the JMRI paths in their provided property.)
2. Documenting how to use jmri.conf to setup the JMRI launchers to work with the Fedora provided RXTX, and possibly providing a script to do this for the user.

Do want to also allow the JMRI Posix Launchers to read configurations from something like /etc/jmri/jmri.conf to provide a system-wide configuration that can then be further modified by the individual user's jmri.conf files?

rhwood, 2016-08-25:
I think that #936 also resolves this issue, but that appears to have other problems not yet resolved.

pabender, 2016-08-25:
Randall,

On 08/25/2016 08:53 AM, Randall Wood wrote:
>
> The RXTX package included in JMRI fails to function on Fedora 24 due 
> to the fix for a latent security issue 
> <https://bugzilla.redhat.com/show_bug.cgi?id=581884>.
>

That bug report you linked to was fixed in Fedora 14, and closed in 2010.

> Working around this requires removing *all* instances of 
> |librxtxSerial.so| in |JMRI_HOME/lib/linux/*/|, before creating a 
> symlink to the Fedora provided RXTX libraries.
>

As I mentioned on the users list, there  an alternative to that:
On 08/24/2016 09:11 PM, Paul Bender wrote:
 > On 08/24/2016 07:38 PM, ste_vg@yahoo.ca [jmriusers] wrote:
 >> Sorry. Yes the librxtxSerial.so reference in the JMRI Linux libraries
 >> has to replaced with a symlink to librxtxSerial.so in the operating
 >> system libraries.
 >
 > There is another way to fix this.
 >
 > create a file named "/etc/tmpfiles.d/00rxtx.conf" and put the following
 > contents in it:
 >
 > <start file contents>
 > # create the lock directory if it doesn't exist
 > d /run/lock 0775 root uucp -
 > # if it does, set the permissions
 > z /run/lock 0775 root uucp -
 > <end file contents>
 >
 > For historic reasons, I use uucp on my own system, but you can change
 > that to another group, such as lock, if desired.
 >
 > The advantage this has over copying the librxtxSerial.so is that when
 > you upgrade your JMRI installation, you don't have to remember to
 > replace the file.

Does that not work for you?

> Fixes for this include:
> 1. Updating the launcher so that users can set the Java property
> |java.library.path| and having the JMRI launcher append (not prepend)
> its library paths to the [...]

3. Document how to properly setup the /var/lock directory. and (if 
necessary) SELinux to allow JMRI to access the lock files.

> Do want to also allow the JMRI Posix Launchers to read configurations 
> from something like /etc/jmri/jmri.conf to provide a system-wide 
> configuration that can then be further modified by the individual 
> user's jmri.conf files?
>

That would be ok, but I'm not convinced it is required yet.

Paul

rhwood, 2016-08-25:
> > The RXTX package included in JMRI fails to function on Fedora 24 due 
> > to the [fix for a latent security issue](https://bugzilla.redhat.com/show_bug.cgi?id=581884).
> 
> That bug report you linked to was fixed in Fedora 14, and closed in 2010.
> 
I know. This issue for us goes back a few years.

> Working around this requires removing *all* instances of 
> |librxtxSerial.so| in |JMRI_HOME/lib/linux/*/|, before creating a 
> symlink to the Fedora provided RXTX libraries.
>

>  >> Sorry. Yes the librxtxSerial.so reference in the JMRI Linux libraries
>  >> has to replaced with a symlink to librxtxSerial.so in the operating
>  >> system libraries.

This did not work. It appears that java.library.path property is the roots of searchable paths, not a series of paths (at least in OpenJDK 1.8.0_101 on Fedora 24), so adding a new path before the JMRI lib path in the property it works, but just replacing the one .so file that matches the running architecture does not.

>  > There is another way to fix this.
>  >
>  > create a file named "/etc/tmpfiles.d/00rxtx.conf" and put the following
>  > contents in it:
>  >
```
>  > # create the lock directory if it doesn't exist
>  > d /run/lock 0775 root uucp -
>  > # if it does, set the permissions
>  > z /run/lock 0775 root uucp -
```

I do not want to open a documented security hole in the OS if I can avoid it. We may not know of real exploits, but that is no indication there aren't any and that they are not actively used.

>  > The advantage this has over copying the librxtxSerial.so is that when
>  > you upgrade your JMRI installation, you don't have to remember to
>  > replace the file.

I'm not proposing any copying over of any files in this fix, but I am also proposing a fix that does not lessen the security posture of the system as shipped.

> > Fixes for this include:
> 3. Document how to properly setup the /var/lock directory. and (if 
> necessary) SELinux to allow JMRI to access the lock files.
 
I would rather document a configuration change to JMRI than a security posture change to the underlying OS.

pabender, 2016-08-25:
On 08/25/2016 09:39 AM, Randall Wood wrote:
> The RXTX package included in JMRI fails to function on Fedora
>         24 due
>         to the fix for a latent security issue
>         <https://bugzilla.redhat.com/show_bug.cgi?id=581884>.
>
>     That [...]

Perhaps, but we've been working with it just fine since then, so why the 
sudden rush to fix it by telling users to remove files and replace them?

> The advantage this has over copying the librxtxSerial.so is
>         that when
>         you upgrade your JMRI installation, you don't have to remember to
>         replace the file.
>
> I'm not proposing [...]

It does require replacing files that we distribute, and that step has to 
be performed every single time a user updates JMRI.

This also doesn't help those of us who run from development 
environments, because if we aren't careful, we might over-write the file 
in git.

>         Fixes for this include:
>         3. Document how to properly setup the /var/lock directory. and
>         (if
>         necessary) SELinux to allow JMRI to access the lock files.
>
> I would rather document a configuration change to JMRI than a security 
> posture change to the underlying OS.
>
We can disagree about the process, but there are always tradeoffs in any 
solution to a problem.

I would personally rather have a systems administration process for 
solving distribution specific idiosyncrasies than asking users to modify 
our distributed files.  This is a one time fix, not something you have 
to fix on a regular basis.

If PR #936 does solve the issue, without the possible security change, 
then we should move forward with that.

Paul

rhwood, 2016-08-25:
@pabender You're not understanding the issue. The issue is that we currently have to either modify the system or the JMRI installation to get JMRI working on Fedora systems. The proposed fix eliminates both these changes, unless you are proposing that the ```settings:``` directory (```~/.jmri``` in Linux) should now be part of the JMRI installation.

We ship a ```jmri.conf``` file in ```program:```, but it existing in ```program:``` only to copy into ```settings:``` and modify, and is deliberately not read from ```program:```. I am proposing to enable maintaining the necessary configuration in ```settings:jmri.conf``` and to document how to use that configuration. This means that no changes to the system or JMRI installation except for udev rules would be required or desired.

pabender, 2016-08-25:
On Aug 25, 2016, at 11:22 AM, Randall Wood <notifications@github.com> wrote:
> @pabender You're not understanding the issue. The issue is that we currently have to either modify the system or the JMRI installation to get JMRI working on Fedora systems. The proposed fix [...]
That was not the way I read your proposed solution this morning.  Thanks for clarifying.
> We ship a jmri.conf file in program:, but it existing in program: only to copy into settings: and modify, and is deliberately not read from program:. I am proposing to enable maintaining the [...]
One issue I have is this really feels lie a band aid to me when we have other issues with RxTx that have a wider impact than just Fedora.  I really think it is time we move forward with the move to PJC.

Paul

bobjacobsen, 2017-07-03:
Can we close this Issue?  It seems RXTX-specific.

rhwood, 2017-07-03:
Yes, we can close.

closed 2017-07-03 by rhwood