JMRI/JMRI #1872: Tests timeout on appveyor.

rhwood, 2016-08-25:
We are getting a number of test failures due to timeouts on AppVeyor. These failures are most likely caused by a GUI test failing and leaving a modal window or dialog open and being killed when the build timeout is exceeded.

The most likely cause of the GUI failure is the size of the available headless desktops. (Comments on the AppVeyor forums [suggest these default to a very small size](http://help.appveyor.com/discussions/problems/3906-selenium-testing-screen-size) like 480 x 640).

What size should these be made to be? The same as Travis CI?

bobjacobsen, 2016-08-25:
> The most likely cause of the GUI failure is the size of the available headless desktops. (Comments on the AppVeyor forums suggest these default to a very small size like 480 x 640).
> 
> What size should these be made to be? The same as Travis CI?

The same as Travis CI sounds like a reasonable choice. How big is that?

We have language on a web page that "a display with a resolution of 800x600 or larger is strongly recommended” (http://jmri.org/install/WindowsNew.shtml).  Even that seems very small to me.

rhwood, 2016-08-26:
PR #1877 sets these to 1024 x 786 - the width of an older 4x3 VGA display (and the width of the Travis CI tests) and the height of a currently shipping inexpensive Dell laptop display (shorter than the Travis CI tests).

rhwood, 2016-08-26:
Since PR #1877 has been merged, we need to monitor to see if there is a perceived reduction in test timeouts.

rhwood, 2016-09-03:
Test timeouts appear to have been mitigated.

closed 2016-09-03 by rhwood