JMRI/JMRI #1874: Unable to open Tables window in master.

rhwood, 2016-08-26:
ant -f /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI decoderpro
init:
jjtree:
javacc:
copyfiles:
update-template-code:
Copying 1 file to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/java/tmp
compile-generated-source:
Copying 1 file to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/java/classes/jmri
debug:
Compiling 1 source file to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/java/classes
Note: /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/java/src/jmri/swing/JmriJTablePersistenceManager.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
decoderpro:
runtime-library-selection:
arch.lib.path /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/lib/macosx/x86_64:/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/lib/macosx
Launch normally (no debugger support)
2016-08-25 21:30:59,788 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-08-25 21:31:00,895 util.Log4JUtil                        INFO  - This log is appended to file: /Users/rhwood/Library/Preferences/JMRI-dev/log/messages.log [main]
2016-08-25 21:31:00,898 util.Log4JUtil                        INFO  - This log is stored in file: /Users/rhwood/Library/Preferences/JMRI-dev/log/session.log [main]
2016-08-25 21:31:00,909 apps.AppsBase                         INFO  - DecoderPro version 4.5.4ish-201608260130-rhwood-Rbe452ae starts under Java 1.8.0_92 on Mac OS X x86_64 v10.11.6 at Thu Aug 25 21:31:00 EDT 2016 [main]
2016-08-25 21:31:03,775 gui3.Apps3                            INFO  - Starting with profile NCE_Simulator.3edd8a2f [main]
2016-08-25 21:31:04,324 node.NodeIdentity                     INFO  - Using jmri-9801A7A1BBE1-3edd8a2f as the JMRI Node identity [main]
2016-08-25 21:31:04,573 simulator.SimulatorAdapter            INFO  - NCE Simulator Started [Nce Simulator]
2016-08-25 21:31:04,633 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.1 [AWT-EventQueue-0]
2016-08-25 21:31:05,261 swing.JmriJTablePersistenceManager    INFO  - Table preferences not found.
This is expected on the first time the "NCE Simulator" profile is used on this computer. [main]
2016-08-25 21:31:05,262 util.FileUtilSupport                  INFO  - File path program: is /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/ [main]
2016-08-25 21:31:05,263 util.FileUtilSupport                  INFO  - File path preference: is /Users/rhwood/Dropbox/JMRI/Profiles/NCE_Simulator/ [main]
2016-08-25 21:31:05,263 util.FileUtilSupport                  INFO  - File path profile: is /Users/rhwood/Dropbox/JMRI/Profiles/NCE_Simulator/ [main]
2016-08-25 21:31:05,264 util.FileUtilSupport                  INFO  - File path settings: is /Users/rhwood/Library/Preferences/JMRI-dev/ [main]
2016-08-25 21:31:05,264 util.FileUtilSupport                  INFO  - File path home: is /Users/rhwood/ [main]
2016-08-25 21:31:05,266 util.FileUtilSupport                  INFO  - File path scripts: is /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/jython/ [main]
2016-08-25 21:31:06,311 swing.RosterFrame                     ERROR - Unable to get the double click speed, Using JMRI default of half a secondjava.lang.NullPointerException [main]
2016-08-25 21:31:16.601 java[34232:2577883] 21:31:16.601 WARNING:  140: This application, or a library it uses, is using the deprecated Carbon Component Manager for hosting Audio Units. Support for this will be removed in a future release. Also, this makes the host incompatible with version 3 audio units. Please transition to the API's in AudioComponent.h.
2016-08-25 21:31:16,604 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2016-08-25 21:31:16,639 jmri.InstanceManager                  WARN  - getDefault found no default object for type "jmri.RailComManager" [Listed Table Generation]
2016-08-25 21:31:16,640 beantable.ListedTableFrame            ERROR - Error when adding jmri.jmrit.beantable.RailComTableAction to display
java.lang.NullPointerException [Listed Table Generation]
java.lang.NullPointerException
	at jmri.jmrit.beantable.RailComTableAction$1.getManager(RailComTableAction.java:90)
	at jmri.jmrit.beantable.BeanTableDataModel.<init>(BeanTableDataModel.java:66)
	at jmri.jmrit.beantable.RailComTableAction$1.<init>(RailComTableAction.java:56)
	at jmri.jmrit.beantable.RailComTableAction.createModel(RailComTableAction.java:56)
	at jmri.jmrit.beantable.AbstractTableAction.getTableDataModel(AbstractTableAction.java:81)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.createDataModel(ListedTableFrame.java:372)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.<init>(ListedTableFrame.java:365)
	at jmri.jmrit.beantable.ListedTableFrame.initComponents(ListedTableFrame.java:112)
	at jmri.jmrit.beantable.ListedTableAction$1.run(ListedTableAction.java:66)
	at java.lang.Thread.run(Thread.java:745)

rhwood, 2016-08-26:
Sorry. Hit submit too soon.

This needs #1873 to be merged to be reproducible without other errors.

pabender, 2016-08-26:
I was already working on this.  #1875 fixes this.

rhwood, 2016-08-26:
Fixed by #1875.

closed 2016-08-26 by rhwood