JMRI/JMRI #1880: Audio subsystem hangs JMRI when 'Open Tables' is used

KenC57, 2016-08-27:
Trying to open the tables triggers the audio subsystem and on Roberto's system, that is hanging. A process jogamp* is launched and it prevents populating the tables. Killing that process allows the 'open tables' to complete normally. Other work around is opening specific tables and not the audio table.

This is a subsection of the JOAL support we are using. It seems to match known issues in some versions of that support. Also somebody said the version we are using is getting dated and needs other work, but I've not verified that.

When somebody is able to address this, Roberto should be contacted via the JMRI users group for testing as he is the only user who had run into very consistent failure of this.

mattharris, 2016-10-07:
@KenC57 I should probably take a closer look at this

KenC57, 2016-12-16:
I just did a little debuging and in ListedTableFrame:addPanelModel, it takes a very long time after it gets to the audio table.

KenC57, 2017-06-06:
We just had a case where a user had been working fine then something in his PC changed (like Windows updates). It now takes about an hour for his panel to load. If he runs it without all the audiobuffer statements,  it loads in seconds. See user group message "XML File Load Locks Up on 4.7.4 today, but not yesterday", Randy is the OP.

I still have the case on my system of it taking about 30-60 seconds for JOAL to startup.

bobjacobsen, 2017-06-06:
Is there a way to get a thread dump of where JMRI is spending its time? 

On Mac and Linux, you "send a SIGQUIT to the process".  Yes, that's a bit geeky, but it's pretty easy if running from a terminal: Just hit ctrl-\.  When e.g. running from an icon, it's a bit more complicated, but not that much.

Is there a Windows equivalent?

KenC57, 2017-06-06:
Using Window/Eclipse, cntl-I didn't do anything on Windows. This is using
the current HEAD. Here is thread stack before opening tables:

 

apps.PanelPro.PanelPro at localhost:58808            

               Thread [AWT-EventQueue-0] (Running)    

               Daemon Thread [LocoNetThrottledTransmitter] (Running)

               Daemon Thread [LocoNet transmit handler] (Running)       

               Daemon Thread [LocoNet receive handler] (Running)         

               Thread [Thread-9] (Running)         

               Thread [DestroyJavaVM] (Running)            

 

Here is stack just after opening window but before it populates. Seems the
empty window does get opened, it then take a minute for the contents to show
up. I did a 'suspend' so I could capture things:

PanelPro [Java Application]          

               apps.PanelPro.PanelPro at localhost:58808 (Suspended)    

                              Daemon System Thread [Attach Listener]
(Suspended)       

                              Daemon System Thread [Signal Dispatcher]
(Suspended)    

                              Daemon System Thread [Finalizer] (Suspended)

                              Daemon System Thread [Reference Handler]
(Suspended) 

                              Daemon System Thread [Java2D Disposer]
(Suspended)     

                              Daemon System Thread [AWT-Windows] (Suspended)

                              System Thread [AWT-Shutdown] (Suspended)

                              Thread [AWT-EventQueue-0] (Suspended)

                              Daemon System Thread [TimerQueue] (Suspended)

                              Daemon System Thread [Swing-Shell] (Suspended)

                              Daemon Thread [LocoNetThrottledTransmitter]
(Suspended)          

                              Daemon Thread [LocoNet transmit handler]
(Suspended)  

                              Daemon Thread [LocoNet receive handler]
(Suspended)    

                              Thread [Thread-9] (Suspended)    

                              Thread [DestroyJavaVM] (Suspended)       

                              Thread [Listed Table Generation] (Suspended)

               C:\Program Files\Java\jdk1.8.0_121\bin\javaw.exe (Jun 6,
2017, 10:49:04 AM)        
[... 252 lines cut ...]
               AudioTableAction.lambda$0(AudioManager) line: 122        

               153043339.accept(Object) line: not available         

               Optional<T>.ifPresent(Consumer<? super T>) line: 159       

               AudioTableAction.createModel() line: 120              

               AudioTableAction.getPanel() line: 137       

               ListedTableFrame$TabbedTableItem.addPanelModel() line: 420

               ListedTableFrame$TabbedTableItem.<init>(String, String,
boolean) line: 377             

               ListedTableAction$1$1(ListedTableFrame).initComponents()
line: 117          

               ListedTableAction$1.run() line: 68              

               Thread.run() line: 745      

 

Now letting it run to completion shows:

bobjacobsen, 2017-06-06:
(Trimming for length)

It’s pretty clear that the hang is well inside the JoalAudioFactory (not JMRI code) initialization invoked from DefaultAudioManager (JMRI code).  

It’s possible that we’re not doing some setup or configuration before making that call.  Could be some thing in the code, could be something in the external setup.  But the delay is in that startup.

So there are a couple possibilities:

A) Find why this is happening and fix it.  It seems Windows-specific, but maybe not JMRI-specific so perhaps engaged other users of JOAL audio on Windows know what’s going on.  Does Google show anything?

B) Modify JMRI to defer sound initialization as long as possible.  This might be a good idea anyway. Not trivial, but just some work.  But it won’t help people using sound, like the original source of this particular example.

C) Give JMRI the ability to use some other sound library, if only when JOAL is incredibly slow to start.  There’s some vestigial code for this but I don’t know the status. This solves the problem for people with awful-slow-JOAL who can use another library, but I think JOAL has certain features that other people may will want to access.

Perhaps a combination of these?  Different people may be able to help with different ones. I can work on (B) eventually, but (A) is outside my area.

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

klk32003, 2017-06-09:
> It’s pretty clear that the hang is well inside the JoalAudioFactory (not JMRI code) initialization invoked from DefaultAudioManager (JMRI code).

I did a test on Windows 7 with an environment variable (name: ALSOFT_LOGLEVEL, value: 3), please see log.txt. At JMRI startup I use Perform Action Open Audio Table force Joal initialization.

The log shows the Joal initialization steps. Might be interesting to log the initialization steps on a Windows machine with a long initialization delay. I have a delay of some seconds on my system.

Another point is the file soft_oal.dll. In my case the file is located in lib\windows\x86. JMRI uses version 1.15.1, current version is 1.18.0.
It might be worth to try a newer version of soft_oal.dll for a test, e.g. from here: https://github.com/sgothel/joal/tree/master/make/lib or here: http://openal-soft.org/.

Klaus Killinger
[log.txt](https://github.com/JMRI/JMRI/files/1064031/log.txt)

bobjacobsen, 2017-06-09:
@klk32003 Thanks! Didn't know about that `ALSOFT_LOGLEVEL` option, I think it's going to help if we can get somebody to reproduce the extremely slow startup.  

The attached log is from startup on my Mac.  It's essentially instantaneous. (The delay shown between the last `INFO  - File path scripts:` line and the `INFO  - Initialised JOAL` line is mostly me navigating to the Audio menu item to open the table)

[log-mac.txt](https://github.com/JMRI/JMRI/files/1064159/log-mac.txt)

KenC57, 2017-06-09:
Here is the result from the ALSOFT_LOGLEVEL 3 entry. I'm still having about 1 to 2 minutes to startup the audio table. So what's the next set of options??

AL lib: (II) alc_initconfig: Supported backends: mmdevapi, dsound, winmm, null, wave
AL lib: (II) GetConfigValue: Key general:disable-cpu-exts not found
AL lib: (II) FillCPUCaps: Detected max CPUID function: 0xd (ext. 0x80000008)
AL lib: (II) FillCPUCaps: Vendor ID: ""
AL lib: (II) FillCPUCaps: Name: "      Intel(R) Core(TM) i7-3632QM CPU @ 2.20GHz"
AL lib: (II) FillCPUCaps: Got caps: SSE
AL lib: (II) GetConfigValue: Key general:rt-prio not found
AL lib: (II) GetConfigValue: Key general:resampler not found
AL lib: (II) GetConfigValue: Key general:trap-al-error not found
AL lib: (II) GetConfigValue: Key general:trap-alc-error not found
AL lib: (II) GetConfigValue: Key reverb:boost not found
AL lib: (II) GetConfigValue: Key reverb:emulate-eax not found
AL lib: (II) GetConfigValue: Key general:drivers not found
AL lib: (II) MMDevApiMsgProc: Starting message thread
AL lib: (II) MMDevApiMsgProc: Starting message loop
AL lib: (II) alc_initconfig: Initialized backend "mmdevapi"
AL lib: (II) alc_initconfig: Added "mmdevapi" for playback
AL lib: (II) alc_initconfig: Initialized backend "dsound"
AL lib: (II) alc_initconfig: Added "dsound" for capture
AL lib: (II) GetConfigValue: Key general:excludefx not found
AL lib: (II) GetConfigValue: Key general:default-reverb not found
AL lib: (II) GetConfigValue: Key general:channels not found
AL lib: (II) GetConfigValue: Key general:sample-type not found
AL lib: (II) GetConfigValue: Key general:format not found
AL lib: (II) GetConfigValue: Key general:frequency not found
AL lib: (II) GetConfigValue: Key general:periods not found
AL lib: (II) GetConfigValue: Key general:period_size not found
AL lib: (II) GetConfigValue: Key general:sources not found
AL lib: (II) GetConfigValue: Key general:slots not found
AL lib: (II) GetConfigValue: Key general:sends not found
AL lib: (II) GetConfigValue: Key general:cf_level not found
AL lib: (II) MMDevApiMsgProc: Got message 1024
AL lib: (II) alcOpenDevice: Created device 0000000018123C20, "Speakers (Realtek High Definition Audio)"
AL lib: (II) UpdateDeviceParams: Pre-reset: Stereo, Float, 44100hz, 1024 update size x4
AL lib: (II) MMDevApiMsgProc: Got message 1025
AL lib: (II) UpdateDeviceParams: Post-reset: Stereo, Float, 48000hz, 960 update size x4
AL lib: (II) GetConfigValue: Key general:layout_stereo not found
AL lib: (II) GetConfigValue: Key general:layout not found
AL lib: (II) GetConfigValue: Key general:hrtf not found
AL lib: (II) UpdateDeviceParams: HRTF disabled
AL lib: (II) UpdateDeviceParams: BS2B disabled
AL lib: (II) GetConfigValue: Key general:wide-stereo not found
AL lib: (II) MMDevApiMsgProc: Got message 1026
AL lib: (II) alcCreateContext: Created context 000000001811D700
2017-06-09 12:32:50,489 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]

bobjacobsen, 2017-06-09:
> Here is the result from the ALSOFT_LOGLEVEL 3 entry. I'm still having about 1 to 2 minutes to startup the audio table. So what's the next set of options??

Does it stall somewhere specific in that trace?  Maybe that’d be a hint.

It’s becoming more and more clear that this is an interaction between Windows and the underlying audio libraries.  Some Googling by a Windows-literate person might come up with somebody who’s encountered and fixed this on some other project.

I don’t know enough Windows to know what’s relevant and what’s not, but a quick look found:

http://www.overclock.net/t/903497/opening-playback-devices-is-slow

http://www.tomshardware.com/answers/id-1665804/sound-property-long-time-open.html

https://www.ableton.com/answers/live-9-0-4-slow-at-startup

KenC57, 2017-06-09:
The whole log message showed up as one right when the table finally was
populating.

-ken c

bobjacobsen, 2017-06-09:
> The whole log message showed up as one right when the table finally was
populating.

That's consistent with what the earlier thread traces were showing. Those were hung in a initconfig step.

Next, it would be good to update the Windows DLL components to see if that helps. (See @klk32003 post a couple above)

bobjacobsen, 2017-06-09:
Download location for OpenAL: http://openal.org/downloads/

Another clue about things that might matter for Windows (where there can be lots of different device support software bits lying around):
http://openal.org/pipermail/openal/2014-October/000239.html

klk32003, 2017-06-09:
> Download location for OpenAL: http://openal.org/downloads/

Handling the binaries is a little bit tricky. Please see the readme.txt.

[readme.txt](https://github.com/JMRI/JMRI/files/1064963/readme.txt)

KenC57, 2017-06-10:
Dropping the soft_oal.dll for 1.18 to replace the 1.15 didn't make any change. Also used the ini gui in the download to edit the ini and no change, still near 2 minutes to load.

klk32003, 2017-06-10:
> Dropping the soft_oal.dll for 1.18 to replace the 1.15 didn't make any change. Also used the ini gui in the download to edit the ini and no change, still near 2 minutes to load.

Thanks for running all these tests. Your audio device seams to work well.

I found this yesterday: https://github.com/kcat/openal-soft/blob/master/docs/env-vars.txt ... about  ```ALSOFT_LOGLEVEL 4```: 
> Same as 3, but also device and context reference count changes. This will print out *a lot* of info, and is generally not useful unless you're trying to track a reference leak within the library.

Not sure if this addresses timing issues.

klk32003, 2017-06-10:
Had a look at this case in https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/138954.
I downloaded Ontario Southern Railway.xml and added some files to avoid log messages (there is still an error due to a missing Jython script).
Finally I could open the panels manually or per "Open File" startup setting.
No issues.

[logcase.txt](https://github.com/JMRI/JMRI/files/1065714/logcase.txt)

KenC57, 2017-06-10:
I tried looking at the stack again. One thing I see is below the joal.jar is gluegen-rt.jar which is (I think) where this thread is spending time waiting. And it looks like the 4 for the log setting did increase the lines of the log, but still shows as a single entry at completion of startup of the tables.
I am using the newer dll but this is still the original joal.jar file. 

Thread [Listed Table Generation] (Suspended)	
	owns: Class<T> (com.jogamp.common.util.IOUtil) (id=71)	
	owns: PrintStream  (id=87)	
	owns: Class<T> (com.jogamp.common.util.cache.TempJarCache) (id=73)	
	owns: Class<T> (com.jogamp.openal.util.ALut) (id=78)	
	owns: DefaultAudioManager  (id=88)	
	ProcessImpl.waitForInterruptibly(long) line: not available [native method]	
	ProcessImpl.waitFor() line: 449	
	IOUtil.testDirExec(File) line: 909	
	IOUtil.testDirImpl(File, boolean, boolean, String) line: 939	
	IOUtil.getSubTempDir(File, String, boolean, String) line: 999	
	IOUtil.getTempDir(boolean) line: 1114	
	TempFileCache.<clinit>() line: 80 [local variables unavailable]	
	TempJarCache.initSingleton() line: 88 [local variables unavailable]	
	Platform$1.run() line: 309 [local variables unavailable]	
	AccessController.doPrivileged(PrivilegedAction<T>) line: not available [native method]	
	Platform.<clinit>() line: 287 [local variables unavailable]	
	ALFactory.<clinit>() line: 68 [local variables unavailable]	
	ALut.alutInit() line: 64 [local variables unavailable]	
	JoalAudioFactory.init() line: 217	
	DefaultAudioManager.init() line: 140	
	AudioTableAction.lambda$0(AudioManager) line: 122	
	1214925788.accept(Object) line: not available	
	Optional<T>.ifPresent(Consumer<? super T>) line: 159	
	AudioTableAction.createModel() line: 120	
	AudioTableAction.getPanel() line: 137	
	ListedTableFrame$TabbedTableItem.addPanelModel() line: 420	
	ListedTableFrame$TabbedTableItem.<init>(String, String, boolean) line: 377	
	ListedTableAction$1$1(ListedTableFrame).initComponents() line: 117	
	ListedTableAction$1.run() line: 68	
	Thread.run() line: 745

klk32003, 2017-06-10:
> I tried looking at the stack again. One thing I see is below the joal.jar is gluegen-rt.jar which is (I think) where this thread is spending time waiting. And it looks like the 4 for the log setting [...]
I am using the newer dll but this is still the original joal.jar file.

```joal.jar``` is an own-built JMRI version with modifications to correct the load of WAV files with appended metadata. Together with ```gluegen-rt.jar```, ```joal.dll``` and ```gluegen-rt.dll``` this is the JOAL package. JMRI ```joal.jar``` is the newest version. Last offical JOAL was released in October 2015 at http://jogamp.org/  (JOAL is a module of jogamp).
```soft_oal.dll``` is a futher development of OpenAL and is based on JOAL. A newer version than 1.15.1 should be working too.

The Thread Stack may help to narrow down the extreme latency. Will have a look on it.

klk32003, 2017-06-11:
Interesting response today from Roberto in https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/139079

So I think one part of the issue seams to be solved. Please see the opening comment from Ken.

But this doesn't concern the long loading time issue.

KenC57, 2017-06-12:
I was checking with 4.7.5, still hanging for a couple of minutes. But I read something about the jogamp process that seems to run during this period and that killing it helped a similar case with the audio stuff hanging. When I did that, I got completion right away, but with a stack dump. Maybe this is another clue:

Warning: Caught Exception while retrieving executable temp base directory:
java.io.IOException: Could not determine a temporary executable directory
	at com.jogamp.common.util.IOUtil.getTempDir(IOUtil.java:1168)
	at com.jogamp.common.util.cache.TempFileCache.<clinit>(TempFileCache.java:80)
	at com.jogamp.common.util.cache.TempJarCache.initSingleton(TempJarCache.java:88)
	at com.jogamp.common.os.Platform$1.run(Platform.java:309)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.common.os.Platform.<clinit>(Platform.java:287)
	at com.jogamp.openal.ALFactory.<clinit>(ALFactory.java:68)
	at com.jogamp.openal.util.ALut.alutInit(ALut.java:64)
	at jmri.jmrit.audio.JoalAudioFactory.init(JoalAudioFactory.java:217)
	at jmri.jmrit.audio.DefaultAudioManager.init(DefaultAudioManager.java:140)
	at jmri.jmrit.beantable.AudioTableAction.lambda$0(AudioTableAction.java:122)
	at java.util.Optional.ifPresent(Unknown Source)
	at jmri.jmrit.beantable.AudioTableAction.createModel(AudioTableAction.java:120)
	at jmri.jmrit.beantable.AudioTableAction.getPanel(AudioTableAction.java:137)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.addPanelModel(ListedTableFrame.java:420)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.<init>(ListedTableFrame.java:377)
	at jmri.jmrit.beantable.ListedTableFrame.initComponents(ListedTableFrame.java:117)
	at jmri.jmrit.beantable.ListedTableAction$1.run(ListedTableAction.java:68)
	at java.lang.Thread.run(Unknown Source)
2017-06-11 21:58:09,120 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]

klk32003, 2017-06-12:
OK, back to zero. I thought that jogamp-process issue was solved. At least it is solved for Roberto, but it's still present.

Had to re-read https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/130784, which led to this current issue.

Some background on that jogamp-process:
http://forum.jogamp.org/Why-is-IOUtil-testDirExec-needed-td4035542.html

For short: The jogamp-developers are aware of this problem. But it is not solved. Could be a virus scanner issue, after all we are talking about an executable.

Former reaction by jogamp-developers to solve it:
https://jogamp.org/bugzilla/show_bug.cgi?id=1231 - title "Sporadic failure to end jogamp_exe_tst while determine 'executable temp directory' on Windows"

At least those links may answer "A) Find why this is happening" from a comment above.

klk32003, 2017-06-13:
> For short: The jogamp-developers are aware of this problem. But it is not solved.

I think I misinterpreted the bug list. If I look at the closed bugs, there are ID 1219 and ID 1231, both with status RESOLVED and with resolution FIXED.
See https://jogamp.org/bugzilla/buglist.cgi?bug_status=__closed__&component=core&product=Gluegen.

The last comment, which irritated me (https://jogamp.org/bugzilla/show_bug.cgi?id=1219#c73), was about a "false positive".

KenC57, 2017-06-13:
My suspect is now the gluegen-rt.jar as where I'm seeing the trace hanging longer. But I'm not sure about which version we are running right now. Looking at the wiki it seems the 2.0 is what they have listed now.

rhwood, 2017-06-13:
Where is the source code for the special build made for JMRI and have those changes been contributed back to the upstream project?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-06-13:
The `lib/README.md` file says:

> NOTE: joal.jar is currently replaced by an own-built version with modifications to correct the load of WAV files with appended metadata - see [GitHub PR](https://github.com/sgothel/joal/pull/15) for details of modifications.

Unfortunately, the sgothel/joal project doesn't seem to be being maintained:  The two PR's proposed since 2015 haven't been merged.

klk32003, 2017-06-13:
I couldn't find a source in JMRI. Might be it's in a fork of sgothel/joal.

bobjacobsen, 2017-06-13:
The source is in the `bug_1280` branch of the forked repo in the `mattharris` account.  See the [PR](https://github.com/sgothel/joal/pull/15) that @mattharris sent back to the original project or look at  [this commit](https://github.com/sgothel/joal/pull/15/commits)

klk32003, 2017-06-14:
> My suspect is now the gluegen-rt.jar as where I'm seeing the trace hanging longer. But I'm not sure about which version we are running right now. Looking at the wiki it seems the 2.0 is what they have listed now.

Yes, I think it has to do with ```gluegen-rt.jar```. Failure appears sporadic. I'm aware of minimum 4 cases. One case headed to a [bug](https://jogamp.org/bugzilla/show_bug.cgi?id=1231) report, fixed and [released](http://jogamp.org/deployment/archive/rc/v2.3.2/jar/) on 10-Oct-2015.

Could someone please tell me the creation date of ```JMRI/lib/gluegen-rt.jar```? I couldn't find a way to detect it.

bobjacobsen, 2017-06-14:
> Could someone please tell me the creation date of JMRI/lib/gluegen-rt.jar? I couldn't find a way to detect it.

`jar tv` shows "Fri Mar 27 16:12:38 PDT 2015".

Manifest contents are:
```
Manifest-Version: 1.0
Ant-Version: Apache Ant 1.9.4
Created-By: 1.8.0_25-b17 (Oracle Corporation)
Application-Name: GlueGen Run-Time
Specification-Title: GlueGen Java Bindings Generator
Specification-Version: 2.3
Specification-Vendor: JogAmp Community
Implementation-Title: GlueGen Run-Time
Implementation-Version: 2.3.1
Implementation-Build: 2.3-b857-20150327
Implementation-Branch: origin/master
Implementation-Commit: f1eba171ddd887a2daa3dfa2ac90ac864c5b22d4
Implementation-Vendor: JogAmp Community
Implementation-Vendor-Id: com.jogamp
Implementation-URL: http://jogamp.org/
Extension-Name: com.jogamp.common
Trusted-Library: true
Permissions: all-permissions
Application-Library-Allowable-Codebase: *
Codebase: *.jogamp.org

Name: jogamp/common/
Sealed: true

Name: com/jogamp/common/
Sealed: true
```

klk32003, 2017-06-14:
@bobjacobsen Thanks!
@KenC57 There is a chance, that a test with a [**newer**](http://jogamp.org/deployment/archive/rc/v2.3.2/jar/) ```gluegen-rt.jar``` gives a clue. I was able to open a reporter table and run VSDecoder on Windows 7 and XP. Could you please give it a try?

klk32003, 2017-06-19:
I've found a way to run a Joal-test on Windows.

To setup the test:

1. download and extract http://jogamp.org/deployment/v2.3.2/archive/jogamp-all-platforms.7z
2. copy ```jogamp-all-platforms\joal-java-src.zip``` and ```jogamp-all-platforms\jar\joal-test.jar``` into local JMRI ```lib``` directory
3. expand Windows Path with the OS specific path, e.g. ```C:\...\lib\windows\x86```
4. open Windows command-line, navigate to JMRI ```lib``` directory and run: ```java -Djogamp.debug=true -cp gluegen-rt.jar;joal.jar;joal-test.jar com.jogamp.openal.test.manual.OpenALTest >> joal-test.log 2>&1```

Test results will be available in ```...\lib\joal-test.log```. See attached example.

Remark on last error in ```joal-test.log```: since ```joal.jar``` is JMRI-specific, it fails. Using ```jogamp-all-platforms\jar\joal.jar``` will play an audio (at least in my case).

[joal-test.log.txt](https://github.com/JMRI/JMRI/files/1084670/joal-test.log.txt)

klk32003, 2017-07-13:
Ongoing discussion in [jmriusers](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/139797).

One user did a test with a newer gluegen-rt.jar (which should cover a ```resolved fixed``` bug), but without success.

KenC57, 2017-07-25:
I went back to normal jars and then added this to the startup options in the Eclipse run/debug:

-Djogamp.gluegen.UseTempJarCache=false
This cured the problem of slow table opening. I also copy/created the jmri.conf for normal executables to include:

default_options="-J-Djogamp.gluegen.UseTempJarCache=false"

And all the normal JMRI programs now work too with very quick table opens. So this helps to confirm which issue within the audio seems to be the slow spot.

rhwood, 2017-08-17:
Duplicate of #2732

rhwood, 2017-08-17:
Closing as a duplicate of another issue.

closed 2017-08-17 by rhwood