JMRI/JMRI #1906: Warrant code throws error when running headless

bobjacobsen, 2016-09-01:
JmriFaceless on RPi throws error from `jmri.jmrit.logix.WarrantRoute` (but keeps running) 

Traceback from an email thread:

```
pi@J-20160307-6:~ $ /home/pi/JMRI/JmriFaceless &
[1] 3181
pi@J-20160307-6:~ $ 2016-08-31 21:56:55,629 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2016-08-31 21:56:55,781 util.Log4JUtil INFO - This log is appended to file: /home/pi/.jmri/log/messages.log [main]
2016-08-31 21:56:55,786 util.Log4JUtil INFO - This log is stored in file: /home/pi/.jmri/log/session.log [main]
2016-08-31 21:56:55,818 apps.AppsBase INFO - JmriFaceless version 4.4-R1ccf76b starts under Java 1.8.0_65 on Linux arm v4.1.18-v7+ at Wed Aug 31 21:56:55 CDT 2016 [main]
2016-08-31 21:56:56,615 apps.AppsBase INFO - Starting with profile SW__Faceless.3eb5c7d8 [main]
2016-08-31 21:56:56,975 node.NodeIdentity INFO - Using jmri-0013EF203762-3eb5c7d8 as the JMRI Node identity [main]
2016-08-31 21:56:58,116 hexfile.LnHexFilePort INFO - LocoNet Simulator Started [Thread-6]
2016-08-31 21:57:00,103 util.FileUtilSupport INFO - File path program: is /home/pi/JMRI/ [main]
2016-08-31 21:57:00,107 util.FileUtilSupport INFO - File path preference: is /home/pi/Documents/JMRI/SW/ [main]
2016-08-31 21:57:00,110 util.FileUtilSupport INFO - File path profile: is /home/pi/.jmri/SW__Faceless/ [main]
2016-08-31 21:57:00,113 util.FileUtilSupport INFO - File path settings: is /home/pi/.jmri/ [main]
2016-08-31 21:57:00,116 util.FileUtilSupport INFO - File path home: is /home/pi/ [main]
2016-08-31 21:57:00,119 util.FileUtilSupport INFO - File path scripts: is /home/pi/Documents/JMRI/SW/scripts/ [main]
2016-08-31 21:57:05,078 apps.AppsBase ERROR - java.awt.HeadlessException [Thread-12]
java.awt.HeadlessException
at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204)
at java.awt.Window.<init>(Window.java:536)
at java.awt.Frame.<init>(Frame.java:420)
at java.awt.Frame.<init>(Frame.java:385)
at javax.swing.JFrame.<init>(JFrame.java:189)
at jmri.util.JmriJFrame.<init>(JmriJFrame.java:82)
at jmri.jmrit.logix.WarrantRoute.<init>(WarrantRoute.java:78)
at jmri.jmrit.logix.NXFrame.<init>(NXFrame.java:87)
at jmri.jmrit.logix.NXFrame.getInstance(NXFrame.java:75)
at jmri.jmrit.logix.WarrantPreferences.setNXFrame(WarrantPreferences.java:313)
at jmri.jmrit.logix.WarrantPreferences.apply(WarrantPreferences.java:310)
at jmri.jmrit.logix.WarrantPreferencesPanel.setGUI(WarrantPreferencesPanel.java:94)
at jmri.jmrit.logix.WarrantPreferencesPanel.<init>(WarrantPreferencesPanel.java:66)
at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
at java.lang.Class.newInstance(Class.java:442)
at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:173)
at apps.AppsBase$1.run(AppsBase.java:118)
at java.lang.Thread.run(Thread.java:745)
2016-08-31 21:57:07,536 withrottle.FacelessServer INFO - Published ZeroConf service for 'sw._withrottle._tcp.local.' on 10.0.0.100:12090 [WiThrottleFacelessServer]
2016-08-31 21:57:10,299 withrottle.FacelessServer INFO - Published ZeroConf service for 'sw._withrottle._tcp.local.' on 10.0.0.10:12090 [WiThrottleFacelessServer]
2016-08-31 21:57:10,301 withrottle.FacelessServer INFO - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleFacelessServer]
```

petecressman, 2016-09-26:
peteC-headlessWarrant#1906 has a proposed fix, but i cannot test this in a headless environment

petecressman, 2016-10-02:
PR #2003 should fix this.  How can Preferences run headless?

bobjacobsen, 2016-10-10:
PR #2003 from @petecressman was meant to fix this.  Could somebody try development package 881 or later from http://jmri.tagadab.com/jenkins/job/Development/job/Packages/ please? Thanks.

closed 2016-10-23 by bobjacobsen

petecressman, 2016-10-24:
Installed.  Ran a few LocoNet simulated warrants.   Modified and saved several Preferences values.  Looks OK. @bobjacobsen

petecressman, 2016-10-25:
Installed onto win 10

| JMRI.4.5.6ish-201610240455-jenkins-Rd899e67.exe

 |

Raun a few LocoNet simulated warrants.   Modified and saved several Preferences values.  Looks OK.
PeteC 

 

 Closed #1906.—