JMRI/JMRI #1917: rxtx issue on Raspberry Pi 3 Jessie

fiz1962, 2016-09-03:
I got a new Raspberry PI 3 to run JMRI with a Digitrax PR3 interface.  I installed JMRI 4.4 (and also tried 4.5.3).  I added my user to the dialout and uucp groups.  I created a new profile and tried both /dev/ttyAMA0 and /dev/ttyACM0 ports.  The first complains about bad ioctl but the other seems to open initially but then just as jmri finishes initializing it prints an error about the lock file existing.  Then I can't toggle the power and can't do anything beyond that.  Output pasted below.

====================
2016-09-03 14:12:54,780 apps.Apps                             INFO  - PanelPro version 4.4-R1ccf76b starts under Java 1.8.0_65 on Linux arm v4.4.13-v7+ at Sat Sep 03 14:12:54 CDT 2016 [main]
2016-09-03 14:13:01,691 apps.Apps                             INFO  - Starting with profile AlleghenySierra.3f5dde98 [main]
2016-09-03 14:13:01,930 node.NodeIdentity                     INFO  - Using jmri-B827EB0A2F78-3f5dde98 as the JMRI Node identity [main]
2016-09-03 14:13:03,124 locobuffer.LocoBufferAdapter          INFO  - /dev/ttyACM0 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-09-03 14:13:06,483 util.FileUtilSupport                  INFO  - File path program: is /home/pi/Desktop/JMRI/ [main]
2016-09-03 14:13:06,488 util.FileUtilSupport                  INFO  - File path preference: is /home/pi/.jmri/AlleghenySierra/ [main]
2016-09-03 14:13:06,492 util.FileUtilSupport                  INFO  - File path profile: is /home/pi/.jmri/AlleghenySierra/ [main]
2016-09-03 14:13:06,497 util.FileUtilSupport                  INFO  - File path settings: is /home/pi/.jmri/ [main]
2016-09-03 14:13:06,501 util.FileUtilSupport                  INFO  - File path home: is /home/pi/ [main]
2016-09-03 14:13:06,505 util.FileUtilSupport                  INFO  - File path scripts: is /home/pi/Desktop/JMRI/jython/ [main]
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File existsRXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists

pabender, 2016-09-03:
On Sep 3, 2016, at 5:55 PM, fiz1962 <notifications@github.com> wrote:
> I got a new Raspberry PI 3 to run JMRI with a Digitrax PR3 interface. I installed JMRI 4.4 (and also tried 4.5.3). I added my user to the dialout and uucp groups. I created a new profile and tried [...]
/dev/AMA0 on the Raspberry Pi 3 is connected ( by default ) to the Bluetooth device.  You can change that with an overlay but that doesn't have anything to do with your problem, since /dev/ttyAMA0 will never be assigned to a USB device.
> RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File existsRXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
> 

Please give us the output of
 
'ls -l /var/lock/LCK..ttyACM'

Even if you have all the permissions set correctly, If the device lock was created by another user or the port is actually in use, you may not be able to remove ( and recreate ) the lock file as the user you are logged in as.

Paul

fiz1962, 2016-09-03:
It does leave /var/lock/LCK..ttyACM0 behind.  And I can remove it and still get the same error about it existing. If I run PanelPro and the file already exists it does say it is removing a stale lock file and then proceeds as in the original post.

ls -all /var/lock/LCK*

gives

-r--r--r-- 1 pi pi 11 Sep 3 17:27 /var/lock/LCK..ttyACM0

and if I run PanelPro now, in addition I see a line in the output like

RXTX Warning: Removing stale lock file /var/lock/LCK..ttyACM0

followed by the line saying the port was opened.  If I delete the lock file, I don't get the line about removing the stale lock file but have the same behavior.

fiz1962, 2016-09-05:
And I see the same problem on an old Macbook Pro running FC24 and JMRI 4.2

rsion 4.2.1-R6b484b4 starts under Java 1.8.0_92 at Mon Sep 05 13:40:21 CDT 2016 [main]
2016-09-05 13:40:24,801 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3f5875cb [main]
2016-09-05 13:40:25,092 node.NodeIdentity                     INFO  - Using jmri-001B6304ECDD-3f5875cb as the JMRI Node identity [main]
RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyACM0
2016-09-05 13:40:25,526 locobuffer.LocoBufferAdapter          INFO  - /dev/ttyACM0 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-09-05 13:40:27,475 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2016-09-05 13:40:27,805 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=3.0} [WebServer]
2016-09-05 13:40:29,585 util.FileUtilSupport                  INFO  - File path program: is /home/fiz/Desktop/JMRI/ [main]
2016-09-05 13:40:29,586 util.FileUtilSupport                  INFO  - File path preference: is /home/fiz/.jmri/My_JMRI_Railroad/ [main]
2016-09-05 13:40:29,587 util.FileUtilSupport                  INFO  - File path profile: is /home/fiz/.jmri/My_JMRI_Railroad/ [main]
2016-09-05 13:40:29,587 util.FileUtilSupport                  INFO  - File path settings: is /home/fiz/.jmri/ [main]
2016-09-05 13:40:29,588 util.FileUtilSupport                  INFO  - File path home: is /home/fiz/ [main]
2016-09-05 13:40:29,588 util.FileUtilSupport                  INFO  - File path scripts: is /home/fiz/Desktop/JMRI/jython/ [main]
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
2016-09-05 13:40:37,545 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 5959, waiting for incoming connection... [WiThrottleGUIServer]

fiz1962, 2016-09-11:
Here's the latest progress I've made.

* Fresh install of Ubuntu 16.04 on a thumb drive
* Fresh download of JMRI 4.4
* Add user ubuntu to group dialout, log out, log back in
* Install java (sudo apt-get install default-jre), installs openjdk
* Run PanelPro and create Digitrax PR3 connection with /dev/ttyACM0 (in black text at this point)
* Save connection and restart PanelPro
* Get "RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists" error twice as PanelPro starts

And now, get the above error message each time I select "Edit/Preferences" menu from main PanelPro screen and select "/dev/ttyACM0" (now in red) drop down list on the LocoNet connection tab and select "Save".  If I do not click on the Serial Port drop down list, I don't get the error.

pabender, 2016-09-11:
On Sep 11, 2016, at 2:12 PM, fiz1962 <notifications@github.com> wrote:

> Here's the latest progress I've made.
>
> Fresh install of Ubuntu 16.04 on a thumb drive
> Fresh download of JMRI 4.4
> Add user ubuntu to group dialout, log out, log back in
> Install java (sudo apt-get [...]
What happens in you try to do something like turn on the track power at this point?
> Get "RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists" error twice as PanelPro starts
> And now, get the above error message each time I select "Edit/Preferences" menu [...]
I think you're focusing too much on this error.

What is causing this when you open the preferences is that we request from RxTx a list of serial ports that are available.  As part of that check, RxTx verifies which of the serial ports on the system are in use.

Once you select /dev/ttyACM0, it will appear in the selection list for two basic reasons: 1) there is some kind of communication problem that prevents RxTx from talking to the port ( e.g.  port no longer exists ( you unplugged it, or there was some other failure )) or 2) the port is currently in use by some program ( including the currently running JMRI instance ).  

Basically, the red text indicates the port is currently selected, but RxTx did not report it as an available port.

Paul

pabender, 2016-09-27:
@fiz1962,

Can you try one of the alpha builds that uses Pure Java Comm instead of RXTX to see if your issue goes away?

You can download that here:

http://jmri.tagadab.com/jenkins/job/TestReleases/job/alpha-builds/job/rxtx-to-pjc/

Paul

fiz1962, 2016-09-27:
The problem does seem to have gone away.  It seems like I have more trouble with the command station than anything.  I'm learning that when I have problems, clearing the command station memory, clearing the command station slots, etc seems to do it.  One day I may write a script to automatically do these things each time we power up.

closed 2016-09-27 by fiz1962