JMRI/JMRI #1947: A number of tests fail when run individually.

rhwood, 2016-09-10:
A number of tests fail or have errors when run individually. See the attached logs from running
```
rm ./test_results.txt ; for jmri_test in $( find java/test -name \*Test.java ! -name AllTest.java ! -name HeadLessTest.java ! -name PackageTest.java ! -name FailTest.java ) ; do jmri_test=${jmri_test#java/test/} ; jmri_test=${jmri_test%\.java} ; echo ${jmri_test} >> ./test_results.txt ; ./runtest.csh ${jmri_test} >> ./test_results.txt 2>&1 ; done
```

Similar results can be generated by running ```mvn test``` after installing [Apache Maven](http://maven.apache.org).

Of note, one test class (jmri.implementation.swing.SwingShutDownTaskTest) had a test waiting endlessly on two dialogs to be dismissed, and one test class (jmri.jmrit.vsdecoder.
VSDecoderTest) had a test that needed to be killed with Ctrl-C.

[test_results.txt](https://github.com/JMRI/JMRI/files/465052/test_results.txt)

pabender, 2016-09-10:
Some of those are pretty subtle issues.

Thanks for pointing those out.

bobjacobsen, 2016-09-11:
This one:
```
jmri/jmrit/consisttool/ConsistToolActionTest
JUnit version 4.12
.log4j:WARN No appenders could be found for logger (jmri.InstanceManager).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
```
and a couple similar ones nearby are interesting.  That usually means that `apps.tests.Log4JFixture.setUp();` hasn't been invoked.  That typically happens in the (earlier JUnit) `protected void setUp()` method.  This test is in Junit 4 format and has:

```
    @Before
    public void setUp() {
        JUnitUtil.resetInstanceManager();
        Log4JFixture.setUp();
        InstanceManager.setDefault(ConsistManager.class, new TestConsistManager());
    }
```

Does anybody see anything wrong with that?  Looks like it should work to me, but the error message indicates it's not working.

pabender, 2016-09-11:
Sent from my iPad

On Sep 10, 2016, at 10:07 PM, Bob Jacobsen <notifications@github.com> wrote:

> This one:
>
> jmri/jmrit/consisttool/ConsistToolActionTest
> JUnit version 4.12
> .log4j:WARN No appenders could be found for logger (jmri.InstanceManager).
> log4j:WARN Please initialize the log4j system [...]
Looking at this now, I think this is a statement ordering problem.

Note that the log4j warning is coming from the instance manager, and not any part of the 
consisting tool, so the first two lines need to be swapped. ( I can't verify that at the moment. )

Paul

pabender, 2016-09-12:
On 09/11/2016 12:37 AM, Paul Bender wrote:
> Looking at this now, I think this is a statement ordering problem.
>
> Note that the log4j warning is coming from the instance manager, and not
> any part of the
> consisting tool, so the first two lines need to be swapped. ( I can't
> verify that at the moment. )
It looks like I was correct.

PR#1955 includes a correction.

Paul

pabender, 2016-09-13:
Just looking through the jmri.jmrix tree, I think the only failing tests at this point are the Abstract manager tests.  These are all testing the abstract classes, and the intention is that tests for derived classes will include those. 

I generally find that inheritance in tests doesn't work the way it should, with tests not necessarilly being run, even though they were inherited from a base class.  (maybe this works better with JUnit4 formated tests, but the tests in question are not....)

Paul

bobjacobsen, 2016-09-14:
> Just looking through the jmri.jmrix tree, I think the only failing tests at this point are the Abstract manager tests. These are all testing the abstract classes, and the intention is that tests for derived classes will include those.

Right. I think it's an artifact of the test shell script that they were being run. Normally they don't, because the abstract class doesn't create a TestSuite for them.

> I generally find that inheritance in tests doesn't work the way it should, with tests not necessarilly being run, even though they were inherited from a base class. (maybe this works better with JUnit4 formated tests, but the tests in question are not....)

The Turnout, Sensor, etc tests use an abstract base to reduce code duplication and increase coverage.  Are any of those not running right?

pabender, 2016-09-14:
On 09/13/2016 11:39 PM, Bob Jacobsen wrote:
> I generally find that inheritance in tests doesn't work the way it
>     should, with tests not necessarilly being run, even though they
>     were inherited from a base class. (maybe this works better [...]

It isn't that the tests don't run.

The problem is when you have an overridden method in a derived class and 
your test class is using inheritance,  the wrong method is sometimes run.

I found that out with all those Bundle tests I added at the beginning of 
the summer.  The tests all ran with the derived classes, but the 
coverage report showed the derived classes were not tested at all.

With methods that are only in the Abstract class, there shouldn't be any 
problem.

Paul

pabender, 2016-09-14:
One there note on those Abstract*MgrTest classes...

From what I can tell, It looks like we barely use them.  
The Turnout version is used extensively.  
The Sensor and Reporter Manager versions are not used at all currently.  
The Light Manager version is currently used by 3 classes.

I think I'll go ahead and convert all of them but the Turnout manager version to JUnit 4 format, along with any tests that use those.  I'll save the Turnout version for later, because it is a bigger project.

bobjacobsen, 2016-09-18:
Would it make sense to rename the abstract test classes to reduce confusion?  E.g. `AbstractTurnoutTest` isn't really a test class that should be run (or be able to run) by itself.  How about renaming it to `AbstractTurnoutTestBase`, leaving the pattern that `*Test` means 'a test class to be run'.

pabender, 2016-09-18:
On Sep 18, 2016, at 12:35 PM, Bob Jacobsen <notifications@github.com> wrote:
> Would it make sense to rename the abstract test classes to reduce confusion? E.g. AbstractTurnoutTest isn't really a test class that should be run (or be able to run) by itself. How about renaming it to AbstractTurnoutTestBase, leaving the pattern that *Test means 'a test class to be run'.
> 
After thinking about this ( and getting some of the manager tests working with the abstract classes that previously did not) I think it is actually correct as it is.  

The thought here is that the class tests the concrete functions of an abstract class I.e. it included tests, but, like the abstract class it covers, it won't function by itself.

Paul

bobjacobsen, 2016-09-18:
> The thought here is that the class tests the concrete functions of an abstract class I.e. it included tests, but, like the abstract class it covers, it won't function by itself.

I think our naming convention conflates two different things:

- Abstract(FooTest) - an abstract base class for a family of (test the Foo class) classes
- (AbstractFoo)Test - a single class to test the concrete methods of the AbstractFoo class

My earlier comment was referring the 1st, and you're thinking of the 2nd.

I was focusing on the 1st because it _also_ does the 2nd automatically. All those concrete test methods are invoked when a subclass is run.  For example, the `AbstractTurnoutMgrTest#testCreate()` test method is run when the `jmri.jmrix.loconet.LnTurnoutManagerTest` class is run.

So I think we should do one of two things:
- Rename the existing abstract `AbstractFooTest` classes to e.g. `AbstractFooTestBase`, to preserve the convention that `*Test` classes can be run as test.
- Alternately, if we want those abstract test classes to be runnable to test the abstract base class (case 2), we should update those classes to actually run properly. (Not quite sure how to do that, willing to look into it, but I prefer the previous solution)

pabender, 2016-09-26:
PR #1997 finishes the work of converting the existing tests based on the Abstract*MgrTest classes to the JUnit4 format.

There may still be some applicable manager tests that don't extend these.

Paul

rhwood, 2016-10-07:
Note that the tests in __jmri.implementation.swing.SwingShutDownTaskTest__ deliberately wait on user input when run using ```./runtest.csh jmri.implementation.swing.SwingShutDownTaskTest```, but do not do so when run using ```ant test-single -Dtest.includes=jmri.implementation.swing.SwingShutDownTaskTest``` or ```mvn surefire:test -Dtest=jmri.implementation.swing.SwingShutDownTaskTest```

rhwood, 2016-11-27:
The latest [test_results.txt](https://github.com/JMRI/JMRI/files/614955/test_results.txt)

Interestingly enough, I think this ~200 more tests than are run in ant alltest.

bobjacobsen, 2017-01-22:
PR #2912 improves this, by renaming all the classes that are true abstract bases, not intended to be run independently, into *TestBase. 

PR #2914 improves output when running this way, but doesn't fix any failures.

Issue #2915 is for the DuplicateTest, which fails when run individually and isn't in the corresponding PackageTest

bobjacobsen, 2017-01-23:
Most recent log from running this is attached.
[test_results.txt](https://github.com/JMRI/JMRI/files/725213/test_results.txt)

bobjacobsen, 2017-01-24:
#2924 does a bit more work on this.

bobjacobsen, 2017-07-03:
Another update on this. Run with `master` today, there were three failures:
- `jmri/jmrit/ussctc/StationTest`, which is new, had several failures. I'll fix that.
- `jmri/jmrit/decoderdefn/DuplicateTest` found duplicate models in decoder definition. This is #2455 redux
- `jmri.jmrit.audio.AudioCommandTest` threw a `java.lang.ExceptionInInitializerError`. I've embedded that part of the log below, as I don't know what caused that and perhaps somebody else does.

The full log is also attached.

In addition, `jmri/jmplementation/swing/SwingShutDownTaskTest` opened a window and waited.  See:
[image]

After clicking "Do Something and Stop", it presented a similar window. Clicking on that dismissed it and the run continued.

AudioCommandTest log snippet:
```
jmri/jmrit/audio/AudioCommandTest
JUnit version 4.12
.Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
E
Time: 2.136
There was 1 failure:
1) testCTor(jmri.jmrit.audio.AudioCommandTest)
java.lang.ExceptionInInitializerError
	at jmri.jmrit.audio.AudioCommandTest.testCTor(AudioCommandTest.java:19)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runners.Suite.runChild(Suite.java:128)
	at org.junit.runners.Suite.runChild(Suite.java:27)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:115)
	at org.junit.runner.JUnitCore.runMain(JUnitCore.java:77)
	at org.junit.runner.JUnitCore.main(JUnitCore.java:36)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:27)
Caused by: java.lang.NullPointerException
	at jmri.jmrit.audio.AbstractAudioSource.<clinit>(AbstractAudioSource.java:66)
	... 33 more

FAILURES!!!
Tests run: 1,  Failures: 1
```
[test_results.txt](https://github.com/JMRI/JMRI/files/1120041/test_results.txt)

rhwood, 2017-07-03:
> jmri/jmrit/decoderdefn/DuplicateTest found duplicate models in decoder definition. This is #2455 redux

Note that the test *is* (still) failing correctly, as the underlying failure has not been fixed. See #2601

dheap, 2017-07-03:
@rhwood 
> jmri/jmrit/decoderdefn/DuplicateTest found duplicate models in decoder definition. This is #2455 redux
> Note that the test is (still) failing correctly, as the underlying failure has not been fixed. See #2601

This has just made its way back to the top of my list and I will hopefully be working on it today.

rhwood, 2017-10-06:
@bobjacobsen Can the test process in this bug to ensure that unit tests can be run individually be run daily on Jenkins. #4214 includes db9939a which was found incidentally while fixing a persistent but intermittent failure in a LayoutEditor test, which could not be tested individually, but only if some other specific test (I don't know which) had been run previously.

bobjacobsen, 2017-10-08:
I created a Jenkins [Separate_Tests](http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/) job to run this daily.  

Jenkins shows the most recent job as failing. The [console log](http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/lastBuild/console) shows it ending in the `jmri/jmrit/display/layoutEditor/LayoutEditorWindowTest` test. The [test_results.txt](http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/ws/test_results.txt) file shows why, see below.  So I think this is working as a test job, if perhaps not doing very user-friendly reporting.

```
jmri/jmrit/display/layoutEditor/LayoutEditorWindowTest
./runtest.csh: 75: ./runtest.csh: [[: not found
JUnit version 4.12
.Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'useDefaultTrackColour' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'darkGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'grey' sent to stringToColor [main] jmri.util.ColorUtil.?()
ERROR - unknown color text 'lightGrey' sent to stringToColor [main] jmri.util.ColorUtil.?()
E
Time: 3.266
There was 1 failure:
1) testShowAndClose(jmri.jmrit.display.layoutEditor.LayoutEditorWindowTest)
java.lang.AssertionError: Unexpected ERROR or FATAL messages emitted
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.assertTrue(Assert.java:41)
	at org.junit.Assert.assertFalse(Assert.java:64)
	at apps.tests.Log4JFixture.tearDown(Log4JFixture.java:38)
	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:118)
	at jmri.jmrit.display.layoutEditor.LayoutEditorWindowTest.tearDown(LayoutEditorWindowTest.java:47)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[... cut ...]
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:115)
	at org.junit.runner.JUnitCore.runMain(JUnitCore.java:77)
	at org.junit.runner.JUnitCore.main(JUnitCore.java:36)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:27)

FAILURES!!!
Tests run: 1,  Failures: 1
```

silverailscolo, 2017-10-08:
This test failing could be the result of it running on British English as default Locale, under which ``grey`` is spelled as ``gray``.

bobjacobsen, 2017-10-08:
> This test failing could be the result of it running on British English as default Locale, under which grey is spelled as gray.

Good catch!  That Jenkins server is in the UK.  Will try to figure out how to reset the default locale via the test arguments.

rhwood, 2017-10-08:
That Jenkins server is headless isn't it?

bobjacobsen, 2017-10-08:
> That Jenkins server is headless isn't it?

It has no physical screen, but it starts Xvfb:

```
Xvfb starting$ /usr/bin/Xvfb :0 -fbdir /var/lib/jenkins/xvfb-8-9088030887392661367.fbdir
```

The most recent run made it to a known-failing test, further than before:

```
+ jmri_test=jmri/jmrit/decoderdefn/DuplicateTest.java
+ jmri_test=jmri/jmrit/decoderdefn/DuplicateTest
+ echo jmri/jmrit/decoderdefn/DuplicateTest
+ ./runtest.csh jmri/jmrit/decoderdefn/DuplicateTest
```

```
jmri/jmrit/decoderdefn/DuplicateTest
./runtest.csh: 75: ./runtest.csh: [[: not found
JUnit version 4.12
.Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
found duplicate for TCS DP2X decoder][DP2X
found duplicate for TCS DP2X decoder][DP2X
found duplicate for TCS DP2X decoder][DP2X-UK
found duplicate for TCS DP2X decoder][DP2X-UK
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://javax.xml.XMLConstants/property/accessExternalDTD
Warning:  org.jdom2.transform.JDOMSource$DocumentReader: http://www.oracle.com/xml/jaxp/properties/entityExpansionLimit
found duplicate for TCS Mx series decoders][M1
checked total of 2246
E
Time: 23.469
There was 1 failure:
1) testForDuplicateModels(jmri.jmrit.decoderdefn.DuplicateTest)
java.lang.AssertionError: test failed, see System.err
	at org.junit.Assert.fail(Assert.java:88)
	at jmri.jmrit.decoderdefn.DuplicateTest.testForDuplicateModels(DuplicateTest.java:39)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runners.Suite.runChild(Suite.java:128)
	at org.junit.runners.Suite.runChild(Suite.java:27)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:115)
	at org.junit.runner.JUnitCore.runMain(JUnitCore.java:77)
	at org.junit.runner.JUnitCore.main(JUnitCore.java:36)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:27)

```

bobjacobsen, 2017-10-08:
The failing test is related to Issue #2915 and #2601

dheap, 2017-10-08:
> The failing test is related to Issue #2915 and #2601
Which I'm working on again now.

dheap, 2017-11-05:
#4372 resolves the `DuplicateTest` issue and also re-enables the `DuplicateTest` in `pom.xml`.

Is there anything else outstanding or can this be closed now?

bobjacobsen, 2017-11-05:
I've started the Jenkins job at http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/  If that works, we can close this.  But let's wait to see whether something else is uncovered.

dheap, 2017-11-05:
It appears that there is something else:

[Separate_Tests/30/consoleText](http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/30/consoleText)

bobjacobsen, 2017-11-05:
The Jenkins job gets past DuplicateTest now, and fails on
```
There was 1 failure:
1) testCTor(jmri.jmrit.audio.AudioCommandTest)
java.lang.ExceptionInInitializerError
	at jmri.jmrit.audio.AudioCommandTest.testCTor(AudioCommandTest.java:17)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
...
Caused by: java.lang.NullPointerException
	at jmri.jmrit.audio.AbstractAudioSource.<clinit>(AbstractAudioSource.java:66)
	... 33 more
```

bobjacobsen, 2017-11-11:
The Jenkins job is now running to completion! So we're getting a complete set of tests run individually every morning (US time)

closed 2018-01-03 by bobjacobsen