JMRI/JMRI #1972: Operations - Add Route/Cancel NPE

silverailscolo, 2016-09-17:
Cancelling the "Add Route" pane in Operations (current master) by clicking the close box causes an NPE (on OS X 10.11.6) as the empty Route name is processed. The pane stays open, effectively freezing JMRI. There's no Cancel button, so I assume this is a common method for users to try. Clicking the "Add Route" button politely asks to enter a name, but that does not restore control.
The Ops "Add Train" pane handles a similar situation without a hitch.

DanielBoudreau, 2016-09-17:
Thanks Egbert,

You found a bug that was recently introduced with the new TableRowSorter and persisting.  The table wasn’t initialized until the user entered a name for the route and pressed “Add”. Now fixed.

Dan

DanielBoudreau, 2016-09-18:
Commit #1974 Fixes NPE when canceling/closing "Add Route" frame

closed 2016-09-18 by DanielBoudreau