JMRI/JMRI #1978: startup scripts does not work when called by link

pischky, 2016-09-19:
Sorry just reported this on sf.net (http://jmri.org/help/en/html/doc/Technical/index.shtml still points to sf.net)

The startup scripts use `dirname `to get the installation directory. This does not work when called by link. To fix readlink should be used:

The scripts should use:
`SCRIPTDIR=$(cd "$( dirname "$( readlink -e "${0}" )" )" && pwd)`
instead of
`SCRIPTDIR=$(cd "$( dirname "${0}" )" && pwd)`

To reproduce install in `/opt`. Use `sudo ln -s /opt/JMRI/DecoderPro /usr/local/bin/decoderpro` and start by `decoderpro`.

rhwood, 2016-09-19:
The proposed fix breaks these scripts on macOS, although it implements readlink, it does not use GNU readlink.

The startup scripts are intended to be able to run on any POSIX system that supports bash 3, and the readlink command is not specified in POSIX, so there may be other operating systems that do not have it (nor is it a requirement that a Linux distribution include it, so there is no guarentee that readlink is available on a Linux system either).

BTW, is there a reason, you cannot run DecoderPro directly out of /opt/JMRI or add /opt/JMRI to your PATH environment variable?

pabender, 2016-09-19:
On Sep 19, 2016, at 7:18 AM, pischky <notifications@github.com> wrote:
> Sorry just reported this on sf.net (http://jmri.org/help/en/html/doc/Technical/index.shtml still points to sf.net)
> 
We get things there as well.
> The startup scripts use dirnameto get the installation directory. This does not work when called by link. To fix readlink should be used:
>
> The scripts should use:
> SCRIPTDIR=$(cd "$( dirname "$( [...]
In addition to Randall's comments, is there a reason you didn't create the symbolic link with the correct name.  In other words, why didn't you run:

ln -s /opt/JMRI/DecoderPro /usr/local/bin/DecoderPro
> 

That "${0}" tells the shell we want the name of the 0th argument to the program, which is the program name.

Since Linux is case sensitive, DecoderPro and decoderpro are not the same thing, and the name of the program we are starting is DecoderPro.

Paul

pischky, 2016-09-19:
Hello,
Is `readlink` or `readlink -e` or the problem on OSX? 
I guess that `realname` also is not supported. right?
I would suggest adding a test if readlink is supported or $(uname) is Linux to fix.

Using lowercase is convention on Unix. Nobody ist typeing LibreOffice. The link in /usr/bin is named libeoffice.

I stumbled a

rhwood, 2016-09-19:
> Is readlink or readlink -e or the problem on OS X?

```readlink -e``` is the problem on macOS.

 > I guess that realname also is not supported. right?

```realname``` is not supported in RHEL/Fedora/CentOS Linux either, so its invalid to assume that command is available in Linux.

> I would suggest adding a test if readlink is supported or $(uname) is Linux to fix.

See http://refspecs.linuxfoundation.org/LSB_5.0.0/LSB-Common/LSB-Common/rcommands.html#TBL-CORECMD for the list of commands you can expect in Linux. Neither ```readlink``` nor ```realname``` are in that list.

> Using lowercase is convention on Unix. Nobody ist typeing LibreOffice. The link in /usr/bin is named libreoffice.

This is irrelevant to @pabender's concerns. The scripts are deliberately written to use the script's name to determine which preferences to load, so ```decoderpro``` would load different preferences than ```DecoderPro```. (This allows a user to copy a script to a new name to get a new JMRI instance with a different preferences set.)

pischky, 2016-09-19:
Thanks for the list of LSB supported commands. So the only way to make ist work is to test that `readlink` is support on the current system. Or to rewrite readlink as a shell funktion :-(

I did not know about the usage of $(basename $0) for configuration but its written in the comments of the script. Should have read this before. But I think this is strange behaviour. `--config` parameter should be used in my opinion.

Martin

pabender, 2016-09-19:
On Mon, Sep 19, 2016 at 8:47 AM, pischky <notifications@github.com> wrote:

> Using lowercase is convention on Unix. Nobody ist typeing LibreOffice. The
> link in /usr/bin is named libeoffice.
>

While this is true by convention, there is no rule requiring it.  Also,
this isn't universally followed on all POSIX systems.

Paul

pabender, 2016-09-19:
On Mon, Sep 19, 2016 at 9:54 AM, pischky <notifications@github.com> wrote:

> I did not know about the usage of $(basename $0) for configuration but its
> written in the comments of the script. Should have read this before. But I
> think this is strange behaviour. --config parameter should be used in my
> opinion.
>

You have to have a way to find the default configuration, which is what the
whole $(basename $0) does for us.

Using the --config switch over-rides the default configuration.

Paul

bobjacobsen, 2017-07-03:
Can we close this?

pischky, 2017-07-03:
You can close it, I did not do any further research on it.
But I still believe there should be an integration of JMRI in the program menu on systems like Ubuntu or OpenSuse to allow non technical persons to start it.

closed 2017-07-03 by bobjacobsen