JMRI/JMRI #1979: Create test release 4.5.5

bobjacobsen, 2016-09-19:
For discussion of timing and what to include in 4.5.5

bobjacobsen, 2016-10-02:
"Replacement of RXTX with PureJavaComm (2nd version)" #936 has some recent comments about whether the RTX replacement could be merged for this (next) test release.

bobjacobsen, 2016-10-10:
I'd like to build this soon.  Should I wait a little longer for the PureJavaComm merge, or go ahead and build tomorrow?

rhwood, 2016-10-10:
I'd go ahead and build tomorrow without waiting for the PureJavaComm merge. We don't know enough about the observed NPEs in the PureJavaComm code to know that they don't prevent (re)configuration on Windows.

bobjacobsen, 2016-10-12:
See comment in #2058 "Signal mast Aspects don't work"

bobjacobsen, 2016-10-16:
The 4.5.5 candidate has a couple outstanding problems:
- #2077 C/MRI can't be set up from scratch
- #2081 TMCC throttle support broken
and probably others. (#2074 issues on macOS Sierra has a workaround: I'll release signed and unsigned versions to see how that goes)

I think it makes sense to release that build anyway, so unless some fixes come up between now and the time I get back to The Land of Bandwidth in about 10 hours, I'm going to release it this evening.

Thanks for all your patience.

pabender, 2016-10-16:
@bobjacobsen,

Sent from my iPad

On Oct 16, 2016, at 12:16 PM, Bob Jacobsen <notifications@github.com> wrote:

> The 4.5.5 candidate has a couple outstanding problems:
>
> #2077 C/MRI can't be set up from scratch
> #2081 TMCC throttle support broken and probably others. (#2074 issues on macOS Sierra has a [...]
PR #2078 should take care of #2077.  This at least let's you pick a serial port.  I don't have real C/MRI hardware available to test with further.

Paul

bobjacobsen, 2016-10-16:
Thanks.  I've pulled the current master into the `release-4.5.5` branch to pick up changes to date, including the C/MRI fix in #2078. 

Files are now on SourceForge (where they can be replaced if need be), with links in the release note at http://jmri.org/releasenotes/jmri4.5.5.shtml

I've added a link for a 4th (2nd on the list) download for "macOS", and a warning at the top describing why that's needed.  Would appreciate tests of those.

bobjacobsen, 2016-10-18:
Again moving the 4.5.5 build to the HEAD of master, to pick up C/MRI and table-sort fixes.  Jenkins build number 3 working on it; files will be made available via SourceForge shortly.

bobjacobsen, 2016-10-18:
Released

closed 2016-10-18 by bobjacobsen