JMRI/JMRI #2028: Unable to set text for Sensor states

rhwood, 2016-10-03:
This appears to be an I18N issue:

> Created a Sensor Icon of an existing sensor, right-clicked 'Change to Text' (and it does), right-clicked 'Set Text For Sensor States', and the window that should have the four states and spots to modify the text opens up correct size, but blank - no title text, just Window and Help menus, which work.
>
> System console shows:
>
> [WT-EventQueue-0] Exception: java.util.MissingResourceException: Can't find resource for bundle java.util.PropertyResourceBundle, key EnterInActiveToolTip

[as reported in JMRI Users Group](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/131724;_ylc=X3oDMTM4b3VhbTRlBF9TAzk3MzU5NzE0BGdycElkAzY2MDI3NTgEZ3Jwc3BJZAMxNzA1MDYzNTc2BG1zZ0lkAzEzMTcyNARzZWMDZnRyBHNsawN2dHBjBHN0aW1lAzE0NzUyNzA0NzEEdHBjSWQDMTMxNzI0)

bobjacobsen, 2016-10-03:
That symbol used to be in 

````
java/src//jmri/jmrit/display/DisplayBundle.properties:EnterInActiveToolTip = Enter the text to be displayed when the sensor is Inactive
````

pabender, 2016-10-03:
Hopefully that is fixed now with @rhwood's #2031 fixing the issue and my #2032 flagging if it happens again.

closed 2016-10-04 by rhwood