JMRI/JMRI #2043: There are too many issue trackers for JMRI

rhwood, 2016-10-05:
There are too many issue trackers for JMRI spread across multiple sites. These should be consolidated to a single site, and preferably to a single tracker.

To complete this, the following needs to be done:
- [ ] Update website to point to a single issue tracker
- [ ] Update documentation to use a single issue tracker
- [ ] Bar new issues at no-longer used issue trackers

rhwood, 2016-10-05:
Since [the query](https://sourceforge.net/p/jmri/mailman/message/35377231/) asking if we should move all new issues to GitHub has met with silence, I am assuming it is acceptable.

bobjacobsen, 2016-10-05:
I think this is great. One additional item, maybe 2nd or 3rd on the list:

- [ ]  Announce the change so that helpful people on jmriusers don't point back to the old tracker(s)

rhwood, 2016-10-05:
@bobjacobsen thanks. I've added that to the list.

rhwood, 2016-10-09:
I sent the following to jmriusers-owner@yahoogroups.com on 6 Oct 2016:

> We are changing the mechanism we use to track issues and requests for enhancement to use GitHub instead of SourceForge (see https://github.com/JMRI/JMRI/issues/2043 for more information).
>
>Can you please change the following line on the JMRI User’s group about page (https://groups.yahoo.com/neo/groups/jmriusers/info) from:
>
>	To  Report a Bug, Submit a Patch or Feature Request, please visit: https://sourceforge.net/p/jmri/_list/tickets
>
>to read:
>
>	Report an issue, Submit a patch, or Request an enhancement at https://github.com/JMRI/JMRI/issues
>
>The new tracker is already in use, so this change can be immediate.

I am unable to make this change myself.

bobjacobsen, 2016-10-10:
Done

rhwood, 2016-10-11:
Message sent to user's group at https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/131835

pabender, 2016-10-23:
@rhwood 

I see you are migrating issues now (thanks),  but there isca problem.

It looks like closed issues are being migrated along with the open ones.  

If those are not automatically closed, we are going to have several hundred issues that someone needs to go through  those and verify their status.

rhwood, 2016-10-23:
I don't have the time available to write a tool to import the SF.net trackers JSON export format and import that into GitHub (using the [GitHub API](https://developer.github.com/v3/)). [Recommended tools](https://github.com/cmungall/gosf2github) persistently fail to read the JSON export, although multiple other tools do verify it is valid JSON. [Other tools](https://github.com/ttencate/sf2github) get cutoff before completion by triggering abuse of API blocking.

Given these issues, We need to periodically remind users who submitted issues to the SourceForge trackers that if they want their open issues dealt with, they should open new issues in GitHub before eventually closing all issues at SourceForge.

closed 2016-10-23 by rhwood

pabender, 2016-10-23:
Personally, I don't think it was time to close this yet.

What I saw this morning was the issues from sourceforge successfully created in github, and then closed.

I think the only real issue I see is that the automated tools didn't filter the lists to only grab open issues.  If we were only pulling open issues from sourceforge, there should be about 200 of them, and most of those are feature requests.

So, a three step process should make the migration possible:
1) Pull the issues (all of them if required) from sourceforge.
2) delete any closed issues from the download
3) Upload the new file to github.

Paul