JMRI/JMRI #2053: Lua scripts don't work?

GosubGreg, 2016-10-08:
Tried entering a Swing example from lua.org and a simple io.write statement and when I press "Execute" with lua selected in the combo box it just echoes the script commands in the script output window.

rhwood, 2016-10-08:
The Script Output window *always* echoes the input in the Script Entry window as well as provide output from the script. Is there any output in the JMRI logs (available in the System Console under Help)?

GosubGreg, 2016-10-08:
Here's what I see in the console:
[AWT-EventQueue-0]
2016-10-08 16:48:01,506 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.ClassCastException: javax.script.SimpleScriptContext cannot be cast to org.luaj.vm2.script.LuajContext
	at org.luaj.vm2.script.LuaScriptEngine$LuajCompiledScript.eval(Unknown Source)
	at org.luaj.vm2.script.LuaScriptEngine.eval(Unknown Source)
	at org.luaj.vm2.script.LuaScriptEngine.eval(Unknown Source)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:264)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:245)
	at jmri.jmrit.jython.InputWindow.buttonPressed(InputWindow.java:268)
	at jmri.jmrit.jython.InputWindow.lambda$2(InputWindow.java:130)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

My script input:
io.write('Hello, what is your name? ')

Script output:
>>> io.write('Hello, what is your name? ')

I already had LuaJ working on my Mac from other Java/Lua work I'm doing - maybe that caused issues?

Thanks so much for your help!

rhwood, 2016-10-08:
This has been reported as a bug already: https://sourceforge.net/p/luaj/bugs/49/

JMRI provides a standard Java Scripting API context object that allows JSR-223 scripts to access JMRI internals. It appears that LuaJ does not like that context object, but this is an issue with LuaJ, not JMRI.

GosubGreg, 2016-10-08:
Guess I lucked out by avoiding JSR-223 on my Java/luaJ application.

Is there any simple work-around in the mean time or is it time to learn python?

Appreciate the quick turn-around - you da man!

rhwood, 2016-10-08:
You'll get the most support from us using python, but JavaScript support is also built in (it's actually built into the JRE, so we get it for free as it were). There appear to be multiple [JSR-223 Lua](https://www.google.com/#q=JSR-223+Lua) implementations, you could try a version other than LauJ.

rhwood, 2016-10-16:
@GosubGreg Have you found a working implementation?

closed 2016-10-22 by rhwood