JMRI/JMRI #2058: Signal mast Aspects don't work

silverailscolo, 2016-10-09:
As noted by a user on the jmriusers group:
In the Signal Mast table, new masts lack a default Aspect after creation and their Aspect can't be set from the drop down list as possible in earlier versions. Seen on LocoNet, LoceNet Simulator and NCE Simulator under macOS and Windows.
[image]

KenC57, 2016-10-09:
What type of signal mast?? They have a lot of differences to how a mast
operates and what the issue might be would be very different for the many
types. Please post an example (step by step) to see the issue.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

silverailscolo, 2016-10-09:
On Panels all Mast types work fine. But any mast displays bad behavior when trying to set it from the table. I suspect it to be caused by the drop downs not working correctly in tables that use
``import javax.swing.table.TableCellRenderer;``
``import javax.swing.table.TableRowSorter;``
Compare the Block table drop down in the last column to the Aspect column of the Signal Mast table. This is what he drop down looked like in 4.5.3:
[image]
In 4.5.4 and later, the drop downs don't show (see earlier screen shot).

bobjacobsen, 2016-10-12:
Is this significant enough to hold the test release? Otherwise, it'll start building later today.

silverailscolo, 2016-10-12:
Although important, this was in 4.5.4 and we don't have a fix yet so to me it seems OK to build for 4.5.5 with this issue open for now.

silverailscolo, 2016-10-13:
Looking for a working Drop Down menu in a TableRowSorter table, I noticed that as long as you don't shift the Column order, there's one in SignalingMastPairs > Add... pane (on the Signal Masts tab). Sorting doesn't work for the Aspect column, hiding and showing States and when you shift Columns some entries stay in their old place:
[image]
So we should look into the combination of the TableRowSorter and Drop-down Editors. In the meantime I will try to pick some working code to fix settting Aspects in the SignalMast table. Currently the following errro is thrown hen trying to set one:
``2016-10-13 21:02:52,717 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.ClassCastException: java.lang.String cannot be cast to java.lang.Boolean
	at jmri.jmrit.signalling.SignallingPanel$SignalMastModel.setValueAt(SignallingPanel.java:1648)
	at javax.swing.JTable.setValueAt(JTable.java:2741)
	at javax.swing.JTable.editingStopped(JTable.java:4723)
	at javax.swing.AbstractCellEditor.fireEditingStopped(AbstractCellEditor.java:141)
	at javax.swing.DefaultCellEditor$EditorDelegate.stopCellEditing(DefaultCellEditor.java:368)
	at javax.swing.DefaultCellEditor$3.stopCellEditing(DefaultCellEditor.java:158)
	at javax.swing.DefaultCellEditor.stopCellEditing(DefaultCellEditor.java:233)
	at javax.swing.plaf.basic.BasicTableUI$Handler.mousePressed(BasicTableUI.java:1010)
	at javax.swing.plaf.basic.BasicTableUI$MouseInputHandler.mousePressed(BasicTableUI.java:798)
	at java.awt.AWTEventMulticaster.mousePressed(AWTEventMulticaster.java:280)
	at java.awt.Component.processMouseEvent(Component.java:6532)``

silverailscolo, 2016-10-23:
Found the cause, working on a fix. As the creation of a JTable uses 3 params instead of 1 since 4.5.4, the constructor in signaling.SignalMastTableDataModel that used to override makeJTable() is being bypassed. Thus, the ComboBox renderer etc. are not created, leaving the Aspect cells blank.
[ ] move Table (display) code out of SignalMastDataModel class (leave deprecated class)
[ ] write new cellRenderer and cellEditor in signalMastTableAction extending JTable
[ ] convert columnIndex between DataModel and TableView to fix display issues after sorting
[ ] have column width adjust for i18n

silverailscolo, 2016-11-07:
After some study I conclude that the createModel() implementation breaks the SignalMast Aspect comboBoxes. Without extending the standard JTable it is not possible to create mast (row) specific lists (JTable has just one class of editors/renderers) per column). So I suggest to add some code in an extended JTable.

rhwood, 2016-11-07:
> On Nov 7, 2016, at 16:03, Egbert Broerse <notifications@github.com> wrote:
> 
> After some study I conclude that the createModel() implementation breaks the SignalMast Aspect comboBoxes. Without extending the standard JTable it is not possible to create mast (row) specific lists (JTable has just one class of editors/renderers) per column). So I suggest to add some code in an extended JTable.
> 
Make the model return a specific subclass of JPanel for the problematic column and have that render whatever you need (a label, combobox, button, image, another table). The table model is not broken, you just aren't specifying the correct renderer/editor.

silverailscolo, 2016-11-13:
First - partially working - code for fix is in [my branch](https://github.com/silverailscolo/JMRI/blob/eb-sigmasttable/java/src/jmri/jmrit/beantable/signalmast/SignalMastTableDataModel.java). Doesn't yet update the current value for Aspect and needs 2 clicks to open.

silverailscolo, 2016-11-23:
Fixed in PR #2573 while using TableRowSorter.

closed 2016-11-26 by silverailscolo