JMRI/JMRI #2064: Ops car roster confused after changing Locale

silverailscolo, 2016-10-10:
Having added a translated term for Loads/Empties confuses Operations when I create a new car and try to save it. Instead of L/E the German translation uses B/L from the CarsBundle. The pop up for creating a new car however still shows a choice of either L or E, created from my OperationsCarRoster.xml file.
When I press Save (Speichere), the Console replies with:
``2016-10-10 21:56:42,333 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NoSuchMethodError: jmri.util.PhysicalLocation.equals(Ljmri/util/PhysicalLocation;)Ljava/lang/Boolean;
	at jmri.jmrit.operations.locations.Location.store(Location.java:1445)
	at jmri.jmrit.operations.locations.LocationManager.store(LocationManager.java:421)
	at jmri.jmrit.operations.locations.LocationManagerXml.writeFile(LocationManagerXml.java:60)
	at jmri.jmrit.operations.OperationsXml.writeOperationsFile(OperationsXml.java:31)
	at jmri.jmrit.operations.OperationsXml.writeFileIfDirty(OperationsXml.java:97)
	at jmri.jmrit.operations.OperationsXml.save(OperationsXml.java:210)
	at jmri.jmrit.operations.rollingstock.cars.CarEditFrame.writeFiles(CarEditFrame.java:584)
	at jmri.jmrit.operations.rollingstock.cars.CarEditFrame.buttonActionPerformed(CarEditFrame.java:559)``
When next opening JMRI > Operations > Cars, the file can't be parsed caused by an unexpected EOF. Had to resore from an auto backup.
Manually editing my OperationsCarRoster for the translated keys as follows:
``  <loads>
    <defaults empty="L" load="B" />
  </loads>``
fixes reading/saving, but that means users should not create cars and than change Locale. While we sort this out, it seems wise not to translate the E/L key (in OpsCarsBundle).

DanielBoudreau, 2016-10-10:
Egbert,

The exception is most likely due to a wrong vecrmath jar file.  Are you sure you have an up to date set of files?

Dan

rhwood, 2016-10-10:
Note that the web and json clients for operations do not necessarily use the Locale that JMRI is using, so its important that codes that have special meaning within JMRI be localized only when displayed, but not when stored for persistence.

DanielBoudreau, 2016-10-10:
Egbert,

One more thing.

If you first start the program using the German local, and with a clean start with regards to operations, no saved files, you should find that the default load names for German is B/L.  Once you save the files using a specific locale the default load names are saved since the user is allowed to change them.

Dan

pabender, 2016-10-10:
On Oct 10, 2016, at 5:00 PM, Randall Wood <notifications@github.com> wrote:
> Note that the web and json clients for operations do not necessarily use the Locale that JMRI is using, so its important that codes that have special meaning within JMRI be localized only when displayed, but not when stored for persistence.
> 
Randall is absolutely right here.

The L and E are the default loads.  If the user wishes to change those, they should really be custom load types, which the user can specify in whatever language desired.

Paul

DanielBoudreau, 2016-10-10:
The default load names are loaded from the active language bundle file.  So if a user starts operations in English, they get English default names, if in German then German names, and in either case only when operation files are saved do the default names get stored.  Default load names can be modified by the user to anything they want via the GUI, using the same tool that creates custom load names for their rolling stock.

There should never be a need to directly edit the operation xml files.

Dan

silverailscolo, 2016-10-10:
Thanks for the replies. This at least answers an earlier question as to why newly translated Ops color names (and L/E) weren't showing up in my setup: Operations effectively freezes the properties at the time of first use. Seems reasonable to protect the way we store things in the xml.
However, the error I got was not caused by editing the xml, but by creating a new car in the GUI under a different Locale, something any user might do unaware of a change in the properties file. Haven't yet understood all the details, but this could be caused by the code comparing the stored defaults to the current values in the cars Bundle and getting confused.
I expect the false pointer to vecmath is caused by a parsing "error" of the previous field, maybe looking for a character that isn't there in the translation and adding that to the following instruction (which explains the capital L in Ljava...) resulting in the first error.

DanielBoudreau, 2016-10-11:
Egbert,

I can’t reproduce your error, and I understand where the exception is coming from.  The trace shows that this line in operations is causing the exception:

java.lang.NoSuchMethodError: jmri.util.PhysicalLocation.equals(Ljmri/util/PhysicalLocation;)Ljava/lang/Boolean;
at jmri.jmrit.operations.locations.Location.store(Location.java:1445)

        // save physical location if not default
if (getPhysicalLocation() != null && !getPhysicalLocation().equals(PhysicalLocation.Origin)) {
            e.setAttribute(Xml.PHYSICAL_LOCATION, getPhysicalLocation().toString());
        }

As I stated before, I’ve seen this error when the wrong jar file is in use.  Confirm which vecmath jar file is in use.

Dan

silverailscolo, 2016-10-11:
Didn't find any older vecmath, except one stored in the Documents folder. After deleting it and running ``ant clean`` , I no longer get this error. Thanks for the pointers.

closed 2016-10-11 by silverailscolo

silverailscolo, 2016-10-12:
Paul and Daniel, if that’s the case, shouldn’t we remove the translatable L/E keys from the properties? I translated them to match the full length keyword in the interface (Beladen/Leer = Loaded/Empty). It would be confusing to the user when these would be selected as L/E instead of B/L from the drop down. We could guide the user by including the abbreviation in parenthesis behind the name, like "Beladen (L)”.
Egbert

> Op 10 okt. 2016, om 23:44 heeft Paul Bender <notifications@github.com> het volgende geschreven:
> > Note that the web and json clients for operations do not necessarily use the Locale that JMRI is using, so its important that codes that have special meaning within JMRI be localized only when displayed, but not when stored for persistence.
> 
> Randall is absolutely right here.
> The L and E are the default loads. If the user wishes to change those, they should really be custom load types, which the user can specify in whatever language desired.
> 
> Paul

DanielBoudreau, 2016-10-12:
No, I would not remove them.  They are correct when the user starts the program in their appropriated language.  Once they “Save” their setting, then whatever they are called is saved, which could be anything the user desires.  The default names for loads is only a starting point, and can be changed at any time.

Dan