JMRI/JMRI #2065: MERG simulation cannot aquire a throttle

petecressman, 2016-10-10:
The Tools->Throttles->New Throttle screen throttle cannot set an address. The message
"Throttle request (RLOC) timed out" is returned.  

Also, attempting to run a warrant will get the same error message.  Note: Preferences Defaults should be set to MERG Throttles.  No other simulators have this problem. Problem may be related to Bug #554.

rhwood, 2016-10-10:
#554 is the translation of some strings into French. Is that the correct issue?

petecressman, 2016-10-11:
No different issue.  this is https://sourceforge.net/p/jmri/bugs/554/

#554 Throttle speed in warrant not recognized.

pabender, 2016-10-11:
On 10/10/2016 07:35 PM, petecressman wrote:
> The Tools->Throttles->New Throttle screen throttle cannot set an
> address. The message
> "Throttle request (RLOC) timed out" is returned.
>
> Also, attempting to run a warrant will get the same error [...]

Has the throttle actually been implemented on the MERG simulator?

I see where we define the throttle interface, but that doesn't mean we 
respond to the commands when the simulated connection is used. In that 
case, the error message you note above just acknowledges that the 
command was sent, but no response was ever received (which may be 
correct behavior, even if it isn't the behavior you desire).

Paul

petecressman, 2016-10-11:
If it's the case that a MERG simulator has not been implemented, then the issue is it should not be listed as an available connection type in Preferences.

This then would separate Bug #554 as a different problem. In #554, the MERG throttle was broken in release 3.9 and still is not fixed.

PeteC 

 

Has the throttle actually been implemented on the MERG simulator?

I see where we define the throttle interface, but that doesn't mean we 
respond to the commands when the simulated connection is used. In that 
case, the error message you note above just acknowledges that the 
command was sent, but no response was ever received (which may be 
correct behavior, even if it isn't the behavior you desire).

Paul

pabender, 2016-10-11:
On Oct 11, 2016 3:28 PM, "petecressman" <notifications@github.com> wrote:
>
> If it's the case that a MERG simulator has not been implemented, then the
issue is it should not be listed as an available connection type in
Preferences.
>

There are actually two ways to think about our simulators.  One aspect is
the communication channel itself and the other aspect is responding to
specific commands.

The MERG simulator definitely implements the communication channel.

My suggestion here is that the MERG simulator does not respond to the
throttle commands.

Paul

pnbgit, 2016-10-26:
The MERG simulator allows a simulated CBUS connection, which enables CBUS panels to be loaded and worked on without an actual layout connection.  You can also see the CBUS messages that the panel sends in the MERG console.  So it is a valid connection option and very useful for those of us developing panels for CBUS.  However, as Paul suggests, it does not implement a simulated throttle protocol (yet?).  

So it isn't, IMHO, a bug, it is just a feature that we don't have yet. 

You could do a jython script, to run in simulation mode, that responds to the RLOC command with the appropriate PLOC command, then you would be able to get a throttle in simulation mode.  There is not much else comes back from the Command station in a basic throttle session, so that might be all you need.

Pete

bobjacobsen, 2016-10-26:
> On Oct 26, 2016, at 3:52 PM, pnbgit <notifications@github.com> wrote:
> 
> The MERG simulator allows a simulated CBUS connection, which enables CBUS panels to be loaded and worked on without an actual layout connection. You can also see the CBUS messages that the panel sends in the MERG console. So it is a valid connection option and very useful for those of us developing panels for CBUS. However, as Paul suggests, it does not implement a simulated throttle protocol (yet?).

Some of the Simulator connections just include a debug ThrottleManager. That lets various throttle tools (appear) to work, but it doesn't simulate the actual traffic. 

It would straight-forward to add that to the MERG simulator.

icklesteve, 2018-12-22:
See #6341 which adds a MERG Command Station Simulation for both JMRI and external throttles,
[http://jmri.org/help/en/package/jmri/jmrix/can/cbus/swing/simulator/SimulatorPane.shtml](http://jmri.org/help/en/package/jmri/jmrix/can/cbus/swing/simulator/SimulatorPane.shtml)
This issue can now be closed.

closed 2021-03-12 by icklesteve