JMRI/JMRI #2068: NPE in SPROG OpsMode programming

bobjacobsen, 2016-10-12:
From a [JMRIusers thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/16558) where a user reports an NPE when programming in ops mode with an SPROG:

````
2016-10-12 16:45:37,547 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2016-10-12 16:45:38,255 util.Log4JUtil INFO - This log is appended to file: /Users/Simon/Library/Preferences/JMRI/log/messages.log [main]
2016-10-12 16:45:38,257 util.Log4JUtil INFO - This log is stored in file: /Users/Simon/Library/Preferences/JMRI/log/session.log [main]
2016-10-12 16:45:38,268 apps.AppsBase INFO - DecoderPro version 4.5.5ish-201610102140-jenkins-R5cb214e starts under Java 1.8.0_101 on Mac OS X x86_64 v10.11.6 at Wed Oct 12 16:45:38 AEDT 2016 [main]
2016-10-12 16:45:41,473 gui3.Apps3 INFO - Starting with profile Sprog_POM.3f0d20bb [main]
2016-10-12 16:45:41,532 node.NodeIdentity INFO - Using jmri-E4CE8F5EF7B0-3f0d20bb as the JMRI Node identity [main]
2016-10-12 16:45:41,653 serialdriver.SerialDriverAdapter INFO - /dev/cu.usbserial-feedcode port opened at 9600 baud, sees DTR: true RTS: true DSR: true CTS: true CD: false [main]
2016-10-12 16:45:42,126 util.FileUtilSupport INFO - File path program: is /Volumes/JMRI 4.5.5ish-201610102140-jenkins-R5cb214e/JMRI/ [main]
2016-10-12 16:45:42,126 util.FileUtilSupport INFO - File path preference: is /Users/Simon/Dropbox/JMRI/ [main]
2016-10-12 16:45:42,126 util.FileUtilSupport INFO - File path profile: is /Users/Simon/Library/Preferences/JMRI/Sprog_POM/ [main]
2016-10-12 16:45:42,126 util.FileUtilSupport INFO - File path settings: is /Users/Simon/Library/Preferences/JMRI/ [main]
2016-10-12 16:45:42,127 util.FileUtilSupport INFO - File path home: is /Users/Simon/ [main]
2016-10-12 16:45:42,127 util.FileUtilSupport INFO - File path scripts: is /Volumes/JMRI 4.5.5ish-201610102140-jenkins-R5cb214e/JMRI/jython/ [main]
2016-10-12 16:45:42,734 swing.RosterFrame ERROR - Unable to get the double click speed, Using JMRI default of half a secondjava.lang.NullPointerException [main]
2016-10-12 16:45:52,125 throttle.AddressPanel INFO - Asked for 1214 got 1214(L) [AWT-EventQueue-0]
2016-10-12 16:46:17,924 symbolicprog.CvValue WARN - Exception during write CV '16.1.315' to '119' [AWT-EventQueue-0]
java.lang.NullPointerException
at jmri.jmrix.sprog.SprogOpsModeProgrammer.writeCV(SprogOpsModeProgrammer.java:53)
at jmri.implementation.AddressedHighCvProgrammerFacade.writeCV(AddressedHighCvProgrammerFacade.java:70)
at jmri.implementation.MultiIndexProgrammerFacade.writeCV(MultiIndexProgrammerFacade.java:111)
at jmri.jmrit.symbolicprog.CvValue.write(CvValue.java:536)
at jmri.jmrit.symbolicprog.DecVariableValue.writeAll(DecVariableValue.java:349)
at jmri.jmrit.symbolicprog.DecVariableValue.writeChanges(DecVariableValue.java:332)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.executeWrite(PaneProgPane.java:858)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.nextWrite(PaneProgPane.java:1149)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.writePaneChanges(PaneProgPane.java:1083)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane$3.itemStateChanged(PaneProgPane.java:379)
at javax.swing.AbstractButton.fireItemStateChanged(AbstractButton.java:2050)
````

bobjacobsen, 2016-10-12:
From just reading code, without recreating this, the fault here:
````
java.lang.NullPointerException
at jmri.jmrix.sprog.SprogOpsModeProgrammer.writeCV(SprogOpsModeProgrammer.java:53)
````

is pointing at:

````
        // Add the packet to the queue rather than send it directly
        // [AC 23/01/16] Check that there is a free slot for the ops mode packet.
        // Delay the reply to allow time for the ops mode packet to be sent and prevent all slots from filling up
        // when writing multiple CVs, e.g. writing a sheet in the comprehensive programmer.
        if (_memo.getCommandStation().opsModepacket(mAddress, mLongAddr, CV, val) != null) {
````

Either `memo` or `memo.getCommandStation()` must be `null`.  Could this be a fault in the configuration process much earlier?

crosland, 2016-10-13:
I suspect this is a fault in the configuration due to changes for multi system. Similar to the issue that caused the F keys top stop working in SPRPG programmer mode. I will investigate.

crosland, 2016-10-13:
I can't see how to assign this issue to myself. Does someone else have to do it?

KenC57, 2016-10-13:
Andrew,

 

While logged into GitHub, bring up the issue. On the right side you will see
things like 'Assignees' and a very small gear icon next to that word. Click
that and you get a list of the names, find your name and click to add (or
double click, I'm not that sure). But I'm not seeing your name in the list,
so maybe you aren't joined to the project?? If that's the case, it's been so
long I don't recall that part of it. So maybe Bob or one of the others might
be able to help on that.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

crosland, 2016-10-13:
I don't even see the gear icon. May be because I'm not joined to the project?

mattharris, 2016-10-13:
@crosland Yes, it's due to not being assigned to one of the project groups. I'll take a look at it later on this evening  (I'm in CET timezone)

crosland, 2016-10-13:
Found an issue with adapterMemo not being assigned in one of the SprogProgrammerManager constructors but when I fix that the application hangs. The next line after the one that was causing the NPE starts a timer for the programming operation timeout. I see a 'TimerQueue' suspended at 'Unsfae.Park' in the debugger but don't really know what that means. I will continue digging.

mattharris, 2016-10-13:
@crosland I've invited you to be part of the JMRI organisation as a member of the Developers group. This should give you the possibility to assign an issue to yourself

crosland, 2016-10-13:
I still don't see the gear icon for assignees. I would also like to mark #2073 as 'resolved' or 'fixed' or 'rebuilt'. I'm not sure what terminology we use or when issues are marked thus.

crosland, 2016-10-13:
I tried log out/log in. No luck.

mattharris, 2016-10-13:
Hmmm, I guess that the GitHub permissions don't quite do what I thought.

Anyhow, I've now been able to assign this issue to you as you're now a member.

pabender, 2016-10-25:
Andrew,

On Oct 21, 2016 1:34 PM, "Andrew Crosland" <notifications@github.com> wrote:
>
> I am stuck on this and will be unable to work on it for at least a week.
Can anyone else help? WHat I have fpund so far:
>
> "Found an issue with adapterMemo not being assigned in one of the

As i mentioned in another e-mail ( off list ) i was able to replicate this,
along with the  fix.

Could you go ahead and push this up to your github  repository so we can
work on this from a common point of reference?  You don't need to submit a
pull request, but you can if you like.

> SprogProgrammerManager constructors but when I fix that the application
hangs. The next line after the one that was causing the NPE starts a timer
for the programming operation timeout. I see a 'TimerQueue' suspended at
'Unsfae.Park' in the debugger but don't really know what that means. I will
continue digging."
>

I have not been able to replicate the hanging.  I've still only tested with
a PiSprogOne though.

Paul

crosland, 2016-11-06:
Paul,

Sorry for the delay. I have pushed my change to my fork crosland/JMRI.

Andrew

crosland, 2016-11-06:
I also updated my fork to JMRI:master and I no longer see any problem. I will do a bit more testing.

closed 2017-01-22 by bobjacobsen