JMRI/JMRI #2069: Migrate from jmri.headlesstest to native java.awt.headless

bobjacobsen, 2016-10-12:
With Java 1.8, we can directly control the setting of the `java.awt.headless` property, so we no longer need to track that with our own `jmri.headlesstest ` property. Thanks to @rhwood and @petecressman for figuring this out.  Our code will be simpler if we migrate from checking the `jmri.headlesstest ` property to asking Java itself whether it's running headless with `GraphicsEnvironment.isHeadless()`

The steps are (in approximate order):
- [ ] change the documentation pages (see below)
- [ ] do an automated migration from `"jmri.headlesstest"` to `"java.awt.headless"` to catch the places where tests are checking our shadow property value
- [ ] clear up a couple special cases: `java/test/jmri/HeadLessTest.java`, `runtest.csh`, `build.xml` (several places)

That completes the basic migration. After that, ee can slowly migrate tests as we work on them and they're converted to JUnit4.

New recommendations to be added to the JUnit page:

- When checking whether to run a JUnit4 test, use `Assume.assumeFalse(GraphicsEnvironment.isHeadless())`
- If you have to explicitly test for some reason, check the value of `GraphicsEnvironment.isHeadless()` instead of the `jmri.headlesstest` or `java.awt.headless` properties

pabender, 2016-10-12:
@bobjacobsen,

Instead of an automated change from jmri.headlesstest to java.awt.headless, why don't we do an automated change to java.awt.GraphicsEnvironment.isHeadless()?

That should work with the JUnit3 tests we haven't migrated yet, because we're typically using that boolean in an if statement to protected against running GUI code in headless mode.

Paul

bobjacobsen, 2016-10-12:
> Instead of an automated change from jmri.headlesstest to java.awt.headless, why don't we do an automated change to java.awt.GraphicsEnvironment.isHeadless()?

Mostly because I think the jmri.headless -> java.awt.headless can be quickly done with 100% removal of jmri.headless, which lets the rest of the process proceed.  There are a lot of idioms that _use_ jmri.headless, e.g. `!System.getProperty("jmri.headlesstest", "false").equals("true")`, `Boolean.getBoolean("jmri.headlesstest")` so it's more work/time to track them all down.

I'd rather get the first change done, to pull out jmri.headless, and then do the next steps as a "remove java.awt.headless when we can" campaign.

rhwood, 2016-11-08:
#2474 removes any setting of ```jmri.headlesstest```, but leaves the tests that use it intact. This was done to make it easy to find the uses of jmri.headlesstest, since most uses of jmri.headlesstest bypass/ignore an entire test suite, when the tests for headlessness should be bypassing/ignoring individual test methods instead of classes or suites (I don't think that change from suite to method can be automated).

rhwood, 2016-11-09:
#2477 removes all mention of ```jmri.headlesstest``` (except the French technical documentation).

```java.awt.headless``` is still used in tests where it prevents an entire suite from being run if headless, since this indicates where ```GraphicsEnvironment.isHeadless()``` tests need to be added to methods within the suite before removing the suite-wide test.

commit 4d1b8e12 (2016-11-24): Merge pull request #2574 from rhwood/headless-tests

Complete migration from headless checks in suites to tests.
Closes #2069
closed 2016-11-24 by rhwood