JMRI/JMRI #2074: JMRI runs afoul of macOS Sierra Security

rhwood, 2016-10-14:
Downloads of JMRI copied from an unsigned DMG are unable to launch when double clicked in the Finder (or launched using the ```open``` command) due to [Gatekeeper Path Randomization](http://mjtsai.com/blog/2016/06/16/gatekeeper-path-randomization/) (lots of links to discussions about the issue).

We basically have two alternatives:
1. Sign our JMRI builds using an Apple developer certificate
2. Ship everything JMRI needs within the applications bundles in Contents/Resources (this is confirmed to work, but since we ship DecoderPro, PanelPro, and SoundPro triples the size of the macOS download)

rhwood, 2016-10-14:
It is only a one line change to the StartJMRI script to make alternative 2 work (I don't know the scope of the build.xml changes).

rhwood, 2016-10-14:
Further research suggests that alternative 1 is not a workable alternative unless an application is shipped from the App Store since applications, even signed ones, appear to be subject to Gatekeeper Path Randomization (GPR) if copied from an unsigned DMG (which I think is only supported in OS X 10.11 and macOS 10.12+) using the Finder or run from anywhere other than /Applications

We could also provide an installation script that downloads JMRI and installs it using a Terminal command (since this would avoid a copy using the Finder that would trigger the GPR protections).

bobjacobsen, 2016-10-15:
Can we make option 1 (signed builds) work if we distribution on a signed .dmg?

I have the JMRI developer credentials from Apple.  I can share those with Matt (the other current release pumpkin; it would be good to have one or two others).  

Right now, I don't understand whether signed .dmg is openable on older MacOS X systems (back to 10.7.3, which I think is the oldest on which Java 1.8 will work). It could be that they're opened, but the signing isn't checked.  Or it could be that they don't open.

- If it opens, then I think that distributing an officially-signed .dmg is the right approach.  
- If it doesn't open, then we'd probably have to distribute to the Mac in two forms for now:  (1) as now, for Mac OS X and (2) in a signed .dmg for macOS.

Why is this better than building all-inclusive bundles?  In a perfect world, we'd have read-only program directories.  But I don't think we're there yet.  Lots of people have panels that reference their contents as program-directory resources, for example. If we really are going to _enforce_ read-only program directories (via Apple's changes), we should have a version or two that warns of the situation and helps migrate.

rhwood, 2016-10-15:
I'm not confident option 1 will "just work" unless combined with option 2 due to https://developer.apple.com/library/prerelease/content/technotes/tn2206/_index.html#//apple_ref/doc/uid/DTS40007919-CH1-TNTAG17

The issue being that unless the app is copied from a signed DMG to /Applications (and only /Applications) prior to running, it gets treated the same as JMRI is treated now.

Signed DMGs can only be created on OS X 10.11 or macOS, and do not appear to function on 10.9 or earlier (same reference, different section).

Randall Wood

bobjacobsen, 2016-10-15:
> @rhwood wrote:
> 
> I'm not confident option 1 will "just work" unless combined with option 2 due to https://developer.apple.com/library/prerelease/content/technotes/tn2206/_index.html#//apple_ref/doc/uid/DTS40007919-CH1-TNTAG17

I think the relevant quote from that page is "Starting with macOS Sierra, running a newly-downloaded app from a disk image, archive, or the Downloads directory will cause Gatekeeper to isolate that app at a unspecified read-only location in the filesystem.”

> The issue being that unless the app is copied from a signed DMG to /Applications (and only /Applications) prior to running, it gets treated the same as JMRI is treated now.

I read it as running from ~/Downloads or leaving it on the disk image won’t work, but that copying it to other locations will work, including but not limited to /Applications.  

> Signed DMGs can only be created on OS X 10.11 or macOS, and do not appear to function on 10.9 or earlier (same reference, different section).

I don’t see that. Could you point me more specifically?  I do see references to needing later versions to _sign_ images, but that’s not the same as being unable to open/mount them.

Even in the worst-case understanding, this means the choice is between:

1) Keeping our current structure, but having to have the program directory be in /Applications

2) Changing to a multi-program-directory structure with two complete app bundles for DecoderPro and PanelPro.  

There are good and bad aspects to both of those; the choice is not at all obvious to me.

rhwood, 2016-10-15:
>>  Signed DMGs can only be created on OS X 10.11 or macOS, and do not appear to function on 10.9 or earlier (same reference, different section).

> I don’t see that. Could you point me more specifically?  I do see references to needing later versions to _sign_ images, but that’s not the same as being unable to open/mount them.

Sorry. I can't find that either, but I did read it over the last two days.

Note that the .app bundle itself must be copied or moved (to anywhere), copying or moving a folder containing the .app bundle will trigger the GPR. An application could be moved out of ~/Downloads and back in and run without issue, just so long as the application itself is what's moved (there is no discussion I've found of what happens if an app + external resources are moved as a group).

Its seems more and more that anything not incorporating #2075 (option 2) will only be fragile.

bobjacobsen, 2016-10-15:
I've taken the 4.5.5 candidate DMG file and 
- signed the three app bundles
- then signed the DMG
(these are signed with my personal certificate, not JMRI's, but we can resolve that later)

It works on my Yosemite and Sierra machines, in the sense that I can mount, copy to Applications and run with just the usual "Downloaded from the Internet, OK?" warnings. But that might be because I'm known on those machines.  

I've posted the file to Sourceforge at https://sourceforge.net/projects/jmri/files/support%20files/JMRI.4.5.5-Rbf46600%20sign%20test.dmg/download

It would be great if somebody could test these
- [ ] on Sierra
- [ ] on Yosemite
- [ ] on some older MacOS X versions

Thanks

rhwood, 2016-10-16:
These work on my Sierra system, including network-based systems (port may matter). However, even if we do begin signing the DMGs for test and production releases, I think we want the nightlies to also work (since we sometimes point users to them to see if a specific issue has been fixed), but I also think we don’t want to sign everything automatically on the build server either, so we may want to also create self-contained .app bundles.

Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

bobjacobsen, 2016-10-16:
> @rhwood wrote:
> 
> These work on my Sierra system, including network-based systems (port may matter).

Thanks!

Does anybody else have an older OS X system to try this on?

> However, even if we do begin signing the DMGs for test and production releases, I think we want the nightlies to also work (since we sometimes point users to them to see if a specific issue has been [...]

I don’t know how to sign within Jenkins while still protecting the certificate. There might be a way, as it seems a common enough need. I’ll ask around.

We do use a “signing server” at the day job. You scp a file to a location, and then pick up the signed file via scp at a related location.  The authorization is via scp's sign-in credentials.  I’ll ask about how it’s done.

If we can’t get the test packages signed somehow, I agree we should provide self-contained bundles.

pabender, 2016-10-16:
@bobjacobsen,

It was very slow too install, but the test package worked on my mid 2010 Mac Mini running El Capitan.

Paul

bobjacobsen, 2016-10-16:
> It was very slow too install, but the test package worked on my mid 2010 Mac Mini running El Capitan.

Do you mean the copy from the disk image was slow?

Not sure what that’s happening. I thought that a signed DMG was just inspected at mount time, not while it’s being accessed.

rhwood, 2016-10-16:
This works from my 2008 El Capitan iMac — the copy feels slower than from an unsigned DMG. I would not be surprised if everything within the signed .app bundles is also being inspected, and that each inspection is triggering a complete revivification of the entire chain of trust.

Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

pabender, 2016-10-16:
Sent from my iPad

On Oct 16, 2016, at 2:28 AM, Bob Jacobsen <notifications@github.com> wrote:
> > It was very slow too install, but the test package worked on my mid 2010 Mac Mini running El Capitan.
> 
> Do you mean the copy from the disk image was slow?

Yes, the actual copy was slow.  I was doing some other stuff at the same time ( web browsing mostly) but the copy took at least 10 minutes, which is longer than normal.

> 

Paul

bobjacobsen, 2016-10-16:
It might have been slow because that signed .dmg wasn't compressed, hence was twice as big as it should have been.  

I've rebuilt 4.5.5 with the C/MRI fix, and put a (still manually made) signed and compressed .dmg on SourceForge at https://sourceforge.net/projects/jmri/files/test%20files/JMRI.4.5.5-Rcb78224-signed.dmg/download

You can also get that from the _draft_ release note at http://jmri.org/releasenotes/jmri4.5.5.shtml

bobjacobsen, 2016-10-23:
JMRI 4.5.5 was released with unsigned and signed .dmg files.  The signed version was created manually.

User reports say that the signed one worked back to Mac OS 10.8. 

For 4.5.6, perhaps we can just create a signed dmg for distribution. We can create some automation for creating the signed .dmg, but so far we don't know how to do that on Jenkins without possibly exposing our certificate.  More to follow.

bobjacobsen, 2016-10-31:
I'm testing a Jenkins setup that will send the macOS package off to a slave Jenkins instance for signing.  That'll let us have a remote machine that (reasonably securely) holds the certificates.

If I can get that to work, we can use it for signing the Jenkins 'Packages' and release-build job products.

rhwood, 2016-10-31:
That's cool. I'd still like to be able to entirely encapsulate JMRI in the macOS launchers anyway since a self contained app can be copied, moved, etc independently of the other apps without breaking.

pabender, 2016-10-31:
On Oct 31, 2016 11:55 AM, "Randall Wood" <notifications@github.com> wrote:
>
> That's cool. I'd still like to be able to entirely encapsulate JMRI in
the macOS launchers anyway since a self contained app can be copied, moved,
etc independently of the other apps without breaking.
>

Just thinking about this, what implications does that have for add on
applications l, such as CATS that we don't dostribute?

Paul

bobjacobsen, 2016-10-31:
> Just thinking about this, what implications does that have for add on applications, such as CATS that we don't distribute?

Good question!

CATS now drops files into a standard JMRI installation, and uses its own launcher.  I _think_ that will still work, but will check.  They might have to sign their launcher bundle.

bobjacobsen, 2016-10-31:
> I'd still like to be able to entirely encapsulate JMRI in the macOS launchers anyway since a self contained app can be copied, moved, etc independently of the other apps without breaking.

I asked about that on the jmriusers list.  I’d like to understand what users think of the alternatives.

rhwood, 2016-11-09:
>> I'd still like to be able to entirely encapsulate JMRI in the macOS launchers anyway since a self contained app can be copied, moved, etc independently of the other apps without breaking.

> I asked about that on the jmriusers list.  I’d like to understand what users think of the alternatives.

A summary of responses so far (in 53 messages by 30 respondents):
- 2 for keeping the current structure
- 7 for self-contained apps
  - 2 for distributing in a single DMGs (as in #2075 )
  - 2 for distributing as DecoderPro and PanelPro in separate DMGs
  - 3 had no preference on distribution
- 8 for combining DecoderPro and PanelPro into a single self-contained app
- 2 for using the App Store (implies self contained, but unclear if as a single app, or two different apps)
- 11 were non-comittal (mostly along the lines of "whatever's easiest" or "whatever the developers prefer")

bobjacobsen, 2016-11-14:
I read the record differently.  Multiple people expressed concerns about restructuring into separate monolithic apps.

In any case, I think the question has become moot, as the "not workable" approach of signing our usual distribution is now working in Jenkins. See the sample build [here](http://jmri.tagadab.com/jenkins/job/TestReleases/job/alpha-builds/job/Signed/).  This was also used for the 4.5.6 distribution that was just put out. 

The next step (day or two) is a bit of configuration work to make sure it's robust, then I'll add that to the Packages target.  We'll then be back to having our usual form of distribution working fine on all supported OS X / macOS versions.

The only remaining advantage for separate apps then seems to be that you can move them around. Given the various ways that people now modify program directory files (whether they should or not, they do, and that would break with monolithic app bundles), I think moving _solely_ to those is a step too far for now.

If somebody wants to adapt the build.xml to create _both_ the current form and a .dmg with separate apps (with or without SoundPro), that might be a way to get some experience with monolithic apps on Mac.  If people end up preferring those, a solution is found for CATS, etc, we can then stop creating the current form some time in the future with a small change to the release build process.

rhwood, 2016-11-15:
> I read the record differently. Multiple people expressed concerns about restructuring into separate monolithic apps.

If I recall correctly, three people expressed concerns about that: one about the possibility that a user might have multiple versions of JMRI installed (many do now, that's not a change; one (a Windows user) about having any installation of JMRI differ by platform (it's different on all platforms now, that's not a change; and one about needing to modify the default.lcf ( that's been fixed, although the default.lcf needs to be edited to reflect that). There was some confusion about the apps being different, but that's been a constant for years. In any event, there were people supporting the change than arguing against it.

pabender, 2016-11-15:
@rhwood, @bobjacobsen,

My take on the user discussion was that most users were indifferent about the change, deferring to "whatever is easiest for the developers".

Given the technical issues related to 3rd party applications (mainly CATS, but possibly others) that use JMRI in it's original intended purpose (as a library), I personally believe the approach that Bob has outlined is easier for us.

Paul

bobjacobsen, 2016-11-20:
The [Jenkins 'Builds' target]() is now producing signed Mac product files, with both the .dmg file and the three included app bundles signed.  Test and production releases from 4.5.7 will be in that form too.

With this, I think JMRI is OK on all currently supported Mac OS (macOS) versions.