JMRI/JMRI #2077: Setting up C/MRI from scratch is failing

bobjacobsen, 2016-10-14:
(From a jmriusers thread by James LeFevre)

At PanelPro startup, create a new profile, go to configure connection, select "C/MRI", then when you select "Serial" you get an exception which halts operation:

````
     [java] 2016-10-14 07:00:28,178 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrix.cmri.serial.serialdriver.ConnectionConfig.loadDetails(ConnectionConfig.java:36)
     [java] 	at jmri.jmrix.JmrixConfigPane.selection(JmrixConfigPane.java:334)
     [java] 	at jmri.jmrix.JmrixConfigPane.lambda$new$302(JmrixConfigPane.java:262)
     [java] 	at jmri.jmrix.JmrixConfigPane$$Lambda$49/261621514.actionPerformed(Unknown Source)
     [java] 	at javax.swing.JComboBox.fireActionEvent(JComboBox.java:1258)
````

pabender, 2016-10-14:
I'm sure this is related to #1662 which was fixed by #1668, but that should have fixed this for all C/MRI systems.

Did James say what version of JMRI he was using?

Paul

pabender, 2016-10-14:
Of course, I can replicate what James is seeing....

pabender, 2016-10-14:
I've found the cause of the NPE, adapter is null on the line in question.

If adapter is set prior to the call, then the display comes up, but with a "2" prefix and name, so there is an adapter being created, it just isn't associated with the ConnectionConfig object.

Now, I just need to find where the disconnect occurs between creating the new adapter and linking it to the ConnectionConfig object.

bobjacobsen, 2016-10-17:
Thanks.

The current HEAD of master (with #2078 included) still has a problem.  I've got an existing TMCC simulator connection.  Open PanelPro, add C/MRI, select Simulator, click Save, and get:

````
     [java] 2016-10-17 09:18:59,451 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrix.cmri.serial.serialdriver.configurexml.ConnectionConfigXml.extendElement(ConnectionConfigXml.java:40)
     [java] 	at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.store(AbstractSerialConnectionConfigXml.java:64)
     [java] 	at jmri.jmrix.configurexml.AbstractConnectionConfigXml.store(AbstractConnectionConfigXml.java:32)
     [java] 	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:495)
     [java] 	at jmri.jmrix.ConnectionConfigManager.lambda$savePreferences$298(ConnectionConfigManager.java:142)
     [java] 	at jmri.jmrix.ConnectionConfigManager$$Lambda$130/131779814.accept(Unknown Source)
     [java] 	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
     [java] 	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
     [java] 	at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:140)
     [java] 	at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:132)
     [java] 	at jmri.jmrix.swing.ConnectionsPreferencesPanel.savePreferences(ConnectionsPreferencesPanel.java:283)
     [java] 	at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:241)
     [java] 	at apps.gui3.TabbedPreferences.lambda$init$56(TabbedPreferences.java:154)
     [java] 	at apps.gui3.TabbedPreferences$$Lambda$56/1114143722.actionPerformed(Unknown Source)
     [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
     [java] 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
     [java] 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
````

Thinking from reading the code that this might be due to no C/MRI nodes existing, I repeated the same thing except configured nodes by adding an SMINI node, but on hitting the "Add" for that got:

````
     [java] 2016-10-17 09:22:51,577 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrix.cmri.serial.nodeconfig.NodeConfigFrame.addButtonActionPerformed(NodeConfigFrame.java:354)
     [java] 	at jmri.jmrix.cmri.serial.nodeconfig.NodeConfigFrame$3.actionPerformed(NodeConfigFrame.java:283)
     [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
     [java] 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
     [java] 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
     [java] 	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
     [java] 	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
     [java] 	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
````

pabender, 2016-10-18:
PR #2085 at least makes the simulator not throw those two NPEs.

I'm working on tests that will capture something like this in the future.

closed 2016-10-23 by bobjacobsen