JMRI/JMRI #2080: Race condition in LocoNet Simulator startup

bobjacobsen, 2016-10-15:
Occasionally, during startup with a LocoNet Simulator connection you'll see:

````
2016-10-13 18:48:09,951 node.NodeIdentity                     INFO  - Using jmri-00023F082735-3eb5b6e2 as the JMRI Node identity [main]
2016-10-13 18:48:12,379 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-4]
2016-10-13 18:48:12,655 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-6]
````

The middle message seems to be due to a race condition, where a LocoNet message send is attempted before the Simulator thread(s) are fully created.

This may be due to the LnSensorUpdateThread started in LnSensorManager.

Not yet known whether this ever hits a real (non-simulated) LocoNet connection. If it does, it becomes much more important to fix it.

klk32003, 2016-10-17:
I can confirm this log message when using real LocoNet hardware (Intellibox1 SW-V2.0).
Normally this happens once after PC startup.

```
2016-10-17 12:17:33,421 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-10-17 12:17:36,890 util.Log4JUtil                        INFO  - This log is appended to file: C:\Documents and Settings\Klaus\JMRI\log\messages.log [main]
2016-10-17 12:17:36,906 util.Log4JUtil                        INFO  - This log is stored in file: C:\Documents and Settings\Klaus\JMRI\log\session.log [main]
2016-10-17 12:17:37,046 apps.Apps                             INFO  - PanelPro version 4.4-R1ccf76b starts under Java 1.8.0_101 on Windows XP x86 v5.1 at Mon Oct 17 12:17:36 BST 2016 [main]
2016-10-17 12:17:48,125 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ec71bd7 [main]
2016-10-17 12:17:51,343 node.NodeIdentity                     INFO  - Using jmri-000296045F85-3ec71bd7 as the JMRI Node identity [main]
2016-10-17 12:17:54,953 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-10-17 12:17:55,343 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-6]
2016-10-17 12:18:14,703 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-10-17 12:18:14,734 util.FileUtilSupport                  INFO  - File path preference: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-17 12:18:14,734 util.FileUtilSupport                  INFO  - File path profile: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-17 12:18:14,750 util.FileUtilSupport                  INFO  - File path settings: is C:\Documents and Settings\Klaus\JMRI\ [main]
2016-10-17 12:18:14,828 util.FileUtilSupport                  INFO  - File path home: is C:\Documents and Settings\Klaus\ [main]
2016-10-17 12:18:14,890 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-10-17 12:18:36,593 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
2016-10-17 12:18:43,140 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-10-17 12:18:43,171 util.Log4JUtil                        INFO  - This log is appended to file: C:\Documents and Settings\Klaus\JMRI\log\messages.log [main]
2016-10-17 12:18:43,171 util.Log4JUtil                        INFO  - This log is stored in file: C:\Documents and Settings\Klaus\JMRI\log\session.log [main]
2016-10-17 12:18:43,203 apps.Apps                             INFO  - PanelPro version 4.4-R1ccf76b starts under Java 1.8.0_101 on Windows XP x86 v5.1 at Mon Oct 17 12:18:43 BST 2016 [main]
2016-10-17 12:18:44,921 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ec71bd7 [main]
2016-10-17 12:18:45,296 node.NodeIdentity                     INFO  - Using jmri-000296045F85-3ec71bd7 as the JMRI Node identity [main]
2016-10-17 12:18:47,031 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-10-17 12:18:55,296 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-10-17 12:18:55,375 util.FileUtilSupport                  INFO  - File path preference: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-17 12:18:55,421 util.FileUtilSupport                  INFO  - File path profile: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-17 12:18:55,484 util.FileUtilSupport                  INFO  - File path settings: is C:\Documents and Settings\Klaus\JMRI\ [main]
2016-10-17 12:18:55,531 util.FileUtilSupport                  INFO  - File path home: is C:\Documents and Settings\Klaus\ [main]
2016-10-17 12:18:55,609 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-10-17 12:19:15,140 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
```

klk32003, 2016-10-17:
Here are some more details ...

I added a line in LnSensorUpdateThread (started in LnSensorManager.java) asking for log.debug information:

```
2016-10-17 15:16:48,906 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-10-17 15:16:49,656 util.Log4JUtil                        INFO  - This log is appended to file: C:\Documents and Settings\Klaus\JMRI\log\messages.log [main]
2016-10-17 15:16:49,656 util.Log4JUtil                        INFO  - This log is stored in file: C:\Documents and Settings\Klaus\JMRI\log\session.log [main]
2016-10-17 15:16:49,796 apps.Apps                             INFO  - PanelPro version 4.4ish-201610171256-Klaus starts under Java 1.8.0_101 on Windows XP x86 v5.1 at Mon Oct 17 15:16:49 BST 2016 [main]
2016-10-17 15:17:00,156 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ec71bd7 [main]
2016-10-17 15:17:02,968 node.NodeIdentity                     INFO  - Using jmri-000296045F85-3ec71bd7 as the JMRI Node identity [main]
2016-10-17 15:17:07,046 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-10-17 15:17:07,437 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-6]
2016-10-17 15:17:07,750 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 78 27 00 [Thread-7]
2016-10-17 15:17:08,250 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 79 27 10 [Thread-7]
2016-10-17 15:17:08,750 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 7A 27 11 [Thread-7]
2016-10-17 15:17:09,250 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 7B 27 12 [Thread-7]
2016-10-17 15:17:09,765 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 78 07 13 [Thread-7]
2016-10-17 15:17:10,265 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 79 07 30 [Thread-7]
2016-10-17 15:17:10,765 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 7A 07 31 [Thread-7]
2016-10-17 15:17:11,265 .LnSensorManager$LnSensorUpdateThread DEBUG - message: B0 7B 07 32 [Thread-7]
2016-10-17 15:17:26,625 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-10-17 15:17:26,656 util.FileUtilSupport                  INFO  - File path preference: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-17 15:17:26,656 util.FileUtilSupport                  INFO  - File path profile: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-17 15:17:26,656 util.FileUtilSupport                  INFO  - File path settings: is C:\Documents and Settings\Klaus\JMRI\ [main]
2016-10-17 15:17:26,656 util.FileUtilSupport                  INFO  - File path home: is C:\Documents and Settings\Klaus\ [main]
2016-10-17 15:17:26,656 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
2016-10-17 15:19:29,546 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
```

Finally please find attached a session.log with full debug information:
[session.txt](https://github.com/JMRI/JMRI/files/533774/session.txt)

Regards,
Klaus

devel-bobm, 2016-10-17:
Klaus,

 

The full session.txt link you included gives the *critical clue*.  From your log:

 

2016-10-17 15:45:01,765 loconet.LnPacketizer                  DEBUG - queue LocoNet packet: BB 00 00 44 [Thread-6]

2016-10-17 15:45:01,765 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-6]

 

The BB 00 00 44 message is one which requests a read of slot 0.  Searching the source for "slot 0" reveals that jmri.jmrix.loconet.LnPowerManager performs a read of slot 0 to determine the "track power" status.  It is invoked as part of the LnPowerManager constructor.  The message is sent as part of a thread which has the following code:

 

        /**

         * Runs the thread - Waits a while (to allow the managers to initialize)

         * then sends a query of slot 0 so that the power manager can inspect

         * the {@code "<trk>"} byte.

         */

        public void run() {

            // wait a little bit to allow power manager to be initialized

            try {

                // Delay 200 mSec to allow init of traffic controller, listeners.

                Thread.sleep(200);

            } catch (InterruptedException e) {

                Thread.currentThread().interrupt(); // retain if needed later

            }

            LocoNetMessage msg = new LocoNetMessage(4);

            msg.setOpCode(LnConstants.OPC_RQ_SL_DATA);

            msg.setElement(1, 0);

            msg.setElement(2, 0);

            tc.sendLocoNetMessage(msg);

        }

 

That delay of 200 mSec is likely the source of the problem.  I reduced that delay to 5, and I was able to easily get the warning on my machine, and *confirm the slot 0 read is the source of the problem*, as shown in my log snippet here:

 

2016-10-17 11:58:05,029 node.NodeIdentity                     INFO  - Using jmri-0C8BFD976922-3d6abdb7 as the JMRI Node identity [main]

2016-10-17 11:58:05,420 locobuffer.LocoBufferAdapter          INFO  - COM9 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]

2016-10-17 11:58:05,452 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException for message Request data/status for slot 0.

contents: BB 00 00 44

[Thread-6]

2016-10-17 11:58:05,905 util.FileUtilSupport                  INFO  - File path program: is C:\Users\given\Documents\NetBeansProjects\JMRI_4_1_3b\JMRI\ [main]

2016-10-17 11:58:05,905 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\given\JMRI\DCS240_mine\ [main]

2016-10-17 11:58:05,905 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\given\JMRI\DCS240_mine\ [main]

2016-10-17 11:58:05,905 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\given\JMRI\ [main]

2016-10-17 11:58:05,905 util.FileUtilSupport                  INFO  - File path home: is C:\Users\given\ [main]

2016-10-17 11:58:05,905 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\given\Documents\NetBeansProjects\JMRI_4_1_3b\JMRI\jython\ [main]

 

*I expect that increasing that sleep delay from 200 to something like 500 mSec would help or resolve this problem.*  

 

Note that the interrogate sends its first message after a delay of 500 mSec, and subsequent interrogate messages are sent with 500 mSec delays.

devel-bobm, 2016-10-17:
@klk32003, my changes may have resolved this issue.  Can you try one of the install packages at [The Jenkins Project ackages page ](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/) ?  The changes found in Build 893 there ought to remove the LnPacketizer warning from the log.

klk32003, 2016-10-18:
Bob M.,

Thank you for your detailed analysis. 

I installed Build 893 from Jenkins as suggested. The result is that I recieve two messages now. 

```
2016-10-18 20:10:50,609 apps.Apps                             INFO  - PanelPro version 4.5.6ish-201610172040-jenkins-Re9b1033 starts under Java 1.8.0_101 on Windows XP x86 v5.1 at Tue Oct 18 20:10:50 BST 2016 [main]
2016-10-18 20:11:03,265 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ea79590 [main]
2016-10-18 20:11:06,296 node.NodeIdentity                     INFO  - Using jmri-000296045F85-3ea79590 as the JMRI Node identity [main]
2016-10-18 20:11:09,015 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-10-18 20:11:09,843 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-6]
2016-10-18 20:11:09,937 loconet.LnPacketizer                  WARN  - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-7]
2016-10-18 20:11:31,859 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
```

Yesterday I did some other tests with JMRI 4.4. and a delay of 500 mSec and 700 mSec (I did the build with "ant dist").
The result with 500 mSec: not homogeneous *
The result with 700 mSec: no more WARN messages!

It might be relevant here, that my PC is old/slow. Could that be the reason for the need of higher delay values just in my case? 

*) I did all the tests three times.

KenC57, 2016-10-18:
I question if there isn't a way to test if the rest of the initialization
has finished or not and to test for that case instead of trying to pick a
delay that 'might be enough'. As it goes how much is enough could vary quite
a bit from computer to computer. So we end up with still not enough in some
cases but just wasting time for others. Ok, only wasting some multiple of
100ms, but Adm Hooper wouldn't stand for that.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

pabender, 2016-10-18:
On Oct 18, 2016 3:46 PM, "Ken Cameron" <notifications@github.com> wrote:
>
> I question if there isn't a way to test if the rest of the initialization
> has finished or not and to test for that case instead of trying to pick a
> delay that 'might be enough'. As it goes how much is enough could vary
quite
> a bit from computer to computer.

I think Ken has the right idea here.

What we are really seeeing is a flow control problem.

There either needs to be a flag and/or a wait/notify pair that allows rhe
communication initilization to complete before the power manager tries to
ask for a messahe to be sent.

In the general case of AbstractMRTrafficController, there is already a flag
in the adapters ( exposed via okToSend () ) .... but loconet doesn't  (
that i remember  ) use AbstractMRTrafficController.

Paul

devel-bobm, 2016-10-18:
Yes, Klaus, there does seem to be a strong dependence upon the speed of the machine.  My machine is quite recent and fast, so I rarely saw the problem until I dropped the wait to 5mSec.

 

Ken's point and Paul's comment need investigation to see if a "right" solution can be easily found.

 

Regards,

Bob M.

PUlvestad, 2016-10-22:
I just upgraded to JMRI.4.5.6ish-201610221740-jenkins-R65c8836 and after several restarts on both my profiles have not received the error message again as I originally reported prior to the 4.5.5 release.

 2016-10-22 14:44:51,134 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-10-22 14:44:51,433 util.Log4JUtil                        INFO  - This log is appended to file: /home/peter/.jmri/log/messages.log [main]
2016-10-22 14:44:51,447 util.Log4JUtil                        INFO  - This log is stored in file: /home/peter/.jmri/log/session.log [main]
2016-10-22 14:44:51,504 apps.Apps                             INFO  - PanelPro version 4.5.6ish-201610221740-jenkins-R65c8836 starts under Java 1.8.0_101 on Linux i386 v4.0.5-antix.1-486-smp at Sat Oct 22 14:44:51 MDT 2016 [main]
2016-10-22 14:44:53,951 apps.Apps                             INFO  - Starting with profile JMRI.3eb5b6e2 [main]
2016-10-22 14:44:54,097 node.NodeIdentity                     INFO  - Using jmri-00023F082735-3eb5b6e2 as the JMRI Node identity [main]
2016-10-22 14:44:55,062 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-6]
2016-10-22 14:44:57,214 swing.JmriJTablePersistenceManager    INFO  - Table preferences not found.
This is expected on the first time the "JMRI" profile is used on this computer. [main]
2016-10-22 14:44:58,260 util.FileUtilSupport                  INFO  - File path program: is /home/peter/JMRI/ [main]
2016-10-22 14:44:58,262 util.FileUtilSupport                  INFO  - File path preference: is /home/peter/Dropbox/JMRI/ [main]
2016-10-22 14:44:58,263 util.FileUtilSupport                  INFO  - File path profile: is /home/peter/.jmri/JMRI/ [main]
2016-10-22 14:44:58,269 util.FileUtilSupport                  INFO  - File path settings: is /home/peter/.jmri/ [main]
2016-10-22 14:44:58,275 util.FileUtilSupport                  INFO  - File path home: is /home/peter/ [main]
2016-10-22 14:44:58,276 util.FileUtilSupport                  INFO  - File path scripts: is /home/peter/Dropbox/JMRI/Scripts/ [main]

klk32003, 2016-10-23:
Surprisingly I have also an error-free result today after a `git pull` and with real (non-simulated) LocoNet connection ...

```
2016-10-23 17:45:02,578 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-10-23 17:45:03,546 util.Log4JUtil                        INFO  - This log is appended to file: C:\Documents and Settings\Klaus\JMRI\log\messages.log [main]
2016-10-23 17:45:03,546 util.Log4JUtil                        INFO  - This log is stored in file: C:\Documents and Settings\Klaus\JMRI\log\session.log [main]
2016-10-23 17:45:03,671 apps.Apps                             INFO  - PanelPro version 4.5.6ish-201610230929-Klaus starts under Java 1.8.0_111 on Windows XP x86 v5.1 at Sun Oct 23 17:45:03 BST 2016 [main]
2016-10-23 17:45:17,921 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3e818da6 [main]
2016-10-23 17:45:20,218 node.NodeIdentity                     INFO  - Using jmri-000296045F85-3e818da6 as the JMRI Node identity [main]
2016-10-23 17:45:22,875 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2016-10-23 17:45:40,875 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2016-10-23 17:45:40,890 util.FileUtilSupport                  INFO  - File path preference: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-23 17:45:40,890 util.FileUtilSupport                  INFO  - File path profile: is C:\Documents and Settings\Klaus\JMRI\My_JMRI_Railroad\ [main]
2016-10-23 17:45:40,890 util.FileUtilSupport                  INFO  - File path settings: is C:\Documents and Settings\Klaus\JMRI\ [main]
2016-10-23 17:45:40,906 util.FileUtilSupport                  INFO  - File path home: is C:\Documents and Settings\Klaus\ [main]
2016-10-23 17:45:40,906 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]
```

Regards,
Klaus

closed 2017-05-02 by devel-bobm