JMRI/JMRI #2081: TMCC throttle support broken

bobjacobsen, 2016-10-16:
A user reports that TMCC throttle support is broken in 4.5.4 and the current 4.5.5 candidate.

Throttle operations don't send anything in the monitor. The monitor is seeing (unsolicited) messages from the layout, and sending a message (from Packet Send) does send.

Perhaps broken when multi-system was added after JMRI 4.4?

bobjacobsen, 2016-10-16:
Problem seems specific to the throttle support.  The turnout control does send to the serial port (created a new profile with just a Lionel TMCC connection to a dummy serial port to test this).

The TMCC throttle manager doesn't seem to be represented in the TMCC SystemConnectionMemo:

````
% grep -r ThrottleManager java/src/jmri/jmrix/tmcc/
java/src/jmri/jmrix/tmcc//serialdriver/SerialDriverAdapter.java:        jmri.InstanceManager.setThrottleManager(new jmri.jmrix.tmcc.SerialThrottleManager());
java/src/jmri/jmrix/tmcc//SerialThrottleManager.java:import jmri.jmrix.AbstractThrottleManager;
java/src/jmri/jmrix/tmcc//SerialThrottleManager.java: * Implementation of a ThrottleManager.
java/src/jmri/jmrix/tmcc//SerialThrottleManager.java:public class SerialThrottleManager extends AbstractThrottleManager {
java/src/jmri/jmrix/tmcc//SerialThrottleManager.java:    public SerialThrottleManager() {
java/src/jmri/jmrix/tmcc//SerialThrottleManager.java:    private final static Logger log = LoggerFactory.getLogger(SerialThrottleManager.class.getName());
java/src/jmri/jmrix/tmcc//TMCCSystemConnectionMemo.java:        super("T", "Lionel TMCC"); // Prefix from SerialTurnoutManager, UserName from SerialThrottleManager
````

Compare that to e.g. NCE, which has a lot more support in its SystemConnectionMemo: 
````
% grep -r ThrottleManager java/src/jmri/jmrix/nce
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:        if (type.equals(jmri.ThrottleManager.class)) {
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:        if (T.equals(jmri.ThrottleManager.class)) {
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:            return (T) getThrottleManager();
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:    private NceThrottleManager throttleManager;
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:        throttleManager = new jmri.jmrix.nce.NceThrottleManager(this);
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:        InstanceManager.setThrottleManager(throttleManager);
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:    public NceThrottleManager getThrottleManager() {
java/src/jmri/jmrix/nce/NceSystemConnectionMemo.java:            InstanceManager.deregister(throttleManager, jmri.jmrix.nce.NceThrottleManager.class);
java/src/jmri/jmrix/nce/NceThrottle.java: * Based on Glen Oberhauser's original LnThrottleManager implementation
java/src/jmri/jmrix/nce/NceThrottleManager.java:import jmri.jmrix.AbstractThrottleManager;
java/src/jmri/jmrix/nce/NceThrottleManager.java: * NCE implementation of a ThrottleManager.
java/src/jmri/jmrix/nce/NceThrottleManager.java:public class NceThrottleManager extends AbstractThrottleManager {
java/src/jmri/jmrix/nce/NceThrottleManager.java:    public NceThrottleManager(NceSystemConnectionMemo memo) {
java/src/jmri/jmrix/nce/NceThrottleManager.java:    private final static Logger log = LoggerFactory.getLogger(NceThrottleManager.class.getName());
````

pabender, 2016-10-17:
On 10/16/2016 05:25 PM, Bob Jacobsen wrote:
> Problem seems specific to the throttle support. The turnout control
> does send to the serial port (created a new profile with just a Lionel
> TMCC connection to a dummy serial port to test this). [...]
|
@bobjacobsen,

I went back and looked, the TMCC code hasn't actually been migrated to 
the multiple connection format yet.

I'm thinking this problem may have more to do with issue #1014 than 
anything else.

The clue is actually the method that the line above is contained in:

     /**
      * set up all of the other objects to operate connected to this port
      */
     public void configure() {
         // connect to the traffic controller
         SerialTrafficController.instance().connectPort(this);

jmri.InstanceManager.setTurnoutManager(jmri.jmrix.tmcc.SerialTurnoutManager.instance());
         jmri.InstanceManager.setThrottleManager(new 
jmri.jmrix.tmcc.SerialThrottleManager());

         jmri.jmrix.tmcc.ActiveFlag.setActive();
     }

We're still calling the instance methods in both the traffic controller 
and the turnout manager for TMCC, and neither the TurnoutManager or 
ThrottleManager are constructed with a reference to either the memo or 
the traffic controller.

I haven't tried this yet, but my guess is that if you disable the 
internal throttle manager, you will
see traffic from the TMCC throttle.

Paul
|

bobjacobsen, 2016-11-13:
Fixed by #2442, #2473, confirmed by user.

closed 2016-11-13 by bobjacobsen