JMRI/JMRI #2086: Create test release 4.5.6

bobjacobsen, 2016-10-18:
For discussion of timing and what to include in 4.5.6.

This is (intended to be) the last normally-scheduled "from HEAD of master" test release in the 4.5/4.6 cycle this Fall.  

The next is now planned to be 4.5.7 about two weeks later from HEAD, then a short-cycle of "targeted changes only" test releases as needed.

bobjacobsen, 2016-10-23:
We're still working on the macOS distribution question, see #2074

bobjacobsen, 2016-11-13:
Rebuilding help index failed, opened Issue #2505 but will continue.

bobjacobsen, 2016-11-13:
The release-4.5.6 branch has been created. 

From now on, please document your changes in the [jmri4.5.7.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.5.7.shtml) release note file.

Maintainers, please set the 4.5.7 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.5.6/)

devel-bobm, 2016-11-13:
@bobjacobsen : presumably we need a 4.7.1 branch , for those changes not meeting the ""targeted changes only" criteria for 4.5.7.  Correct?

bobjacobsen, 2016-11-13:
@devel-bobm: No, those just stay as PRs to master.  It's the targeted changes that need special handling.

bobjacobsen, 2016-11-13:
Also, 4.5.7 is a typical release from the head of master.  It's 4.5.8 et al (as needed) that get special handling.  We've talked about using another digit (e.g. 4.5.7, then 4.5.7.1, etc) for that. I haven't seen a consensus about that, so haven't put the tooling in place.

devel-bobm, 2016-11-13:
I didn't _carefully_ read the last sentence of the initial comment - the sentence about 4.5.7, and so I incorrectly thought 4.5.7 was for targeted-fixes-only.

bobjacobsen, 2016-11-14:
Released.

closed 2016-11-14 by bobjacobsen