JMRI/JMRI #2088: Bluetooth connection

Milebank, 2016-10-20:
Running JMRI Panel Pro via a Möllehem Buetooth connection to Digitrax Loconet.
Laptop running Win 10 with whatever updates have arrived, latest production release of JMRI.

Keeps losing the Bluetooth connection.

Recovery is to close and restart Panel Pro.

Any ideas?

bobjacobsen, 2016-10-20:
Could you please check the console log (in Help menu) to see if there are any messages?

Milebank, 2016-10-25:
Hi Bob,

It will run for about 3 mins up to maybe 4 hours before stopping.

I have mostly just left it  - no inputs - but the JMRI Panel Pro seems 
to do occasional requests, and resets the speed of two engines to zero - 
again occasionally.

I have two 'routes' set up with which I can set all points to either 
Closed or Open. If I do run these, then they work OK and do not seem to 
prompt the failure.

Just in case it helps, attached are 3 error files from the PanelPro 
console, the overall panel/config file and the roster file.

Any further ideas most welcome.

Regards

Pete W.

2016-10-23 15:53:32,148 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2016-10-23 15:53:32,195 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\user\JMRI\log\messages.log [main]
2016-10-23 15:53:32,195 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\user\JMRI\log\session.log [main]
2016-10-23 15:53:32,210 apps.Apps                             INFO  - PanelPro version 4.4-R1ccf76b starts under Java 1.8.0_111 on Windows 10 x86 v10.0 at Sun Oct 23 15:53:32 BST 2016 [main]
2016-10-23 15:53:32,960 apps.Apps                             INFO  - Starting with profile PetesModel.3f3aab6f [main]
2016-10-23 15:53:34,351 node.NodeIdentity                     INFO  - Using jmri-4CBB580D2FBD-3f3aab6f as the JMRI Node identity [main]
BlueCove log redirected to log4j
BlueCove version 2.1.1-SNAPSHOT on winsock
2016-10-23 15:53:38,604 configurexml.PerformFileModelXml      INFO  - Load file C:\Users\user\Documents\ModelRailway\JMRI\PetesModel\GA21.xml [main]
2016-10-23 15:53:43,776 jmri.Block                            INFO  - Sensor ACTIVE came out of nowhere, no neighbors active for block IB41. Value not set. [AWT-EventQueue-0]
2016-10-23 15:53:43,776 configurexml.PerformFileModelXml      INFO  - Load file C:\Users\user\Documents\ModelRailway\JMRI\PetesModel\roster.xml [main]
2016-10-23 15:53:43,776 jmri.Block                            INFO  - Sensor ACTIVE came out of nowhere, no neighbors active for block IB4. Value not set. [AWT-EventQueue-0]
2016-10-23 15:53:43,776 jmri.Block                            INFO  - Sensor ACTIVE came out of nowhere, no neighbors active for block IB11. Value not set. [AWT-EventQueue-0]
2016-10-23 15:53:43,776 jmri.Block                            INFO  - Sensor ACTIVE came out of nowhere, no neighbors active for block IB33. Value not set. [AWT-EventQueue-0]
2016-10-23 15:53:43,792 jmri.Block                            INFO  - Sensor ACTIVE came out of nowhere, no neighbors active for block IB28. Value not set. [AWT-EventQueue-0]
2016-10-23 15:53:44,417 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2016-10-23 15:53:44,417 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\user\Documents\ModelRailway\JMRI\PetesModel\ [main]
2016-10-23 15:53:44,417 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\user\Documents\ModelRailway\JMRI\PetesModel\ [main]
2016-10-23 15:53:44,417 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\user\JMRI\ [main]
2016-10-23 15:53:44,417 util.FileUtilSupport                  INFO  - File path home: is C:\Users\user\ [main]
2016-10-23 15:53:44,417 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2016-10-23 15:53:45,510 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [init prefs]
2016-10-23 15:53:49,011 logix.WarrantPreferences              INFO  - Found Warrant preferences file: C:\Users\user\Documents\ModelRailway\JMRI\PetesModel\signal\WarrantPreferences.xml [init prefs]
2016-10-23 15:55:34,893 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: IOException: java.io.IOException: Failed to write; [10053] An established connection was aborted by the software in your host machine. [LocoNet transmit handler]
2016-10-23 15:55:34,908 loconet.LnPacketizer                  INFO  - End of file [LocoNet receive handler]
2016-10-23 15:55:35,018 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,158 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,268 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,408 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,518 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,658 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,768 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 15:55:35,909 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
[... cut ...]
2016-10-23 14:30:44,015 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 14:30:54,026 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 14:31:04,038 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet transmit handler]
2016-10-23 14:31:14,047 loconet.LnPacketizer                  WARN  - sendLocoNetMessage: no connection established [LocoNet

bobjacobsen, 2016-10-26:
> On Oct 25, 2016, at 11:21 AM, Milebank <notifications@github.com> wrote:
> It will run for about 3 mins up to maybe 4 hours before stopping.
> 
> I have mostly just left it - no inputs - but the JMRI Panel Pro seems 
> to do occasional requests, and resets the speed of two engines to zero - 
> again occasionally.
> 
> I have two 'routes' set up with which I can set all points to either 
> Closed or Open. If I do run these, then they work OK and do not seem to 
> prompt the failure.
> 
> Just in case it helps, attached are 3 error files from the PanelPro 
> console, the overall panel/config file and the roster file.

> 2016-10-23 15:55:34,893 loconet.LnPacketizer WARN - sendLocoNetMessage: IOException: java.io.IOException: Failed to write; [10053] An established connection was aborted by the software in your host [...]

> 2016-10-23 17:27:13,158 loconet.LnPacketizer WARN - sendLocoNetMessage: IOException: java.io.IOException: Failed to write; [10053] An established connection was aborted by the software in your host [...]

> 2016-10-23 13:52:41,493 loconet.LnPacketizer WARN - sendLocoNetMessage: IOException: java.io.IOException: Failed to write; [10053] An established connection was aborted by the software in your host [...]

Something is happening in the BlueTooth connection (hardware or software) that causes the Windows host to drop the serial connection that JMRI is using. It’s happening outside of JMRI (that’s what the “java.io.IOException: Failed to write; [10053]” means)  

It’s possible, but unlikely, that JMRI’s causing this.  More likely is something is going wrong outboard of JMRI, in the BlueTooth hardware/software or in the Windows support for it.  Loose connection, poor power supply regulation, ??

There are some web pages where people are reporting similar issues with the BlueCove library that handles Windows access to BlueTooth. Unfortunately, most of them don’t have solutions.

Milebank, 2016-12-30:
It seems to be better now I've a Win10 update (or several). However, I still occasionally lose the Bluetooth connection. The recovery is to close / re-open Panel Pro.

Is there a way of re-initialising the Bluetooth with JMRI to save the close/ re-opn?

bobjacobsen, 2016-12-30:
> Is there a way of re-initialising the Bluetooth with JMRI to save the close/ re-opn?

Probably not.  The exception that's being raised (e.g. the software error) indicates that something in the Microsoft WinSock code attached to this run of JMRI has failed and can't cope.  I don't know of a way to fix this other than quoting and restarting JMRI.  That kills the old WinSock instance and creates a new one.  I don't know of a way for a Java program to do this by itself, while still running.  Microsoft probably didn't provide such a method because the 10053 error is one of those "not supposed to happen" things.

Is the problem that the BlueTooth is a the edge of its range, and sometimes does something that the driver code can't cope with?

Milebank, 2017-01-01:
Hi Bob,

Thanks for this.

The next step seems to be to get a USB/Blutooth adapter that is supposed 
to work with Win 10.

More news later.

Regards

Pete W.

rhwood, 2020-04-15:
@Milebank Is this still a problem?

closed 2021-04-14 by stale[bot]