JMRI/JMRI #2091: request: Error when running script RosterCsvExport.py with JMRI 4.5.5 Win10 Pro and Java 1.8.0_101)

alain355, 2016-10-21:
Hi team,

I just updated to 4.5.5 from 4.5.4 and wanted to extract my roster as a CSV file. I then ran the provided script **RosterCsvExport.py** . It took a few seconds to run, but no file name was asked for and no result were provided.

Btw, I had no more resulting csv file using previous version 4.5.4. I also tried going back to 4.4.0, which went a bit further, the script asking for a file name and a location to save it, but creating an empty csv file (0 byte).

I am using a Win10 32 bits Pro Edition, updated to Anniversary, with Java 1.8.0_111 

Thanks if someone can fix the issue.

Alain
Session log file is attached
[session.log.txt](https://github.com/JMRI/JMRI/files/544658/session.log.txt)

rhwood, 2016-10-22:
What happens if you change "Print" to "print" (use a lowercase "p" instead of an uppercase "P") at line 167 in that script?

alain355, 2016-10-22:
Hi
Thanks for your reply.
I changed Print to print, also checked no more Print in the file and ran it getting the same error. Log file and updated script attached with .txt extension

[RosterCsvExport - Copie.py.txt](https://github.com/JMRI/JMRI/files/545663/RosterCsvExport.-.Copie.py.txt)
[session.log.txt](https://github.com/JMRI/JMRI/files/545662/session.log.txt)

closed 2016-10-22 by alain355

alain355, 2016-10-22:
Update: it works with **print** instead of **Print** in line 167: I had some cache issues and the script running was not the updated one (in Program files\JMRI\jython) but JMRI used the previous version cached in users\Alain\AppData\Local\virtualStore\Program File\JMRI\Jython. I discovered the cache issue as I slightly changed the error message but didn't saw the new one in the console...

rhwood, 2016-10-22:
@alain355 Thanks for testing this. #2092 includes the fix and will be in the next release.

dheap, 2016-10-22:
You will encounter VirtualStore issues if you:
- Install JMRI "for this user only".
- Customise any of the JMRI code (definitions, scripts, etc.) in Program Location rather than making your own copy in User Files Location.

Dave

> @alain355
> 
> Update: it works with print instead of Print in line 167: I had some cache issues and the script running was not the updated one (in Program files\JMRI\jython) but JMRI used the previous version cached in users\Alain\AppData\Local\virtualStore\Program File\JMRI\Jython. I discovered the cache issue as I slightly changed the error message but didn't saw the new one in the console...

alain355, 2016-10-23:
Thanks Dave for the explanation,
As far as I remember, the only customization done was adding and checking some new decoder definitions. However, when installing next 4.5.6, I'll double check it's installed for abyone using the computer ;)

Regards from Paris

Alain

dheap, 2016-10-23:
The problem is that if you manually modify scripts, definitions or other items in "C:\Program Files(x86)" or other system locations, Windows UAC simply pits your modified version in VirtualStore. The VirtualStore contents basically invisibly overlay the real directory and conceal the original version.

> @alain355 As far as I remember, the only customization done was adding and checking some new decoder definitions. However, when installing next 4.5.6, I'll double check it's installed for abyone using the computer ;)

alain355, 2016-10-23:
Yes Dave, that's what I discovered. I first launched the updated file from the Virtual Store location, then to update in the real folder, I had to delete the original file as Windows Admin in the real system location, then paste the updated copy from the Virtual Store location.

dheap, 2016-10-25:
You will encounter VirtualStore issues if you:
- Install JMRI "for this user only".
- Customise any of the JMRI code (definitions, scripts, etc.) in Program Location rather than making your own copy in User Files Location.

Dave

> @alain355
>
> Update: it works with print instead of Print in line 167: I had some cache issues and the script running was not the updated one (in Program files\JMRI\jython) but JMRI used the previous [...]