JMRI/JMRI #2093: Soundtraxx TSU2 & Eco

mmosher5501, 2016-10-21:
Added to files ID info from:
http://www.soundtraxx.com/manuals/productID.pdf
changed "productID" to values listed in doc.  Java code will need to be modified to allow with TSU2 and ECO decoders to read CV 256 during decoder ident routine to get exact decoder.
[TSU2.zip](https://github.com/JMRI/JMRI/files/545317/TSU2.zip)

dheap, 2016-10-22:
@mmosher5501 I'll take a look at the ID code and get back to you.
Dave

@mmosher5501
Added to files ID info from:
http://www.soundtraxx.com/manuals/productID.pdf
changed "productID" to values listed in doc. Java code will need to be modified to allow with TSU2 and ECO decoders to read CV 256 during decoder ident routine to get exact decoder.

mmosher5501, 2016-10-22:
@dheap That version will read the product ID CV but does not use it to select just one decoder, all decoders matching CV8 & 7 are selected.
ECO-100 diesel
console
2016-10-22 10:49:09,440 decoderdefn.IdentifyDecoder           INFO  - Decoder returns mfgID:141;modelID:70;productID:2 [AWT-EventQueue-0]

Loconet
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV8 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV8 value 141 (0x8d, 10001101).
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV7 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV7 value 70 (0x46, 1000110).
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV256 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV256 value 2 (0x2, 10).

ECO-21P diesel
console
2016-10-22 10:53:10,960 decoderdefn.IdentifyDecoder           INFO  - Decoder returns mfgID:141;modelID:70;productID:8 [AWT-EventQueue-0]

Loconet
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV8 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV8 value 141 (0x8d, 10001101).
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV7 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV7 value 70 (0x46, 1000110).
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV256 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV256 value 8 (0x8, 1000).

TSU-2200 steam
console
2016-10-22 10:54:58,631 decoderdefn.IdentifyDecoder           INFO  - Decoder returns mfgID:141;modelID:71;productID:17 [AWT-EventQueue-0]

loconet
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV8 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV8 value 141 (0x8d, 10001101).
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV7 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV7 value 71 (0x47, 1000111).
Programming Track Request: Read Byte in Paged Mode on Service Track variable  CV256 value 255 (0xff, 11111111).
Programming Track Response: Read Byte in Paged Mode on Service Track Was Successful, set  CV256 value 17 (0x11, 10001).
[image]
[image]

mmosher5501, 2016-10-22:
@dheap It works on the three decoders I have, ECO-100 diesel, ECO-21P diesel, and TSU-2200 steam.

If you got emailed the pervious comment that I deleted here, ignore it, since my train computer died, using a laptop but did not get the updated files into correct profile's decoder folder.

dheap, 2016-10-22:
Good. I'll add that code as a standalone PR.

The definitions you sent as a .zip showed thousands of changes (probably due to whitespace formatting changes) and I was working on a small screen laptop, so I'll deal with them separately. I want to do a compare ignoring whitespace before committing them.

> @mmosher5501
> 
> @dheap It works on the three decoders I have, ECO-100 diesel, ECO-21P diesel, and TSU-2200 steam.

mmosher5501, 2016-10-22:
remove spaces around "=".
[TSU2.zip](https://github.com/JMRI/JMRI/files/546028/TSU2.zip)

dheap, 2016-10-22:
#2098 has added the necessary ID code.

dheap, 2016-10-23:
@mmosher5501  
Is the second .zip you posted an updated set of files, or a duplicate of the the first set?

dheap, 2016-10-23:
@mmosher5501 worked it out thanks. Will finish tomorrow.

dheap, 2016-10-25:
@mmosher5501 I'll take a look at the ID code and get back to you.
Dave

> @mmosher5501
> 
> Added to files ID info from:
> http://www.soundtraxx.com/manuals/productID.pdf
> changed "productID" to values listed in doc. Java code will need to be modified to allow with TSU2 and ECO decoders to read CV 256 during decoder ident routine to get exact decoder.
> TSU2.zip

mmosher5501, 2016-10-25:
Removed spaces around = signs in second file.Sent from XFINITY Connect Mobile App------ Original Message ------From: Dave HeapTo: JMRI/JMRICc: Michael Mosher, MentionSent: October 23, 2016 at 2:29 AMSubject: Re: [JMRI/JMRI] Soundtraxx TSU2 & Eco (#2093)@mmosher5501
Is the second .zip you posted an updated set of files, or a duplicate of the the first set?

mmosher5501, 2016-10-25:
Yes, I run the re-indent in XML Marker to fix some indent issues, but it 
apparently also put spaces around = as well, I've fixed it and will 
re-upload.

Michael Mosher
Director NJ Div, MER, NMRA  http://www.njdivnmra.org/
Webmaster, ECSFM            http://www.eastcoastsantafemodelers.org
Member SFRH&MS              http://www.atsfrr.net/
DCC Master PVSMR            http://www.patcongvalley.com/

dheap, 2016-10-25:
Sorry, I forgot to deal with this thread. The job is all done. #2098 added the ID code, #2420 added the changes to the decoder definitions and #2422 cleaned up the whitespace according to our NetBeans standards.

closed 2016-10-25 by dheap

dheap, 2016-10-25:
Sorry for the confusion. A couple of comments I tried to add via email 3-4 days ago finally appeared, creating outdated duplicates.